platformio/platformio-core #3710: Unexpected compiler output. This compiler might be unsupported And can not recognize the header file, but it can be compiled

a97077088, 2020-10-22:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:Win10 x64，python3.9(x64)

**PlatformIO Version** (`platformio --version`):platformio.exe -c clion run --target upload -e stc89c52rc

### Description of problem

#### Steps to Reproduce

1.create new project
2.select stc89c52rc
3.open clion pio init

### Actual Results

-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/pouk/go/src/c51/cmake-build-stc89c52rc

Problems were encountered while collecting compiler information:
	Unexpected compiler output. This compiler might be unsupported.
	If you are using GCC/Clang, please report the bug in https://youtrack.jetbrains.com/issues/CPP.

[image]
[image]
[image]

### Expected Results

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:stc89c52rc]
platform = intel_mcs51
board = stc89c52rc
```

**Source file to reproduce issue:**
```# !!! WARNING !!! AUTO-GENERATED FILE, PLEASE DO NOT MODIFY IT AND USE
# https://docs.platformio.org/page/projectconf/section_env_build.html#build-flags
#
# If you need to override existing CMake configuration or add extra,
# please create `CMakeListsUser.txt` in the root of project.
# The `CMakeListsUser.txt` will not be overwritten by PlatformIO.

cmake_minimum_required(VERSION 3.13)
set(CMAKE_SYSTEM_NAME Generic)
set(CMAKE_C_COMPILER_WORKS 1)
set(CMAKE_CXX_COMPILER_WORKS 1)

project("c51" C CXX)

include(CMakeListsPrivate.txt)

if(EXISTS ${CMAKE_CURRENT_SOURCE_DIR}/CMakeListsUser.txt)
include(CMakeListsUser.txt)
endif()

add_custom_target(
    Production ALL
    COMMAND platformio -c clion run "$<$<NOT:$<CONFIG:All>>:-e${CMAKE_BUILD_TYPE}>"
    WORKING_DIRECTORY ${CMAKE_CURRENT_SOURCE_DIR}
)

add_custom_target(
    Debug ALL
    COMMAND platformio -c clion run --target debug "$<$<NOT:$<CONFIG:All>>:-e${CMAKE_BUILD_TYPE}>"
    WORKING_DIRECTORY ${CMAKE_CURRENT_SOURCE_DIR}
)

add_executable(Z_DUMMY_TARGET ${SRC_LIST})

```

CMakeListsPrivate.txt
```
# !!! WARNING !!! AUTO-GENERATED FILE, PLEASE DO NOT MODIFY IT AND USE
# https://docs.platformio.org/page/projectconf/section_env_build.html#build-flags
#
# If you need to override existing CMake configuration or add extra,
# please create `CMakeListsUser.txt` in the root of project.
# The `CMakeListsUser.txt` will not be overwritten by PlatformIO.

set(CMAKE_CONFIGURATION_TYPES "stc89c52rc" CACHE STRING "Build Types reflect PlatformIO Environments" FORCE)

SET(CMAKE_C_COMPILER "$ENV{HOMEDRIVE}$ENV{HOMEPATH}/.platformio/packages/toolchain-sdcc/bin/sdcc.exe")
SET(CMAKE_CXX_COMPILER "C:/Program Files/mingw-w64/x86_64-8.1.0-posix-seh-rt_v6-rev0/mingw64/bin/cc.exe")
SET(CMAKE_CXX_FLAGS "--opt-code-size --peep-return -mmcs51")
SET(CMAKE_C_FLAGS "--std-sdcc11 --opt-code-size --peep-return -mmcs51")

if (CMAKE_BUILD_TYPE MATCHES "stc89c52rc")
    add_definitions(-D'F_CPU=11059200')
    add_definitions(-D'HEAP_SIZE=128')
    add_definitions(-D'PLATFORMIO=50002')

    include_directories("${CMAKE_CURRENT_LIST_DIR}/include")
    include_directories("${CMAKE_CURRENT_LIST_DIR}/src")
    include_directories("$ENV{HOMEDRIVE}$ENV{HOMEPATH}/.platformio/packages/toolchain-sdcc/non-free/include")
    include_directories("$ENV{HOMEDRIVE}$ENV{HOMEPATH}/.platformio/packages/tool-unity")

    FILE(GLOB_RECURSE EXTRA_LIB_SOURCES
        ${CMAKE_CURRENT_LIST_DIR}/.pio/libdeps/stc89c52rc/*.*
    )
endif()

FILE(GLOB_RECURSE SRC_LIST
    ${CMAKE_CURRENT_LIST_DIR}/src/*.*
    ${CMAKE_CURRENT_LIST_DIR}/lib/*.*
)

list(APPEND SRC_LIST ${EXTRA_LIB_SOURCES})

```
### Additional info

Anivie, 2022-10-07:
Do you solvee this problem?
I am get the same error in CLion 2022.3 with windows10 and ubuntu desktop 22.10

aadog, 2022-10-09:
no

XZ8886, 2023-11-07:
Did anyone solve the problem?

CarmJos, 2025-02-21:
Is there somebody have any solutions?