platformio/platformio-core #3712: Timestamp is "wrong" in device monitor output using time filter

hoegge, 2020-10-23:
### Configuration

**Operating system**: MacOS X 10.13.6

**PlatformIO Version** (`platformio --version`): Core 5.0.1·Home 3.3.0

### Description of problem

The timestamp on a new line is shown after a newline / return. That means the time is determined by the previous line and not the current line. So if there is a pause between output, the time of a line will be the time of the previous line and then can be several hours wrong.

#### Steps to Reproduce

1.  Make a solution produce out put with intervals that are "long enough" to see that the time of a line is the time of the previous line
2. E.g. print time as output on every line with e.g. 10 seconds between each output
3. and then it is clear that the time don't match

### Actual Results

### Expected Results
Timestamp should be the time of first output on a line and not the time of newline /return of the previous line. So don't print the time stamp until there is output on a line

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:nanoatmega328]
platform = atmelavr
board = nanoatmega328
framework = arduino
upload_port = /dev/cu.wchusbserialfa140
monitor_port = /dev/cu.wchusbserialfa140
monitor_speed = 74880
monitor_filters = time, colorize
lib_deps = paulstoffregen/OneWire@^2.3.5

```

**Source file to reproduce issue:**
n/a

### Additional info

ivankravets, 2020-10-23:
Could you provide a fix to https://github.com/platformio/platformio-core/blob/develop/platformio/commands/device/filters/time.py ?

hoegge, 2020-10-26:
Yes, I'll try to fix at latest Sunday. Not sure I can build and test, but can try to apply the logic on example monitoring output. Thanks for the file hint

ivankravets, 2020-10-30:
Thanks! Fixed in the latest PlatformIO Core Dev. The final release is planned this week.

commit 96b1a1c7 (2020-10-30): Fixed an issue with a "wrong" timestamp in device monitor output using `"time" filter` // Resolve #3712
closed 2020-10-30 by ivankravets