platformio/platformio-core #3717: Support for the Sipeed Maix Amigo

kallisti5, 2020-10-29:
What kind of issue is this?
**Feature Request**.
 

------------------------------------------------------------------

### Configuration

**Operating system**: Linux
**PlatformIO Version** (`platformio --version`):

### Description of problem

Seeed Studio is now shipping the Sipeed Maix Amigo devices. These are powered via the Kendryte K210 (like the other Maix devices)

### Platform information

* SoC – Kendryte K210 Dual-core 64-bit RISC-V (RV64GC) processor with FPU @ 400 MHz (overclockable to 500MHz), 8MB SRAM, built-in AI accelerators for video and audio
* Storage – 16MB Flash, MicroSD card slot up
* Display – 3.5-inch TFT capacitive touch screen display with 480×320 resolution
* Camera
    * VGA front-facing camera up to 30 fps (GC0328 sensor)
    * VGA rear camera up to 60 fps (OV7740 sensor)
* Audio – 6+1 microphone array
* USB – 1x USB Type-C port for power and programming
* Expansion
    * 3x Grove ports with I2C, GPIO, etc…
    * 3x 8-pin SP-MOD headers with GPIOs, 3.3V, and GND signals; compatible with ESP-01 ESP8266 WiFi module, PSRAM modules, etc…
* Sensor – Accelerometer
* Misc – Reset button, 3x function buttons, 3x LEDs
* Battery – 520 mAh Lithium battery
* Power Supply – Via USB-C port
* Dimensions – 104.3 x 63.3 x 16.5mm

kallisti5, 2020-10-29:
It looks like Sipeed has done "something" here for platformio.  Guessing it wasn't upstreamed
https://dl.sipeed.com/MAIX/platformio/dl-packages

ivankravets, 2020-10-29:
Please move to https://github.com/sipeed/platform-kendryte210/issues

closed 2020-10-29 by ivankravets