platformio/platformio-core #3724: platform_packages clone ruining git history

saloid, 2020-11-01:
What kind of issue is this?
- [*] **PlatformIO Core**.
------------------------------------------------------------------

### Configuration

**Operating system**: Win 10 x64

**PlatformIO Version**: PlatformIO, version 5.0.3a1

### Description of problem

When I am trying to use "staging" version of framework (https://github.com/stm32duino/Arduino_Core_STM32 in my case), whole cloned git repo consists only from single last commit, so all previous commits are lost during this cloning (maybe they are squashed?).

#### Steps to Reproduce

1. Create any project with `platform_packages = framework-arduinoststm32 @ https://github.com/stm32duino/Arduino_Core_STM32` in platformio.ini
2. Build project
3. This repo will be cloned to .platformio folder as `~/.platformio/packages/framework-arduinoststm32`
4. Run `git log --reflog` to view all commits in this repo. There will be only last one.

### Actual Results
```sh
Dmytro@this-pc MINGW64 ~/.platformio/packages/framework-arduinoststm32 (master)
$ git log --reflog
commit 1faf0a0ead7b6ae07258d706c1e9052de0be6cf0 (grafted, HEAD -> master, origin/master, origin/HEAD)
Author: Frederic Pillon <frederic.pillon@st.com>
Date:   Fri Oct 30 19:28:44 2020 +0100

    Merge pull request #1225 from fpistm/UpdateCubeF4L1

    Update STM32 F4 and L1 HAL drivers

```

### Expected Results
List of all commits, not only last one

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:genericSTM32F103RC]
platform = ststm32@8.1.0
framework = arduino
platform_packages = framework-arduinoststm32 @ https://github.com/stm32duino/Arduino_Core_STM32
  ; use upstream Git version
board = genericSTM32F103RC
board_build.variant = Generic_F103Rx
board_build.core = stm32
```

### Additional info
Correct handling of git history is very useful at framework development

saloid, 2020-11-01:
Also I can't clone it here manually - it being replaced at next build:
```
Executing task: C:\Users\Dmytro\.platformio\penv\Scripts\platformio.exe run <

Processing genericSTM32F103RC (platform: ststm32@8.1.0; framework: arduino; board: genericSTM32F103RC)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Tool Manager: Installing git+https://github.com/stm32duino/Arduino_Core_STM32
git version 2.27.0.windows.1
Cloning into 'C:\Users\Dmytro\.platformio\.cache\tmp\pkg-installing-j8sxfixj'...
remote: Enumerating objects: 4867, done.
remote: Counting objects: 100% (4867/4867), done.
remote: Compressing objects: 100% (2110/2110), done.
remote: Total 4867 (delta 3244), reused 3578 (delta 2654), pack-reused 0R
Receiving objects: 100% (4867/4867), 14.98 MiB | 7.52 MiB/s, done.
Resolving deltas: 100% (3244/3244), done.
Updating files: 100% (4517/4517), done.
Tool Manager: framework-arduinoststm32 @ 0.0.0+sha.1faf0a0 has been installed!
```

mcspr, 2020-11-01:
ref. https://github.com/platformio/platformio-core/blob/20a74d1654b239872ba3c13660217ce6477a7fc2/platformio/package/vcsclient.py#L170-L171
This causes `git` to do a 'shallow' clone, fetching bare minimum data from the repo source.

Note that you can 'unshallow' the local copy by running `git fetch --unshallow` from the `~/.platformio/package/framework-arduinoststm32/` (which needs to be done only once)

I wonder if it's worth mentioning in the documentation?

ivankravets, 2020-11-01:
Yes, this is a feature, not a bug. We don't clone packages "for development purpose", we use them as dependencies. If we clone full repository, it can take hours.

closed 2020-11-01 by ivankravets

saloid, 2020-11-01:
`git fetch --unshallow` solves the problem, thanks.