platformio/platformio-core #3729: LDF fails to find project dependencies for espressif32 unless espressif8266 is built first

mairas, 2020-11-09:
What kind of issue is this?

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:

Tested on Macos 10.15.6 and Ubuntu Linux 20.04

**PlatformIO Version** (`platformio --version`):

5.0.2

### Description of problem

LDF fails to resolve the dependencies for a project using SignalK/SensESP when using the espressif32 platform. If ~/.platformio and .pio are deleted and an espressif8266 platform build is performed first, the subsequent espressif32 builds succeed.

#### Steps to Reproduce

1. Clone the test repo at https://github.com/mairas/SensESP-buildtest.git
2. Install PlatformIO CLI: `pip3 install platformio`
3. Run `fail-test.sh` in the cloned repository. The script will demonstrate both expected build failures and successes.

### Actual Results

Builds fail depending on the build order.

### Expected Results

All builds should succeed.

### Additional info

This issue has pretty much crippled the new user experience on [SensESP](https://github.com/SignalK/SensESP) and I'd consider it at least major, if not critical, for us.

Jason2866, 2020-11-10:
It looks like your https://github.com/SignalK/SensESP/blob/master/library.json has dependencies which are not valid for a ESP32 build.
```
In file included from .pio/libdeps/esp32dev/SensESP/src/sensesp_app.h:15:0,
                 from src/main.cpp:3:
.pio/libdeps/esp32dev/SensESP/src/net/http.h:4:31: fatal error: ESPAsyncWebServer.h: No such file or directory

***************************************************************************
* Looking for ESPAsyncWebServer.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ESPAsyncWebServer.h"
* Web  > https://platformio.org/lib/search?query=header:ESPAsyncWebServer.h
*
***************************************************************************

compilation terminated.
Compiling .pio/build/esp32dev/libbc3/WiFi/ETH.cpp.o
```

mairas, 2020-11-10:
If you look at the `library.json` file you linked, all ESP8266 specific libraries are marked so with the `platforms` field. But that's not even the point. There should be absolutely no way builds against a specific platform should impact other platform builds. They maintain their dependencies independently and should have no such side effects.

mairas, 2020-11-10:
Or, well, that actually kinda is the point. :-) To me it seems PIO LDF seems to selectively ignore that field, causing the builds to fail, depending on the build order. This is a new issue in PlatformIO 5.

mairas, 2020-11-19:
The bug seems to be fixed in 5.0.3. The test script now fails at the first expected failure. Closing.

closed 2020-11-19 by mairas