platformio/platformio-core #3730: package pack/publish ignores some folders

maakbaas, 2020-11-10:
The documentation states: 

> Publish a package to the registry so that it can be installed by name. All files in the package directory are included if export field is not declared in a package manifest (for example, see export). The next items are automatically excluded:
> 
> ._*
> __*
> .DS_Store
> .git/
> .hg/
> .svn/
> .pio/
> **/.pio/

However, my library (https://github.com/maakbaas/esp8266-iot-framework) contains three folders not listed above `html/`, `docs/`, and `build/` that are not included in the tarball in PlatformIO 5.0.1. By default the export section in my library.json looks like this:

> "export": {
>         "exclude": [
>             "node_modules",
>             ".vscode"
>         ]
>     }

even if I update my library.json export section to what is shown below, or several other iterations, these folders keep getting ignored. 

> "export": {
>         "exclude": [
>             "node_modules",
>             ".vscode"
>         ],
>         "include": [
>             "docs/*",
>             "build/*",
>         ]
>     }

If I rename `build/` to `build2/`,  `docs/` to `docs2/` and  `html/` to `html2/`, these will be included in the tarball. Is there another hardcoded list of folder names that is not mentioned in the documentation that will be excluded?

maakbaas, 2020-11-10:
Ok, found my answer probably in the `EXCLUDE_DEFAULT` and `EXCLUDE_LIBRARY_EXTRA` lists in [pack.py](https://github.com/platformio/platformio-core/blob/develop/platformio/package/pack.py).

Should this behavior be made more clear in the documentation? In which PlatformIO version was this introduced?

From the folders above, the docs should indeed not be included, but in my library I have a few `extra_scripts` .py files that are now in a folder named `build/`. Is your recommendation to rename this folder, or is there some way in which I can include these without renaming the folder?

ivankravets, 2020-11-11:
Thanks! We will update docs.

Have you tried to include things that you need? See https://docs.platformio.org/en/latest/librarymanager/config.html#include

We try to keep libraries in small size and remove unused files in the build process.

maakbaas, 2020-11-12:
Yes, I tried including, but this did not work. Without fully understanding the code in `pack.py` it seems as if the excludes are applied after the includes, potentially overwriting the includes?

```python
result = ["+<%s>" % p for p in include or ["*", ".*"]]
result += ["-<%s>" % p for p in exclude or []]
result += ["-<%s>" % p for p in exclude_default]
```

ivankravets, 2020-11-12:
Thanks for the report! PlatformIO Core 5.0.3 is going to be released today.

closed 2020-11-12 by ivankravets

maakbaas, 2020-11-12:
Thanks for the quick fix!

ivankravets, 2020-11-12:
Did you test it? Please run “pio upgrade —dev”

ivankravets, 2020-11-12:
Double dash before dev

maakbaas, 2020-11-13:
Yes, I can confirm that it works. Thanks a lot!

commit 2817408d (2020-11-12): Fixed an issue when `pio package pack` ignores some folders // Resolve #3730