platformio/platformio-core #3735: The generated `.ccls` is not updated in builds

pvonmoradi, 2020-11-12:
The following command generates the requires files for LSP for plugins like coc-nvim:  
`pio project init --board my_board --ide vim --project-option "framework=arduino"`
Problem is, after mutating the `platformio.ini` and adding libraries, the `.ccls` file is not kept updated.  
How to make it so that all these auxiliary helper files get updated when project dependencies change? I was able to do it with  
`init`ing again.

ivankravets, 2020-11-13:
Duplicate of https://github.com/platformio/platformio-core/issues/364

pvonmoradi, 2020-11-13:
@ivankravets Thanks. So it is a known issue.  
Is generating the `compile_commands.json` via the post: script and compiledb command, a good alternative? (meaning pointing LSPs to compile_commands.json instead of .ccls)

ivankravets, 2020-11-14:
I updated the title. It is a very good idea to generate `compile_commands.json` for all compatible IDEs. Thanks!

pvonmoradi, 2020-11-14:
@ivankravets But isn't this already implemented? [https://docs.platformio.org/en/latest/integration/compile_commands.html](https://docs.platformio.org/en/latest/integration/compile_commands.html)
Or you mean you are planning the --ide switches to generate this in the project root?

ivankravets, 2020-11-14:
Yes, it makes sense to generate `compile_commands.json` in the root of project.

yunuscukran, 2021-03-14:
> Yes, it makes sense to generate `compile_commands.json` in the root of project.

From my experience, `compile_commands.json` being in the root is required to make `clangd` work properly. As far as ı can tell, this seems like a pre-req for the clangd integration at https://github.com/platformio/platformio-vscode-ide/issues/1802. 

Any timeline for this change?

ivankravets, 2021-03-14:
You can temporarily fix it on your side https://docs.platformio.org/en/latest/integration/compile_commands.html
Just override `COMPILATIONDB_PATH`.

ivankravets, 2022-04-09:
See updated docs https://docs.platformio.org/en/latest/integration/compile_commands.html

Key changes:

1) The latest PlatformIO Core generates a compilation database in the project root 
2) The new build environment was added to include toolchain paths. See docs for details.

Please confirm that it works now upgrading to the latest development version of PIO Core using `pio upgrade --dev`. Thanks!

closed 2022-04-09 by ivankravets

pvonmoradi, 2022-04-09:
@ivankravets confirmed that `pio run -t compiledb` creates a `compile_db.json` which is dependent on libraries included that project ini file. clang is able to jump to definition of library functions/methods.
Thanks.

ivankravets, 2022-04-09:
Happy coding with PlatformIO!

commit 541fcbf0 (2022-04-09): Added a new build variable (COMPILATIONDB_INCLUDE_TOOLCHAIN) to include toolchain paths in the compilation database // Resolve #3735