platformio/platformio-core #3738: Cannot get advanced scripting to work on Travis CI

thijstriemstra, 2020-11-15:
### Configuration

**Operating system**: Ubuntu 16

```
PACKAGES: 
 - framework-arduinoespressif32 3.10004.201016 (1.0.4) 
 - tool-esptoolpy 1.20600.0 (2.6.0) 
 - toolchain-xtensa32 2.50200.80 (5.2.0)
```

```
Updating platformio/tool-scons                4.40001.0 @ ~4.40001.0             [Up-to-date]

Platform Manager
================

Platform espressif32
--------

Updating platformio/espressif32               2.0.0                              [Up-to-date]

Updating platformio/toolchain-xtensa32        2.50200.80 @ ~2.50200.0            [Up-to-date]

Updating platformio/framework-arduinoespressif32 3.10004.201016 @ ~3.10004.191002   [Up-to-date]

Updating platformio/tool-esptoolpy            1.20600.0 @ ~1.20600.0             [Up-to-date]
```

### Description of problem

Cannot get advanced scripting to work on Travis CI.

#### Steps to Reproduce

Run Travis build.

### Actual Results

Error:

```
warning: Calling missing SConscript without error is deprecated.

Transition by adding must_exist=0 to SConscript calls.

Missing SConscript 'apply_patches.py'

File "/home/travis/virtualenv/python3.7.1/lib/python3.7/site-packages/platformio/builder/main.py", line 168, in <module>
```

### Expected Results

File is found. Locally it works as expected.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:test-env]
platform = espressif8266
board = d1_mini
framework = arduino
monitor_speed = 74880
extra_scripts = pre:apply_patches.py
```

**Source file to reproduce issue:**
```python
from os.path import join, isfile

Import("env")

SDFAT_DIR = join(env.get('PROJECT_LIBDEPS_DIR'), env.get('PIOENV'), 'SdFat', 'src')
SDFAT_CFG = join(SDFAT_DIR, 'SdFatConfig.h')
patchflag_path = join(SDFAT_DIR, ".patching-done")

# patch file only if we didn't do it before
if not isfile(patchflag_path):
    original_file = SDFAT_CFG
    patched_file = join(env.get('PROJECT_DIR'), 'patches', '1-SdFat-SPI_DRIVER_SELECT.patch')

    assert isfile(original_file) and isfile(patched_file)

    env.Execute("patch %s %s" % (original_file, patched_file))

    def _touch(path):
        with open(path, "w") as fp:
            fp.write("")

    env.Execute(lambda *args, **kwargs: _touch(patchflag_path))
```

ivankravets, 2020-11-16:
`pio ci` create TMP folder and runs selected files in it. Please use [pio run](https://docs.platformio.org/en/latest/core/userguide/cmd_run.html) directly for advanced things.

closed 2020-11-16 by ivankravets

thijstriemstra, 2020-11-23:
I also tested it with Github actions and getting the same error there:

```
warning: Calling missing SConscript without error is deprecated.
Transition by adding must_exist=0 to SConscript calls.
Missing SConscript 'apply_patches.py'
File "/opt/hostedtoolcache/Python/3.9.0/x64/lib/python3.9/site-packages/platformio/builder/main.py", line 168, in <module>
```

> pio ci create TMP folder and runs selected files in it. Please use pio run directly for advanced things.

I will give that a try, thanks.

thijstriemstra, 2020-11-23:
`pio run` shows a proper error:

```
Verbose mode can be enabled via `-v, --verbose` option
AssertionError: :
  File "/opt/hostedtoolcache/Python/3.9.0/x64/lib/python3.9/site-packages/platformio/builder/main.py", line 168:
    env.SConscript(item, exports="env")
  File "/home/runner/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Script/SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/home/runner/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Script/SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/home/runner/work/foo/bar/apply_patches.py", line 14:
    assert isfile(original_file) and isfile(patched_file)
```

Fixed it by changing from pre to post in platformio.ini:

```ini
[env:test-env]
platform = espressif8266
board = d1_mini
framework = arduino
monitor_speed = 74880
extra_scripts = post:apply_patches.py
```