platformio/platformio-core #3740: PIO Keeps installing a library into temp until filling up the drive.

HamzaHajeir, 2020-11-16:
**Operating system**: Windows 10 Pro Version 10.0.18362 Build 18362

**PlatformIO Version** (`platformio --version`): Core 5.0.3 - Home 3.3.1

### Description of problem

I found that pio keeps installing a custom library dependency inside a lot of my projects repeatedly until filling the C hard disk space.

the library is PangolinMQTT.
Referencing to it in platformio.ini : 
lib_deps =   
  https://github.com/philbowles/PangolinMQTT 

### Actual Results
file target : C:\Users\[UserName]\.platformio\.cache\tmp
screenshot of the folder : (All of them contains the same library with the same version .. etc)
[image]

What could the problem be ?

ivankravets, 2020-11-19:
Do you use external antivirus tools?

HamzaHajeir, 2020-11-19:
Yes, I use 3 different anti-virus software.

What's the responsible part would cause this ?

ivankravets, 2020-11-19:
We don't provide any support for this case. Antiviruses are the biggest evil now. Please read on the Internet what these tools do with your machine and data. I think it's good to start from Avast and the rest, such as Kaspersky, etc.

The only solution for this issue is to exclude `%HOME_DIR%/.platformio` from these bad tools.

HamzaHajeir, 2020-11-26:
Thanks for your input @ivankravets , I'm trying to have a good understanding what's going on. I've disabled all my 3 anti-virus (MalwareBytes, Avast, ESET Endpoint) and restarted VSCode. It continues to do the same thing. Every Rebuilding of Intellisense Index causes this behavior. 

Also I'm curious why it affects only this library *could be a coincidence though*.

Is there a resource to start with to understand what's happening ?

ivankravets, 2020-11-26:
It seems that there is an issue with PlatformIO and that library. @philbowles , an author of this library, recommends using Arduino IDE https://github.com/philbowles/pangolinmqtt#important-note-for-platformio-users . Please follow his guide

If you need an MQTT library for your project, please check these alternatives:
- https://platformio.org/lib/show/89/PubSubClient
- https://platformio.org/lib/show/617/MQTT
- https://platformio.org/lib/show/1092/Adafruit%20MQTT%20Library

HamzaHajeir, 2020-11-26:
His opinion about PIO is about building projects, I'm satisfied with PIO and made sure it's building perfectly.

The current issue applies with any library linked to github under lib_deps.

ivankravets, 2020-11-26:
Could you reproduce your issue when run `pio project init --ide vscode` from CLI https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli?

HamzaHajeir, 2020-11-28:
I've tried it, It successfully installed the library once.

I had to manually assign this to platformio.ini : 
```
[env:esp32doit-devkit-v1]
framework = arduino
monitor_speed = 115200
lib_deps = 
  https://github.com/philbowles/pangolinmqtt ;as an example

platform = espressif32
board = esp32doit-devkit-v1
```

I've then managed to remove a library from an existing project. Hit : Rebuild IntelliSense Index, I observed the following : 
PIO Tries to install it once. 
After it fails, It doesn't try again as before. And the tmp folder got a new folder inside it for that library.

I don't know what goes different at this time, I've also checked out PIO Core&Home versions. No difference.

ivankravets, 2020-11-28:
Something is going wrong with a system privacy policy or 3rd party tools that block software.

There is an alternative solution to this issue. Please manually clone/put the library to `/lib` folder. See the `README` file inside.

HamzaHajeir, 2020-11-28:
I think I've captured some useful info from Console : 

```
Could not fetch project targets for 'esp32doit-devkit-v1' environment => Error: Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\Hamza\.platformio\penv\lib\site-packages\platformio\project\helpers.py", line 155, in load_project_ide_data
    raise exception.PlatformioException(result.output)
platformio.exception.PlatformioException: Processing esp32doit-devkit-v1 (framework: arduino; platform: espressif32; board: esp32doit-devkit-v1)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 (2.0.0) > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 3.10004.200129 (1.0.4) 
 - tool-esptoolpy 1.20600.0 (2.6.0) 
 - toolchain-xtensa32 2.50200.80 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Library Manager: Installing git+https://github.com/philbowles/pangolinmqtt
git version 2.29.2.windows.2
Cloning into 'C:\Users\Hamza\.platformio\.cache\tmp\pkg-installing-shcwufcy'...
C:/Program Files/Git/mingw64/libexec/git-core\git-submodule: line 22: .: git-sh-setup: file not found
VCSBaseException: VCS: Could not process command ['git', 'clone', '--recursive', '--depth', '1', 'https://github.com/philbowles/pangolinmqtt', 'C:\\Users\\Hamza\\.platformio\\.cache\\tmp\\pkg-installing-shcwufcy']:
  File "C:\Users\Hamza\.platformio\penv\Lib\site-packages\platformio\builder\main.py", line 170:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\Hamza\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Hamza\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Hamza\.platformio\platforms\espressif32\builder\main.py", line 223:
    target_elf = env.BuildProgram()
  File "C:\Users\Hamza\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Hamza\.platformio\penv\Lib\site-packages\platformio\builder\tools\platformio.py", line 62:
    env.ProcessProjectDeps()
  File "C:\Users\Hamza\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Hamza\.platformio\penv\Lib\site-packages\platformio\builder\tools\platformio.py", line 140:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\Hamza\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Hamza\.platformio\penv\Lib\site-packages\platformio\builder\tools\piolib.py", line 1062:
    project.install_dependencies()
  File "C:\Users\Hamza\.platformio\penv\Lib\site-packages\platformio\builder\tools\piolib.py", line 898:
    lm.install(spec)
  File "C:\Users\Hamza\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 48:
    pkg = self._install(
  File "C:\Users\Hamza\.platformio\penv\lib\site-packages\platformio\package\manager\library.py", line 81:
    return super(LibraryPackageManager, self)._install(
  File "C:\Users\Hamza\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 97:
    pkg = self.install_from_url(spec.url, spec, silent=silent)
  File "C:\Users\Hamza\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 142:
    assert vcs.export()
  File "C:\Users\Hamza\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 176:
    assert self.run_cmd(args)
  File "C:\Users\Hamza\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 115:
    raise VCSBaseException("VCS: Could not process command %s" % e.cmd)
========================= [FAILED] Took 22.07 seconds =========================
```

HamzaHajeir, 2020-11-28:
-> Disabled 3 Anti-virus software ~ Pretty the same log : 

```
Error: Error: Processing esp32doit-devkit-v1 (framework: arduino; platform: espressif32; board: esp32doit-devkit-v1)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 (2.0.0) > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 3.10004.200129 (1.0.4) 
 - tool-esptoolpy 1.20600.0 (2.6.0) 
 - toolchain-xtensa32 2.50200.80 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Library Manager: Installing git+https://github.com/philbowles/pangolinmqtt
git version 2.29.2.windows.2
Cloning into 'C:\Users\Hamza\.platformio\.cache\tmp\pkg-installing-rkohh5t9'...
C:/Program Files/Git/mingw64/libexec/git-core\git-submodule: line 22: .: git-sh-setup: file not found
VCSBaseException: VCS: Could not process command ['git', 'clone', '--recursive', '--depth', '1', 'https://github.com/philbowles/pangolinmqtt', 'C:\\Users\\Hamza\\.platformio\\.cache\\tmp\\pkg-installing-rkohh5t9']:
  File "C:\Users\Hamza\.platformio\penv\Lib\site-packages\platformio\builder\main.py", line 170:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\Hamza\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Hamza\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Hamza\.platformio\platforms\espressif32\builder\main.py", line 223:
    target_elf = env.BuildProgram()
  File "C:\Users\Hamza\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Hamza\.platformio\penv\Lib\site-packages\platformio\builder\tools\platformio.py", line 62:
    env.ProcessProjectDeps()
  File "C:\Users\Hamza\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Hamza\.platformio\penv\Lib\site-packages\platformio\builder\tools\platformio.py", line 140:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\Hamza\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Hamza\.platformio\penv\Lib\site-packages\platformio\builder\tools\piolib.py", line 1062:
    project.install_dependencies()
  File "C:\Users\Hamza\.platformio\penv\Lib\site-packages\platformio\builder\tools\piolib.py", line 898:
    lm.install(spec)
  File "C:\Users\Hamza\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 48:
    pkg = self._install(
  File "C:\Users\Hamza\.platformio\penv\lib\site-packages\platformio\package\manager\library.py", line 81:
    return super(LibraryPackageManager, self)._install(
  File "C:\Users\Hamza\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 97:
    pkg = self.install_from_url(spec.url, spec, silent=silent)
  File "C:\Users\Hamza\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 142:
    assert vcs.export()
  File "C:\Users\Hamza\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 176:
    assert self.run_cmd(args)
  File "C:\Users\Hamza\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 115:
    raise VCSBaseException("VCS: Could not process command %s" % e.cmd)
========================= [FAILED] Took 11.35 seconds =========================

	at spawnOptions.cwd (c:\Users\Hamza\.vscode\extensions\platformio.platformio-ide-2.2.1\node_modules\platformio-node-helpers\dist\index.js:1:884049)	at ChildProcess.c (c:\Users\Hamza\.vscode\extensions\platformio.platformio-ide-2.2.1\node_modules\platformio-node-helpers\dist\index.js:1:879169)	at ChildProcess.emit (events.js:223:5)	at ChildProcess.cp.emit (c:\Users\Hamza\.vscode\extensions\platformio.platformio-ide-2.2.1\node_modules\cross-spawn\lib\enoent.js:34:29)	at maybeClose (internal/child_process.js:1021:16)	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
t.log @ console.ts:140
$logExtensionHostMessage @ mainThreadConsole.ts:29
_doInvokeHandler @ rpcProtocol.ts:365
_invokeHandler @ rpcProtocol.ts:365
_receiveRequest @ rpcProtocol.ts:306
_receiveOneMessage @ rpcProtocol.ts:174
(anonymous) @ rpcProtocol.ts:105
fire @ event.ts:584
fire @ ipc.net.ts:431
_receiveMessage @ ipc.net.ts:764
(anonymous) @ ipc.net.ts:624
fire @ event.ts:584
[... 43 lines cut ...]
  File "C:\Users\Hamza\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Hamza\.platformio\penv\Lib\site-packages\platformio\builder\tools\platformio.py", line 140:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\Hamza\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
[... cut ...]
  File "C:\Users\Hamza\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 142:
    assert vcs.export()
  File "C:\Users\Hamza\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 176:
    assert self.run_cmd(args)
  File "C:\Users\Hamza\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 115:
    raise VCSBaseException("VCS: Could not process command %s" % e.cmd)
========================= [FAILED] Took 20.52 seconds =========================

```

ivankravets, 2020-11-28:
Could you try this solution https://stackoverflow.com/questions/49256190/how-to-fix-git-sh-setup-file-not-found-in-windows/50833818?

ivankravets, 2020-11-28:
It seems that I could reproduce this issue. Sorry for the wrong advice. I'll investigate it.

HamzaHajeir, 2020-11-29:
No Problem, I'm happy to find bugs. 

And as @philbowles calls me : A Bugger ! xD

> Could you try this solution https://stackoverflow.com/questions/49256190/how-to-fix-git-sh-setup-file-not-found-in-windows/50833818?

Yes, I've tried it and it successfully installed the library!

I think it's solved for me, Waiting what happens at your end, And I'll update news if any.

HamzaHajeir, 2020-11-29:
I found this ERR at Console, I don't know whether it's related or helpful. And I don't encounter any issue using PIO.
Also, It happens at start (Just before or at start of Rebuilding Intellisense Index) : 

```
  ERR Bad progress location: platformio-ide.projectTasks: Error: Bad progress location: platformio-ide.projectTasks
    at E.withProgress (file:///C:/Users/Hamza/AppData/Local/Programs/Microsoft VS Code/resources/app/out/vs/workbench/workbench.desktop.main.js:6315:719)
    at h.$startProgress (file:///C:/Users/Hamza/AppData/Local/Programs/Microsoft VS Code/resources/app/out/vs/workbench/workbench.desktop.main.js:4428:402)
    at p._doInvokeHandler (file:///C:/Users/Hamza/AppData/Local/Programs/Microsoft VS Code/resources/app/out/vs/workbench/workbench.desktop.main.js:4475:707)
    at p._invokeHandler (file:///C:/Users/Hamza/AppData/Local/Programs/Microsoft VS Code/resources/app/out/vs/workbench/workbench.desktop.main.js:4475:399)
    at p._receiveRequest (file:///C:/Users/Hamza/AppData/Local/Programs/Microsoft VS Code/resources/app/out/vs/workbench/workbench.desktop.main.js:4474:59)
    at p._receiveOneMessage (file:///C:/Users/Hamza/AppData/Local/Programs/Microsoft VS Code/resources/app/out/vs/workbench/workbench.desktop.main.js:4472:886)
    at file:///C:/Users/Hamza/AppData/Local/Programs/Microsoft VS Code/resources/app/out/vs/workbench/workbench.desktop.main.js:4471:34
    at c.fire (file:///C:/Users/Hamza/AppData/Local/Programs/Microsoft VS Code/resources/app/out/vs/workbench/workbench.desktop.main.js:99:67)
    at fire (file:///C:/Users/Hamza/AppData/Local/Programs/Microsoft VS Code/resources/app/out/vs/workbench/workbench.desktop.main.js:476:523)
    at t.PersistentProtocol._receiveMessage (file:///C:/Users/Hamza/AppData/Local/Programs/Microsoft VS Code/resources/app/out/vs/workbench/workbench.desktop.main.js:481:629)
    at file:///C:/Users/Hamza/AppData/Local/Programs/Microsoft VS Code/resources/app/out/vs/workbench/workbench.desktop.main.js:478:635
    at c.fire (file:///C:/Users/Hamza/AppData/Local/Programs/Microsoft VS Code/resources/app/out/vs/workbench/workbench.desktop.main.js:99:67)
    at g.acceptChunk (file:///C:/Users/Hamza/AppData/Local/Programs/Microsoft VS Code/resources/app/out/vs/workbench/workbench.desktop.main.js:473:978)
    at file:///C:/Users/Hamza/AppData/Local/Programs/Microsoft VS Code/resources/app/out/vs/workbench/workbench.desktop.main.js:473:330
    at Socket.t (file:///C:/Users/Hamza/AppData/Local/Programs/Microsoft VS Code/resources/app/out/vs/workbench/workbench.desktop.main.js:483:276)
    at Socket.emit (events.js:223:5)
    at addChunk (_stream_readable.js:309:12)
    at readableAddChunk (_stream_readable.js:290:11)
    at Socket.Readable.push (_stream_readable.js:224:10)
    at Pipe.onStreamRead (internal/stream_base_commons.js:181:23)
```

ivankravets, 2020-11-30:
Please open https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli and type `pio upgrade --dev`. Restart VSCode.

Does it work now?

closed 2020-11-30 by ivankravets

HamzaHajeir, 2020-11-30:
> Please open https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli and type `pio upgrade --dev`. Restart VSCode.
> 
> Does it work now?

Ok, I've revert back and removed solution given here : 

> Could you try this solution https://stackoverflow.com/questions/49256190/how-to-fix-git-sh-setup-file-not-found-in-windows/50833818?

Then I removed couple of libraries from .pio/libdeps.
Hit Rebuild IntelliSense Index 

Libraries are successfully installed!

Also, `.platformio/.cache/tmp` folder is clean (for the installed libraries at this time)

ivankravets, 2020-11-30:
Thanks that reported this issue!

Happy coding with PlatformIO! ;)

HamzaHajeir, 2020-12-01:
I'm happy to contribute with bug-finding :) 

Thanks for solving it ;)

commit c38b9a41 (2020-11-30): Fixed a "git-sh-setup: file not found" error when installing project dependencies from Git VCS // Resolve #3740