platformio/platformio-core #3741: LDF only downloading some of the dependencies listed in library manifest

mbarneytu, 2020-11-18:
### Description
My project has 1 library specified in lib_deps, and that library has another 20+ dependencies listed in its library.json manifest. However, only 3 of those dependencies are being automatically downloaded by Library Dependency Finder.
### To Reproduce

1. Create a new project from PIO Home->New Project, and select the board EnviroDIY Mayfly.
2. Replace the contents of platformio.ini with the following, and save it:
`[env:mayfly]`
`platform = atmelavr`
`board = mayfly`
`framework = arduino`
`lib_deps = EnviroDIY_ModularSensors@0.25.1`
`lib_ldf_mode = deep+`
`lib_ignore = RTCZero`
`build_flags = -DSDI12_EXTERNAL_PCINT`
3. Click Build.
### Results
- When you save the ini file, LDF installs only the following into .pio\libdeps\mayfly:
EnviroDIY_ModularSensors and 3 of its dependencies. They happen to be the first 3 dependencies listed in that library's [manifest.](https://github.com/EnviroDIY/ModularSensors/blob/master/library.json)
- When you build, it fails with the error:
`In file included from .pio\libdeps\mayfly\EnviroDIY_ModularSensors\src\modems\DigiXBee3GBypass.cpp:11:0:
.pio\libdeps\mayfly\EnviroDIY_ModularSensors\src\modems\DigiXBee3GBypass.h:37:27: fatal error: TinyGsmClient.h: No such file or directory`
### Expected Behavior
- LDF installs EnviroDIY_ModularSensors as well as all 23 of its dependencies.
- Build succeeds.
### Notes
I can avoid the problem by downgrading the IDE as follows:
1. Downgrade the PlatformIO IDE extension to v2.1.0 or earlier.
2. Close VSCode.
3. Delete the .pio folder.
4. Reopen the project in VSCode.

Then, LDF downloads all dependencies as expected. Any newer version of the extension, including 2.2.0, produces the condition described above, under Results.
### My Environment
PlatformIO Core 5.0.2
PlatformIO Home 3.3.1
PlatformIO IDE extension 2.2.0
VSCode 1.51.0
Windows 10 Pro 64-bit

pfeerick, 2020-12-13:
I have just tried this on Linux, using PIO Core 5.0.3 & VSCode extension 2.2.1 (PIO Home 3.3.1, VSCode 1.51.1), but can't reproduce those results.

```pio init -b mayfly -O "lib_deps = EnviroDIY_ModularSensors@0.25.1" -O "lib_ldf_mode = deep+" -O "lib_ignore = RTCZero" -O "build_flags = -DSDI12_EXTERNAL_PCINT"```

Waited for the intellisense process to finish rebuilding (i.e. downloading the libraries), and then grabbed the simple_sensor example from the library, and used it as the project `main.cpp`. 

[image]

I'll try again later on Windows, but perhaps it was a library internet or other glitch which has been resolved, or is a nasty OS specific issue?

mbarneytu, 2020-12-14:
Interesting- thank you. I tried the pio init command you provided and got the following warning:
`Obsolete PIO Core v4.3.4 is used (previous was 5.0.3)
Please remove multiple PIO Cores from a system:
https://docs.platformio.org/page/faq.html#multiple-pio-cores-in-a-system
`

Not sure how the old PIO Core got there, but I uninstalled it following the FAQ. Then re-tried the pio init, opened VSCode and waited for PIO Rebuild Intellisense Index, but still only get 4 libraries installed by VSCode: 
[image]

My environment:
Windows 10 Pro 64-bit
VSCode 1.52.0
PlatformIO IDE 2.2.1
PIO Core 5.0.3
PIO Home 3.3.1

pfeerick, 2020-12-15:
The `pio init` bit isn't anything magical or different to using the `new project` wizard - it was just a command line and thus more easily reproducible way to init the project. For both others, and myself ;) 

I've just tried again on Windows, and I get the exact same issue... 

I then tried a verbose build (`Advanced -> Verbose Build` in the Project Tasks sidebar panel) and that mentions

```
Found 9 compatible libraries
Scanning dependencies...
Warning: Ignored `Adafruit ADS1X15` dependency for `EnviroDIY_ModularSensors` library
Warning: Ignored `Adafruit AM2315` dependency for `EnviroDIY_ModularSensors` library
Warning: Ignored `Adafruit BME280 Library` dependency for `EnviroDIY_ModularSensors` library
Warning: Ignored `DHT sensor library` dependency for `EnviroDIY_ModularSensors` library
Warning: Ignored `Adafruit INA219` dependency for `EnviroDIY_ModularSensors` library
Warning: Ignored `Adafruit MPL115A2` dependency for `EnviroDIY_ModularSensors` library
Warning: Ignored `Adafruit Unified Sensor` dependency for `EnviroDIY_ModularSensors` library
Warning: Ignored `DallasTemperature` dependency for `EnviroDIY_ModularSensors` library
Warning: Ignored `OneWire` dependency for `EnviroDIY_ModularSensors` library
Warning: Ignored `SDI-12` dependency for `EnviroDIY_ModularSensors` library
Warning: Ignored `KellerModbus` dependency for `EnviroDIY_ModularSensors` library
Warning: Ignored `MS5803` dependency for `EnviroDIY_ModularSensors` library
Warning: Ignored `PubSubClient` dependency for `EnviroDIY_ModularSensors` library
Warning: Ignored `RTCZero` dependency for `EnviroDIY_ModularSensors` library
Warning: Ignored `SensorModbusMaster` dependency for `EnviroDIY_ModularSensors` library
Warning: Ignored `TinyGSM` dependency for `EnviroDIY_ModularSensors` library
Warning: Ignored `YosemitechModbus` dependency for `EnviroDIY_ModularSensors` library
Dependency Graph
|-- <EnviroDIY_ModularSensors> 0.25.1 (E:\Temp\PlatformIO\.pio\libdeps\mayfly\EnviroDIY_ModularSensors)
|   |-- <EnviroDIY_DS3231> 1.3.2 (E:\Temp\PlatformIO\.pio\libdeps\mayfly\EnviroDIY_DS3231)
|   |   |-- <Wire> 1.0 (C:\Users\Peter\.platformio\packages\framework-arduino-avr\libraries\Wire)
|   |-- <EnableInterrupt> 1.1.0 (E:\Temp\PlatformIO\.pio\libdeps\mayfly\EnableInterrupt)
|   |-- <SdFat> 1.1.2 (E:\Temp\PlatformIO\.pio\libdeps\mayfly\SdFat)
|   |   |-- <SPI> 1.0 (C:\Users\Peter\.platformio\packages\framework-arduino-avr\libraries\SPI)
|   |-- <SPI> 1.0 (C:\Users\Peter\.platformio\packages\framework-arduino-avr\libraries\SPI)
|   |-- <Wire> 1.0 (C:\Users\Peter\.platformio\packages\framework-arduino-avr\libraries\Wire)
|-- <EnableInterrupt> 1.1.0 (E:\Temp\PlatformIO\.pio\libdeps\mayfly\EnableInterrupt)
Building in release mode
```

Looking at the `library.json` configuration file for `EnviroDIY_ModularSensors` since it is the one that the dependencies are being ignored from, it looks like they made some mistakes in the `ADS1X15` dependency (`version_note` instead of `version`, `version` instead of `url`, etc, etc). but I don't know if that was enough to break things, or why it would only fail on Windows, not on Linux.

However, it looks like it was fixed in either 0.26 or 0.27, so if you can, just move up to to the newer version and forget about it, or you could temporarily upgrade, let PlatformIO finish installing dependencies, and then go back to 0.25.1 if you need to.

SRGDamia1, 2020-12-15:
The invalid library fields are there intentionally, as extra notes for a curious person reading the file.  I added "version_note" in cases where the version filed was linked to a specific branch or commit, but I wanted someone to be able to see the version number.  In the same way "url" is just a link for a person to look at and "library id", "note", etc are all just decoration.  

In skimming the code for the LibraryJsonManifestParser, it looks like all the expected fields of the manifest are pulled out by the expected key names so as long as those are there and valid, anything left over should be ignored.

Between version 0.25 and now I added the "owner" field to all of the dependencies.  I believe that field is new to pio version 5.

valeros, 2020-12-21:
Thanks for reporting. It looks like this issue has similar symptoms to #3740, so it might be already fixed in the dev branch. Would be great if someone could try to upgrade `platformio-core` to the latest by running the `pio upgrade --dev` command directly in the IDE terminal and try to initialize a new project from scratch. Thanks!

ivankravets, 2020-12-23:
Please re-test with the latest dev-version.

closed 2020-12-23 by ivankravets

SRGDamia1, 2020-12-23:
I'm sorry, I haven't gotten a chance to test again with the dev-version!  I'll try to soon.

mbarneytu, 2020-12-23:
Success! I just tested, after upgrading to the latest dev version (5.0.4b1), all dependencies were properly downloaded for EnviroDIY_ModularSensors@0.25.1, and build succeeds. Thank you all!