platformio/platformio-core #3743: platformio error on pipenv

friedrichaugust, 2020-11-22:
I installed esphome within a pipenv for python 3.8 on Ubuntu 20.10. Before I create the new pipenv I remove `platformio` and `esphome` with pip

When I run `esphome fensterbrattl.yaml compile` I get this error. What is goning wrong?

```
esphome fensterbrattl.yaml compile
INFO Reading configuration xxx.yaml...
INFO Detected timezone 'CET' with UTC offset 1 and daylight savings time from 28 March 02:00:00 to 31 October 03:00:00                                                                                                              
INFO Detected timezone 'CET' with UTC offset 1 and daylight savings time from 28 March 02:00:00 to 31 October 03:00:00                                                                                                              
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d fensterbrattl
******************************************************************************************************************
Obsolete PIO Core v4.3.4 is used (previous was 5.0.3)
Please remove multiple PIO Cores from a system:
https://docs.platformio.org/page/faq.html#multiple-pio-cores-in-a-system
******************************************************************************************************************
Processing fensterbrattl (board: esp32dev; framework: arduino; platform: espressif32@1.12.4)
------------------------------------------------------------------------------------------------------------------
Error: Traceback (most recent call last):
  File "/home/horst/.local/share/virtualenvs/code-vyA6J7rP/lib/python3.8/site-packages/platformio/__main__.py", line 109, in main                                                                                                   
    cli()  # pylint: disable=no-value-for-parameter                                                               
  File "/home/horst/.local/share/virtualenvs/code-vyA6J7rP/lib/python3.8/site-packages/click/core.py", line 829, in __call__                                                                                                        
    return self.main(*args, **kwargs)                                                                             
  File "/home/horst/.local/share/virtualenvs/code-vyA6J7rP/lib/python3.8/site-packages/click/core.py", line 782, in main                                                                                                            
    rv = self.invoke(ctx)                                                                                         
  File "/home/horst/.local/share/virtualenvs/code-vyA6J7rP/lib/python3.8/site-packages/platformio/commands/__init__.py", line 44, in invoke                                                                                         
    return super(PlatformioCLI, self).invoke(ctx)                                                                 
  File "/home/horst/.local/share/virtualenvs/code-vyA6J7rP/lib/python3.8/site-packages/click/core.py", line 1259, in invoke                                                                                                         
    return _process_result(sub_ctx.command.invoke(sub_ctx))                                                       
  File "/home/horst/.local/share/virtualenvs/code-vyA6J7rP/lib/python3.8/site-packages/click/core.py", line 1066, in invoke                                                                                                         
    return ctx.invoke(self.callback, **ctx.params)                                                                
  File "/home/horst/.local/share/virtualenvs/code-vyA6J7rP/lib/python3.8/site-packages/click/core.py", line 610, in invoke                                                                                                          
    return callback(*args, **kwargs)                                                                              
  File "/home/horst/.local/share/virtualenvs/code-vyA6J7rP/lib/python3.8/site-packages/click/decorators.py", line 21, in new_func                                                                                                   
    return f(get_current_context(), *args, **kwargs)                                                              
  File "/home/horst/.local/share/virtualenvs/code-vyA6J7rP/lib/python3.8/site-packages/platformio/commands/run/command.py", line 128, in cli                                                                                        
    process_env(                                                                                                  
  File "/home/horst/.local/share/virtualenvs/code-vyA6J7rP/lib/python3.8/site-packages/platformio/commands/run/command.py", line 170, in process_env                                                                                
    result = {"env": name, "duration": time(), "succeeded": ep.process()}                                         
  File "/home/horst/.local/share/virtualenvs/code-vyA6J7rP/lib/python3.8/site-packages/platformio/commands/run/processor.py", line 79, in process                                                                                   
    result = p.run(build_vars, build_targets, self.silent, self.verbose, self.jobs)                               
  File "/home/horst/.local/share/virtualenvs/code-vyA6J7rP/lib/python3.8/site-packages/platformio/managers/platform.py", line 414, in run                                                                                           
    self.install_packages(silent=True)                                                                            
  File "/home/horst/.local/share/virtualenvs/code-vyA6J7rP/lib/python3.8/site-packages/platformio/managers/platform.py", line 307, in install_packages                                                                              
    self.pm.install(name, version, silent=silent, force=force)                                                    
  File "/home/horst/.local/share/virtualenvs/code-vyA6J7rP/lib/python3.8/site-packages/platformio/managers/package.py", line 672, in install                                                                                        
    package_dir = self.get_package_dir(name, requirements, url)                                                   
  File "/home/horst/.local/share/virtualenvs/code-vyA6J7rP/lib/python3.8/site-packages/platformio/managers/package.py", line 415, in get_package_dir                                                                                
    manifest = self.get_package(name, requirements, url)                                                          
  File "/home/horst/.local/share/virtualenvs/code-vyA6J7rP/lib/python3.8/site-packages/platformio/managers/package.py", line 384, in get_package                                                                                    
    for manifest in self.get_installed():                                                                         
  File "/home/horst/.local/share/virtualenvs/code-vyA6J7rP/lib/python3.8/site-packages/platformio/managers/package.py", line 374, in get_installed                                                                                  
    manifest = self.load_manifest(pkg_dir)
  File "/home/horst/.local/share/virtualenvs/code-vyA6J7rP/lib/python3.8/site-packages/platformio/managers/package.py", line 347, in load_manifest
    manifest = ManifestParserFactory.new_from_file(manifest_path).as_dict()
  File "/home/horst/.local/share/virtualenvs/code-vyA6J7rP/lib/python3.8/site-packages/platformio/package/manifest/parser.py", line 73, in new_from_file
    return ManifestParserFactory.new(
  File "/home/horst/.local/share/virtualenvs/code-vyA6J7rP/lib/python3.8/site-packages/platformio/package/manifest/parser.py", line 123, in new
    return cls(contents, remote_url, package_dir)
  File "/home/horst/.local/share/virtualenvs/code-vyA6J7rP/lib/python3.8/site-packages/platformio/package/manifest/parser.py", line 136, in __init__
    self._data = self.normalize_repository(self._data)
  File "/home/horst/.local/share/virtualenvs/code-vyA6J7rP/lib/python3.8/site-packages/platformio/package/manifest/parser.py", line 174, in normalize_repository
    url = (data.get("repository") or {}).get("url")
AttributeError: 'str' object has no attribute 'get'

============================================================

An unexpected error occurred. Further steps:

[... cut ...]
  - pyserial [required: ==3.4, installed: 3.4]
  - pytz [required: ==2020.1, installed: 2020.1]
  - PyYAML [required: ==5.3.1, installed: 5.3.1]
  - tornado [required: ==6.0.4, installed: 6.0.4]
  - tzlocal [required: ==2.1, installed: 2.1]
    - pytz [required: Any, installed: 2020.1]
  - voluptuous [required: ==0.11.7, installed: 0.11.7]
html5lib==1.0.1
ipaddr==2.2.0
lockfile==0.12.2
msgpack==0.6.2
pep517==0.8.2
pkg-resources==0.0.0
progress==1.5
pyparsing==2.4.6
pytoml==0.1.21
retrying==1.3.3
webencodings==0.5.1
```

friedrichaugust, 2020-11-22:
I played around with the platformio versions. Using platformio = "==5.0.3" will work. Are the versions correct?

ivankravets, 2020-11-22:
> Obsolete PIO Core v4.3.4 is used 

Please upgrade to the latest PlatformIO Core 5.0

closed 2020-11-22 by ivankravets

melyux, 2020-12-16:
That's not going to work, ESPhome uses the pinned PlatformIO Core 4.3.4 version. What is the reason behind the error?