platformio/platformio-core #3744: Clion: PlatformIO utility is not found but installed

danielroehrig, 2020-11-22:
### Configuration

**Operating system**: Linux 5.9.8-2-MANJARO,  Manjaro

**PlatformIO Version** (`platformio --version`): 5.0.3

### Description of problem
[image]
PlatformIO seems to be installed and in path and i can execute pio commands from any terminal, but clion refuses to recognize it.

#### Steps to Reproduce

1. Install platform.io as described [here](https://docs.platformio.org/en/latest/core/installation.html#super-quick-mac-linux)
2. Update path
3. Try executing pio and see that it works
4. Try creating a new PlatformIO project

### Actual Results
Get the error message: PlatformIO utility is not found but installed

### Expected Results
Set up new project

ivankravets, 2020-11-23:
It seems you run CLion from another shell (system) where PATH is not updated. Could you try to run CLion from a Terminal where you have access to `pio --version`?

danielroehrig, 2020-11-23:
You were right. I needed to add the PlatformIO path to my .profile. Thanks for your help.

wirwolf, 2022-02-12:
Hello, @ivankravets. I have the same problem but I already added the path for the bin folder in PATH variable.
```
export PATH="$PATH:/home/wir_wolf/.platformio/penv/bin"

```
[image]
[image]
[image]

wirwolf, 2022-02-14:
If Path to PlatformIO Core CLI  input is empty i can not create a new project using  PlatformIO. On the platform, I see an empty area.
[image]

ivankravets, 2022-02-14:
@elmot what is the mechanism for checking if "PlatformIO utility is not found"? Thanks.

UtechtDustin, 2022-09-25:
I run into the same issue (Ubuntu), is related to https://youtrack.jetbrains.com/issue/CPP-29187/Clion-and-PlatformIO-using-defaults-does-not-find-the-PIO-installation.
Clion don't use the PATH Variable from .bashrc, as suggested in the issue i tried to start clion from cli but i don't fixed the issue for me.

smarteist, 2023-02-04:
> I run into the same issue (Ubuntu), is related to https://youtrack.jetbrains.com/issue/CPP-29187/Clion-and-PlatformIO-using-defaults-does-not-find-the-PIO-installation. Clion don't use the PATH Variable from .bashrc, as suggested in the issue i tried to start clion from cli but i don't fixed the issue for me.
> 
> The following workaround works fine for me (create symlinks instead of appending to the PATH)
> 
> ```
> ln -s ~/.platformio/penv/bin/platformio /home/<username>/.local/bin/platformio
> ln -s ~/.platformio/penv/bin/pio /home/<username>/.local/bin/pio
> ln -s ~/.platformio/penv/bin/piodebuggdb /home/<username>/.local/bin/piodebuggdb
> ```
It does the job:
```bash
mkdir -p $HOME/.local/bin/ && ln -s $HOME/.platformio/penv/bin/platformio $HOME/.local/bin/platformio && ln -s $HOME/.platformio/penv/bin/pio $HOME/.local/bin/pio && ln -s $HOME/.platformio/penv/bin/piodebuggdb $HOME/.local/bin/piodebuggdb
```

EasyG0ing1, 2023-07-28:
Here's the complete steps to the workaround:

Make this folder
`~/.local/bin`

Run these commands:
```BASH
ln -s ~/.platformio/penv/bin/platformio ~/.local/bin/platformio
ln -s ~/.platformio/penv/bin/pio ~/.local/bin/pio
ln -s ~/.platformio/penv/bin/piodebuggdb ~/.local/bin/piodebuggdb
```

Open CLion and in the PlatformIO settings, DELETE ALL THE TEXT in the field asking for the path.

Apply / OK

Open a new PlatformIO project ... TA DA!

cgursche, 2023-11-13:
In CLion 2023.2.2, opened a new project, selected platformIO, selected "settings" in "PlatformIO not found" dialog, erased whatever location I had put in earlier, used the folder button to navigate to /usr/local/bin where the alias to platformIO resides (used shift-command-. to reveal hidden folders), and that's when CLion took over and said "I got this". Now it works.