platformio/platformio-core #3745: Windows: Set "intelliSenseMode" property in c_cpp_properties.json file to "gcc-arm"

JeroenIoT, 2020-11-23:
PlatformIO should set "intelliSenseMode" property in c_cpp_properties.json file to "gcc-arm".

Since PlatformIO uses gcc-arm, the Intellisense should use gcc-arm as well. During startup the IntelliSense mode is changed and following message is displayed:

`[11/23/2020, 8:41:51 AM] For C++ source files, IntelliSenseMode was changed from “gcc-x64” to “gcc-arm” based on compiler args and probing compilerPath: “C:/Users/itsme/.platformio/packages/toolchain-gccarmnoneeabi/bin/arm-none-eabi-gcc.exe”
`

Changing the c_cpp_properties.json manually to "gcc-arm" also work temporary as it is auto generated and platformio overwrites and set it back to "gcc-x64".

ivankravets, 2020-11-23:
Should be fixed in the latest development version. Please open [PlatformIO Core CLI in VSCode]( https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli) and type `pio upgrade --dev`. Does it work now?

JeroenIoT, 2020-11-24:
Upgraded with the development version and it works now. Thanks. CLOSED

AgainPsychoX, 2022-08-20:
@ivankravets / @JeroenIoT  

It seems some regression happened. `intelliSenseMode` is not in the `c_cpp_properties.json`, causing problems, like pointers being hinted as 8 bytes long when in fact they are 4. 

This compiles without any issue (ESP8266), while VS Code IntelliSense tells `static assertion failed`...
```cpp
		static_assert(sizeof(void*) == 4);
		static_assert(sizeof(int*) == 4);
		static_assert(sizeof(const char*) == 4);
```

I can manually modify the `c_cpp_properties.json` and it helps until compiling/changing `platform.ini`, but it updates back itself then, discarding my change.

Can we have issue reopen/re-fixed?

AgainPsychoX, 2022-08-20:
I also tried `pio upgrade --dev` of course, but sadly it doesn't help.

AgainPsychoX, 2022-08-20:
@ivankravets Thank you for reopen. I might fix it myself by pull request, just have few questions and thoughts on the matter.

IMO there should be a way to provide their own base `c_cpp_properties.json` file, but I see that would be a lot of work and edge cases to take care of... However, I guess allowing each platform (like espressif8266) to choose the closest match for `intelliSenseMode` (in my case: gcc-arm or any x86?).

I can see platforms have `platform.json` and `platform.py`. Would be that fine to add the info there? I don't understand whole PlatformIO architecture design yet (libraries (+sdk?) <-> projects <-> platfroms (+sdk?) <-> board <-> tool chain). I imagine some tool chains work with both 32 and 64 bit, so I would be against deducing the mode from tool chain alone. 

My best guess (which might be lacking some assumptions) would be to assume platform has one mode set. And if there were boards/processor with multiple (is there any like this?), it would be listed as other platform, or if the time come the problem would be addressed again.

Also, I don't quite understand which what differs between `arm` and `x86` of `intelliSenseMode` options for C/C++ in VS Code. _(Actually asked at https://github.com/microsoft/vscode-cpptools/discussions/9764)_. Of course I know more or less that ARM (incl. RISC, like xtensa used by ESP8266) and x86 are different architectures, so some other defines are used, I could imagine some architecture specific code would be enabled in libs. 

Other consideration is that having IDE specific (like VS Code specific) property attached to list of rather IDE-independent properties would be inappropriate. While I don't have any solid ideas around how to deal with that small issue, I would love to hear someone else opinion.

ivankravets, 2022-08-21:
Indeed, we should contact the Microsoft team. We pass them the full path to the compiler and they fetch extra data in runtime.

ivankravets, 2022-08-21:
Duplicate of https://github.com/microsoft/vscode-cpptools/issues/9765

AgainPsychoX, 2022-08-21:
So, while we wait for Microsoft fix the issue in their closed-source `cpptools.exe` language server, current workaround would be to add paste
```json
,
            "intelliSenseMode": "gcc-arm"
```
at one of last lines of generated `c_cpp_properties.json` file. Every single time you touch `platformio.ini`, or compile/upload your project... 😕

maxgerhardt, 2022-08-28:
Now that it's logged on the Microsoft side they declared it a feature request instead of a bug and referr to 3-year old other feature request issues. Guess this one may take a few years to resolve.

Can someone prove it's a regerssion with a specific version?

ivankravets, 2022-08-29:
Do we experience this issue with ESP32 as well, or just with ESP8266?

maxgerhardt, 2022-08-29:
`board = esp32dev` works, the C/C++ logs show it fallbacks to `intelliSenseMode: windows-gcc-x86` mode so the pointer sizes happen to be the same.

ivankravets, 2022-08-29:
@maxgerhardt, does it mean that we should force VSCode to the `"intelliSenseMode": "gcc-arm"` for ESP products?

AgainPsychoX, 2023-02-14:
Any update on that?

AgainPsychoX, 2024-08-25:
Today I noticed that even the workaround of adding `"intelliSenseMode": "gcc-arm"` manually each time stopped working.

AgainPsychoX, 2024-08-25:
I made myself extra script to automate applying of the workaround:
```python
import sys
import subprocess
import json
from time import sleep
import inspect

class JSONWithCommentsDecoder(json.JSONDecoder):
	def __init__(self, **kw):
		super().__init__(**kw)

	def decode(self, s: str):
		s = '\n'.join(l if not l.lstrip().startswith('//') else '' for l in s.split('\n'))
		return super().decode(s)

try:
	# PlatformIO/SCons injected stuff
	Import('env' ,'projenv') # type: ignore
	env, projenv, DefaultEnvironment = env, projenv, DefaultEnvironment # type: ignore
except NameError:
	# Running detached
	sleep(1)
	
	with open('.vscode/c_cpp_properties.json', 'r') as f:
		c_cpp_properties = json.load(f, cls=JSONWithCommentsDecoder)

	def check_if_configuration_exists(configuration):
		for existing_configuration in c_cpp_properties['configurations']:
			if existing_configuration['name'] == configuration['name']:
				return True
		return False

	# Add extra configurations
	with open('.vscode/c_cpp_properties.extra.json', 'r') as f:
		extra_properties = json.load(f, cls=JSONWithCommentsDecoder)
		for configuration in extra_properties['configurations']:
			if check_if_configuration_exists(configuration): 
				print(f'[extra_c_cpp_properties] Configuration "{configuration["name"]}" already exists, skipping.')
				continue
			c_cpp_properties['configurations'].append(configuration)

	# Modify `intelliSenseMode` for PlatformIO configuration to ARM to fix pointer size issues
	for configuration in c_cpp_properties['configurations']:
		if configuration['name'] == 'PlatformIO':
			configuration['intelliSenseMode'] = 'gcc-arm'

	with open('.vscode/c_cpp_properties.json', 'w') as f:
		json.dump(c_cpp_properties, f, indent=4)
else:
	# Running directly by PlatformIO/SCons
	this_script_file = inspect.getfile(lambda: None) # can't use sys.argv[0] nor __file__ in SCons loaded scripts
	p = subprocess.Popen([sys.executable, this_script_file], creationflags=subprocess.CREATE_NO_WINDOW)

```