platformio/platformio-core #3751: Problem upgrading jlink

blackketter, 2020-12-03:
### Configuration

macOS 11.0.1

PlatformIO, version 5.0.3

### Description of problem

pio update fails on upgrading jlink with internal server error.  Manually visiting `https://api.registry.platformio.org/v3/packages/platformio/tool/tool-jlink` also errors out.

#### Steps to Reproduce
```
 % pio update --dry-run
Checking platformio/tool-unity                1.20500.200612 @ ~1.20500.0        [Up-to-date]
Checking platformio/contrib-piohome           3.3.1 @ ~3.3.1                     [Up-to-date]
Checking platformio/tool-cppcheck             1.210.0 @ ~1.210.0                 [Up-to-date]

Platform Manager
================
Platform atmelavr
--------
Checking platformio/atmelavr                  3.0.0                              [Up-to-date]
Checking platformio/framework-arduino-avr-attiny 1.3.2 @ ~1.3.2                     [Up-to-date]
Checking platformio/tool-micronucleus         1.200.5 @ ~1.200.0                 [Up-to-date]
Checking platformio/framework-arduino-avr-digistump 1.6.7 @ ~1.6.7                     [Up-to-date]
Checking platformio/framework-arduino-avr     5.1.0 @ ~5.1.0                     [Up-to-date]
Checking platformio/toolchain-atmelavr        1.50400.190710 @ ~1.50400.0        [Up-to-date]
Checking platformio/tool-avrdude              1.60300.200527 @ ~1.60300.0        [Up-to-date]

Platform atmelsam
--------
Checking git+https://github.com/platformio/platform-atmelsam.git 5.0.1+sha.f5729be                  [Up-to-date]
Checking platformio/framework-arduino-samd-seeed 1.7.6 @ ~1.7.6                     [Up-to-date]
Checking platformio/tool-jlink                1.65200.0 @ ^1.63208.0             Error: Traceback (most recent call last):
  File "/usr/local/lib/python2.7/site-packages/platformio/__main__.py", line 109, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "/usr/local/lib/python2.7/site-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python2.7/site-packages/platformio/commands/__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "/usr/local/lib/python2.7/site-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python2.7/site-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/click/decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/platformio/commands/update.py", line 53, in cli
    ctx.invoke(cmd_platform_update, only_check=only_check)
  File "/usr/local/lib/python2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/platformio/commands/platform.py", line 421, in platform_update
    platform, only_packages=only_packages, only_check=only_check, silent=silent
  File "/usr/local/lib/python2.7/site-packages/platformio/package/manager/platform.py", line 120, in update
    p.update_packages(only_check)
  File "/usr/local/lib/python2.7/site-packages/platformio/platform/_packages.py", line 128, in update_packages
    show_incompatible=False,
  File "/usr/local/lib/python2.7/site-packages/platformio/package/manager/_update.py", line 105, in update
    outdated = self.outdated(pkg, to_spec)
  File "/usr/local/lib/python2.7/site-packages/platformio/package/manager/_update.py", line 46, in outdated
    reg_pkg = self.fetch_registry_package(pkg.metadata.spec)
  File "/usr/local/lib/python2.7/site-packages/platformio/package/manager/_registry.py", line 144, in fetch_registry_package
    result = regclient.get_package(self.pkg_type, spec.owner, spec.name)
  File "/usr/local/lib/python2.7/site-packages/platformio/clients/registry.py", line 142, in get_package
    cache_valid="1h",
  File "/usr/local/lib/python2.7/site-packages/platformio/clients/http.py", line 144, in fetch_json_data
    response = self.send_request(method, path, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/platformio/util.py", line 73, in wrapper
    return func(*args, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/platformio/clients/http.py", line 123, in send_request
    return getattr(self._session, method)(path, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/requests/sessions.py", line 521, in get
    return self.request('GET', url, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/platformio/clients/http.py", line 62, in request
    method, urljoin(self.base_url, url), *args, **kwargs
  File "/usr/local/lib/python2.7/site-packages/requests/sessions.py", line 508, in request
    resp = self.send(prep, **send_kwargs)
  File "/usr/local/lib/python2.7/site-packages/requests/sessions.py", line 618, in send
    r = adapter.send(request, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/requests/adapters.py", line 499, in send
    raise RetryError(e, request=request)
RetryError: HTTPSConnectionPool(host='api.registry.platformio.org', port=443): Max retries exceeded with url: /v3/packages/platformio/tool/tool-jlink (Caused by ResponseError('too many 500 error responses',))

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

===================================
```

bbain, 2020-12-03:
I also get this error when trying to install it for the first time

```
pio run --target upload
Processing nrf52840_dk (platform: nordicnrf52; board: nrf52840_dk; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Tool Manager: Installing platformio/tool-jlink @ ^1.63208.0
Error: Traceback (most recent call last):
  File "/private/tmp/platformio/lib/python3.9/site-packages/requests/adapters.py", line 439, in send
    resp = conn.urlopen(
  File "/private/tmp/platformio/lib/python3.9/site-packages/urllib3/connectionpool.py", line 846, in urlopen
    return self.urlopen(
  File "/private/tmp/platformio/lib/python3.9/site-packages/urllib3/connectionpool.py", line 846, in urlopen
    return self.urlopen(
  File "/private/tmp/platformio/lib/python3.9/site-packages/urllib3/connectionpool.py", line 846, in urlopen
    return self.urlopen(
  File "/private/tmp/platformio/lib/python3.9/site-packages/urllib3/connectionpool.py", line 836, in urlopen
    retries = retries.increment(method, url, response=response, _pool=self)
  File "/private/tmp/platformio/lib/python3.9/site-packages/urllib3/util/retry.py", line 573, in increment
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='api.registry.platformio.org', port=443): Max retries exceeded with url: /v3/packages/platformio/tool/tool-jlink (Caused by ResponseError('too many 500 error responses'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/private/tmp/platformio/lib/python3.9/site-packages/platformio/__main__.py", line 109, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "/private/tmp/platformio/lib/python3.9/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/private/tmp/platformio/lib/python3.9/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/private/tmp/platformio/lib/python3.9/site-packages/platformio/commands/__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "/private/tmp/platformio/lib/python3.9/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/private/tmp/platformio/lib/python3.9/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/private/tmp/platformio/lib/python3.9/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/private/tmp/platformio/lib/python3.9/site-packages/click/decorators.py", line 21, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/private/tmp/platformio/lib/python3.9/site-packages/platformio/commands/run/command.py", line 133, in cli
    process_env(
  File "/private/tmp/platformio/lib/python3.9/site-packages/platformio/commands/run/command.py", line 175, in process_env
    result = {"env": name, "duration": time(), "succeeded": ep.process()}
  File "/private/tmp/platformio/lib/python3.9/site-packages/platformio/commands/run/processor.py", line 79, in process
    result = p.run(build_vars, build_targets, self.silent, self.verbose, self.jobs)
  File "/private/tmp/platformio/lib/python3.9/site-packages/platformio/platform/_run.py", line 60, in run
    self.autoinstall_runtime_packages()
  File "/private/tmp/platformio/lib/python3.9/site-packages/platformio/platform/_packages.py", line 69, in autoinstall_runtime_packages
    self.pm.install(self.get_package_spec(name))
  File "/private/tmp/platformio/lib/python3.9/site-packages/platformio/package/manager/_install.py", line 48, in install
    pkg = self._install(
  File "/private/tmp/platformio/lib/python3.9/site-packages/platformio/package/manager/_install.py", line 99, in _install
    pkg = self.install_from_registry(spec, search_filters, silent=silent)
  File "/private/tmp/platformio/lib/python3.9/site-packages/platformio/package/manager/_registry.py", line 83, in install_from_registry
    package = self.fetch_registry_package(spec)
  File "/private/tmp/platformio/lib/python3.9/site-packages/platformio/package/manager/_registry.py", line 144, in fetch_registry_package
    result = regclient.get_package(self.pkg_type, spec.owner, spec.name)
  File "/private/tmp/platformio/lib/python3.9/site-packages/platformio/clients/registry.py", line 136, in get_package
    return self.fetch_json_data(
  File "/private/tmp/platformio/lib/python3.9/site-packages/platformio/clients/http.py", line 144, in fetch_json_data
    response = self.send_request(method, path, **kwargs)
  File "/private/tmp/platformio/lib/python3.9/site-packages/platformio/util.py", line 73, in wrapper
    return func(*args, **kwargs)
  File "/private/tmp/platformio/lib/python3.9/site-packages/platformio/clients/http.py", line 123, in send_request
    return getattr(self._session, method)(path, **kwargs)
  File "/private/tmp/platformio/lib/python3.9/site-packages/requests/sessions.py", line 555, in get
    return self.request('GET', url, **kwargs)
  File "/private/tmp/platformio/lib/python3.9/site-packages/platformio/clients/http.py", line 61, in request
    return super(EndpointSession, self).request(
[... cut ...]
requests.exceptions.RetryError: HTTPSConnectionPool(host='api.registry.platformio.org', port=443): Max retries exceeded with url: /v3/packages/platformio/tool/tool-jlink (Caused by ResponseError('too many 500 error responses'))

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================
```

ellensp, 2020-12-03:
Also getting this on linux_x86_64

valeros, 2020-12-03:
Thanks for reporting, should be fixed.

closed 2020-12-03 by valeros