platformio/platformio-core #3752: Compilation error on Windows 10

snikul, 2020-12-04:
Hi, Guys!
I compile the simplest example for Arduino Uno within VSC Platformio on W10 Pro.
With the following errors. I need your friendly assistance to solve my problem.
I have already read tones of comments and found no straight answer.

platformio.ini
[env:uno]
platform = atmelavr
board = uno
framework = arduino
upload_port = COM3
targets = upload

the result:

Processing uno (platform: atmelavr; board: uno; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/uno.html
PLATFORM: Atmel AVR (3.0.0) > Arduino Uno
HARDWARE: ATMEGA328P 16MHz, 2KB RAM, 31.50KB Flash   
DEBUG: Current (avr-stub) On-board (avr-stub, simavr)
PACKAGES:
 - framework-arduino-avr 5.1.0
 - tool-avrdude 1.60300.200527 (6.3.0)
 - toolchain-atmelavr 1.50400.190710 (5.4.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 5 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio\build\uno\src\main.cpp.o
Archiving .pio\build\uno\libFrameworkArduinoVariant.a
Compiling .pio\build\uno\FrameworkArduino\CDC.cpp.o
Compiling .pio\build\uno\FrameworkArduino\HardwareSerial.cpp.o
avr-g++: error: src\main.cpp: No such file or directory
avr-g++: fatal error: no input files
compilation terminated.
*** [.pio\build\uno\src\main.cpp.o] Error 1
c:\users\greenonion\.platformio\packages\toolchain-atmelavr\bin\../lib/gcc/avr/5.4.0/../../../../avr/bin/ar.exe: .pio\build\uno\libFrameworkArduinoVariant.a: No such file or directory
*** [.pio\build\uno\libFrameworkArduinoVariant.a] Error 1
Assembler messages:
Fatal error: can't create .pio\build\uno\FrameworkArduino\HardwareSerial.cpp.o: No such file or directory
Assembler messages:
Fatal error: can't create .pio\build\uno\FrameworkArduino\CDC.cpp.o: No such file or directory
*** [.pio\build\uno\FrameworkArduino\HardwareSerial.cpp.o] Error 1
*** [.pio\build\uno\FrameworkArduino\CDC.cpp.o] Error 1
================================================== [FAILED] Took 7.99 seconds ==================================================The terminal process "C:\Users\GreenOnion\.platformio\penv\Scripts\pio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

Hope to get your friendly reply.

Sergei Nikulin
Kaliningrad/Russia
serg.nikulin@gmail.com

snikul, 2020-12-11:
Hi, guys! Is anybody there?! I need your friendly assistance. Thanks a lot. Regards.

valeros, 2020-12-11:
Hi @snikul ! Do you have an antivirus in your system? Any chance that it's blocking PlatformIO?

snikul, 2020-12-11:
Hi. Valerii,

I have the standard Windows Defender installed with no problems in 2 years.
On my W10 PRO.
In my brandmauer I have:
                       code - privet box [checked]
Visual Studio Code - privat box [checked]
I tried VS Code with ESP-IDF plugin and could build, transfer to my esp32
target and monitor the output. When I try to repeat
any step-by-step example from youtube within VSC with platformio plugin I
get that same error message I reported in my message for help. I also
followed one of your advices from the internet.
- ivankravets Try to remove
c:\users\alber\.platformio\packages\toolchain-atmelavr folder and restart
build.
I deleted the folder and started the build process. After a while
I got 2 errors:
1. {
"resource":
"/c:/Users/xxx/Documents/PlatformIO/Projects/Blink_new/src/main.cpp",
"owner": "C/C++",
"code": "1696",
"severity": 8,
"message": "Обнаружены ошибки #include. Измените includePath. Волнистые
линии отключены для этой единицы трансляции
(C:\\Users\\xxx\\Documents\\PlatformIO\\Projects\\Blink_new\\src\\main.cpp).",
"source": "C/C++",
"startLineNumber": 7,       <----------------------- #include "Arduino.h"
"startColumn": 1,
"endLineNumber": 7,
"endColumn": 21
}
2. не удается открыть источник файл "stdlib.h" (dependency of "Arduino.h")

The text of example I tried to build:
/**
 * Blink
 *
 * Turns on an LED on for one second,
 * then off for one second, repeatedly.
 */
#include "Arduino.h"

// Set LED_BUILTIN if it is not defined by Arduino framework
// #define LED_BUILTIN 13

void setup()
{
  // initialize LED digital pin as an output.
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop()
{
  // turn the LED on (HIGH is the voltage level)
  digitalWrite(LED_BUILTIN, HIGH);

  // wait for a second
  delay(1000);

  // turn the LED off by making the voltage LOW
  digitalWrite(LED_BUILTIN, LOW);

   // wait for a second
  delay(1000);
}

What could it be?

My Best Regards
Sergei Nikulin