platformio/platformio-core #3759: Issues using advanced scripting

HamzaHajeir, 2020-12-07:
### Configuration

**Operating system**:

**PlatformIO Version** (`platformio --version`):

### Description of problem

Hi
I’m trying to start versioning my own project using platformio pre script

I’ve followed 2 methods, both ends with the same error. First one by adding !python:my_version_file.py into build_flags. (I’ll skip to the second method)

and the other method is by using extra_scripts.

I use git describe output to make version of my project.
I have 2 issues (3rd is less important now) :

When I hit pio run -v, It shows me less defines by 1 in verbose. For example it should define :

```
-DGIT_REVISION="%hash" 
-DGIT_REVISION_NO=0x%hash
-DMEEZANI_VERSION="0.0.1"
-DMEEZANI_VERSION_NO=00001
```
what I really see is only the first 2.

The second (more important issue, I get this error if appended CPP Defines (by this method or the first one) :

xtensa-esp32-elf-g++: fatal error: no input files

a full log with verbose :

> PS C:\Users\Hamza\Documents\PlatformIO\Projects\piotest> pio run -v
> Processing esp32doit-devkit-v1 (framework: arduino; platform: espressif32; board: esp32doit-devkit-v1; monitor_speed: 115200; upload_speed: 921600; build_flags: -w; extra_scripts: pre:git_rev_macro.py; lib_deps: https://github.com/meezani/FastLED.git)
> -------------------------------------------------------------------------------------------------------------------------
> 
> <SCons.Script.SConscript.SConsEnvironment object at 0x0000022B753A3160>
> CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
> PLATFORM: Espressif 32 (2.1.0) > DOIT ESP32 DEVKIT V1
> HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
> DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, 
> olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
> PACKAGES:
>  - framework-arduinoespressif32 3.10004.201016 (1.0.4)
>  - tool-esptoolpy 1.30000.201119 (3.0.0)
>  - toolchain-xtensa32 2.50200.80 (5.2.0)
> LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
> LDF Modes: Finder ~ chain, Compatibility ~ soft
> Found 27 compatible libraries
> Scanning dependencies...
> Dependency Graph
> |-- <FastLED> 3.3.3+sha.1c3fbd0 [git+https://github.com/meezani/FastLED.git] (C:\Users\Hamza\Documents\PlatformIO\Projects\piotest\.pio\libdeps\esp32doit-devkit-v1\FastLED)
> |   |-- <SPI> 1.0 (C:\Users\Hamza\.platformio\packages\framework-arduinoespressif32\libraries\SPI)
> Building in release mode
> xtensa-esp32-elf-g++ -o .pio\build\esp32doit-devkit-v1\lib237\FastLED\FastLED.cpp.o -c -fno-rtti -fno-exceptions -std=gnu++11 -w -Os -g3 -Wall -nostdlib -Wpointer-arith -Wno-error=unused-but-set-variable -Wno-error=unused-variable -mlongcalls -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -Wno-error=deprecated-declarations -Wno-error=unused-function -Wno-unused-parameter -Wno-sign-compare -fstack-protector -fexceptions -Werror=reorder -DGIT_REV="a3a41686779d492f7a235cce926c894d95e19624" -DMEEZANI_VERSION="0.0.1
> xtensa-esp32-elf-g++ -o .pio\build\esp32doit-devkit-v1\lib237\FastLED\bitswap.cpp.o -c -fno-rtti -fno-exceptions -std=gnu++11 -w -Os -g3 -Wall -nostdlib -Wpointer-arith -Wno-error=unused-but-set-variable -Wno-error=unused-variable -mlongcalls -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -Wno-error=deprecated-declarations -Wno-error=unused-function -Wno-unused-parameter -Wno-sign-compare -fstack-protector -fexceptions -Werror=reorder -DGIT_REV="a3a41686779d492f7a235cce926c894d95e19624" -DMEEZANI_VERSION="0.0.1
> xtensa-esp32-elf-g++: fatal error: no input files
> compilation terminated.
> xtensa-esp32-elf-g++ -o .pio\build\esp32doit-devkit-v1\lib237\FastLED\colorpalettes.cpp.o -c -fno-rtti -fno-exceptions -std=gnu++11 -w -Os -g3 -Wall -nostdlib -Wpointer-arith -Wno-error=unused-but-set-variable -Wno-error=unused-variable -mlongcalls -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -Wno-error=deprecated-declarations -Wno-error=unused-function -Wno-unused-parameter -Wno-sign-compare -fstack-protector -fexceptions -Werror=reorder -DGIT_REV="a3a41686779d492f7a235cce926c894d95e19624" -DMEEZANI_VERSION="0.0.1
> xtensa-esp32-elf-g++: fatal error: no input files
> compilation terminated.
> *** [.pio\build\esp32doit-devkit-v1\lib237\FastLED\FastLED.cpp.o] Error 1
> xtensa-esp32-elf-g++ -o .pio\build\esp32doit-devkit-v1\lib237\FastLED\colorutils.cpp.o -c -fno-rtti -fno-exceptions -std=gnu++11 -w -Os -g3 -Wall -nostdlib -Wpointer-arith -Wno-error=unused-but-set-variable -Wno-error=unused-variable -mlongcalls -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -Wno-error=deprecated-declarations -Wno-error=unused-function -Wno-unused-parameter -Wno-sign-compare -fstack-protector -fexceptions -Werror=reorder -DGIT_REV="a3a41686779d492f7a235cce926c894d95e19624" -DMEEZANI_VERSION="0.0.1
> *** [.pio\build\esp32doit-devkit-v1\lib237\FastLED\bitswap.cpp.o] Error 1
> xtensa-esp32-elf-g++: fatal error: no input files
> compilation terminated.
> *** [.pio\build\esp32doit-devkit-v1\lib237\FastLED\colorpalettes.cpp.o] Error 1
> xtensa-esp32-elf-g++: fatal error: no input files
> compilation terminated.
> *** [.pio\build\esp32doit-devkit-v1\lib237\FastLED\colorutils.cpp.o] Error 1

However I’ve created a sample project to reproduce the issue, It’s uploaded here.

For a quick eye :
git_rev_macro.py :

```py
import subprocess

projectTag = (
    subprocess.run(['git', 'describe','--tags','--abbrev=0'], capture_output=True, text=True).stdout
)
revision = (
    subprocess.run(['git', 'rev-parse','HEAD','--short=10'], capture_output=True, text=True).stdout
)
gitDescribeOutput = (
    subprocess.run(['git', 'describe','--tags'], capture_output=True, text=True).stdout
)
info = gitDescribeOutput.split('-')

Import("env")

print(env)

env.Append(CPPDEFINES=[
  ("GIT_REV", "\"%s\""%revision.strip()),
  ("MEEZANI_VERSION", "\"%s\"" % projectTag),
  ("MEEZANI_VERSION_NO", projectTagNumber),
  ("AHEAD_COMMITS", info[1])
])
```

**The content of `platformio.ini`:**
```ini
[platformio]
default_envs = esp32doit-devkit-v1

[common]
framework = arduino
monitor_speed = 115200
lib_deps = 
  https://github.com/meezani/FastLED.git

upload_speed = 921600

[env:esp32doit-devkit-v1]
framework = ${common.framework}
platform = espressif32
board = esp32doit-devkit-v1

monitor_speed = ${common.monitor_speed}
upload_speed = ${common.upload_speed}
build_flags = -w

extra_scripts = pre:git_rev_macro.py

lib_deps = ${common.lib_deps}```

**Source file to reproduce issue:**
```cpp

#include <Arduino.h>
#include <FastLED.h>

const char* version=MEEZANI_VERSION;
void setup() {
  // put your setup code here, to run once:
  
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

sethfischer, 2020-12-07:
[Dynamic build flags](https://docs.platformio.org/en/latest/projectconf/advanced_scripting.html) can be used to achieve a similar result. An example:

`platformio.ini`:

```ini
[common]
build_flags =
    !python3 define-git-revision.py
```

`define-git-revision.py`:

```python
#!/usr/bin/env python3
"""Git revision"""

import subprocess

REVISION = subprocess.check_output(["git", "rev-parse", "HEAD"]).strip()

print("-D SRC_REVISION='\"{}\"'".format(REVISION.decode("UTF-8")))
```

`src/main.cpp`:

```cpp
#ifndef SRC_REVISION
#define SRC_REVISION "(revision not defined)"
#endif

void loop()
{
    Serial.println(SRC_REVISION);
}
```

HamzaHajeir, 2020-12-08:
I've not used python before, But I've modified your solution and used the following code, Which gives the same error again : 

```
#!/usr/bin/env python3
"""Git revision"""

import subprocess

PROJECT_TAG = (
    subprocess.run(['git', 'describe','--tags','--abbrev=0'], capture_output=True, text=True).stdout
)
PROJECT_TAG_NO = PROJECT_TAG.replace('.','0')
REVISION = (
    subprocess.run(['git', 'rev-parse','HEAD','--short=10'], capture_output=True, text=True).stdout
)
gitDescribeOutput = (
    subprocess.run(['git', 'describe','--tags'], capture_output=True, text=True).stdout
)
info = gitDescribeOutput.split('-')

print("-D GIT_REV='\"{}\"'".format(REVISION))
print("-D MEEZANI_VERSION='\"{}\"'".format(PROJECT_TAG))
print("-D MEEZANI_VERSION_NO='{}'".format(PROJECT_TAG_NO))
print("-D AHEAD_COMMITS='{}'".format(info[1]))
```

Where I could be mistaken ?

HamzaHajeir, 2020-12-12:
It was because I'm fresh to python, This worked with me hitting achieving my goal: 

```py
#!/usr/bin/env python3
"""Git revision"""

import subprocess

revision = subprocess.check_output(["git", "rev-parse", "HEAD"]).strip().decode("utf-8")
REVISION = revision[:10]
gitDescribeOutput = subprocess.check_output(["git", "describe", "--tags"]).strip().decode("utf-8")
info = gitDescribeOutput.split('-')
if len(info) == 3:
    PROJECT_TAG = info[0]
    AHEAD_COMMITS = info[1]
else:
    PROJECT_TAG = info[0]+"-"+info[1]
    AHEAD_COMMITS = info[2]

PROJECT_TAG_NO = PROJECT_TAG.replace('.','0')

print("-D GIT_REV='\"{}\"'".format(REVISION))
print("-D MEEZANI_VERSION='\"{}\"'".format(PROJECT_TAG))
print("-D MEEZANI_VERSION_NO='{}'".format(PROJECT_TAG_NO))
print("-D AHEAD_COMMITS='{}'".format(AHEAD_COMMITS))
```

closed 2020-12-12 by HamzaHajeir