platformio/platformio-core #3762: Standard library function branches to invalid location and hardfault

reese-grimsley, 2020-12-13:
### Configuration

**Operating system**:
OSX Catalina
**PlatformIO Version** (`platformio --version`):
5.0.3
### Description of problem
I'm compiling for a custom board and framework, and I find that standard library functions like memset, memcpy, snprintf, etc. always hardfault. Disassembly is showing that these functions will branch unconditionally to a location in FLASH that does not hold any code, which unsurprisingly results in a the fault. The compiler doesn't generate any errors to suggest this would happen. 

I have been able to compile and run the same code in a) Atmel studio (our board uses the Atmel D51 M4F) and b) for a very similar adafruit board whose environment I can use to run basic tests. Specifically, this is the Adafruit Grand Central. No errors for either of these. I used actually use the adafruit linker script and startup/reset handler code as part of my custom framework, and much of my custom platform is based on theirs, removing arduino-specific things. I do notice that memset's disassembly is totally different for the adafruit compiled version vs. my own:

Adafruit env: 
```
0x000012de: 02 44           	add	r2, r0
0x000012e0: 03 46           	mov	r3, r0
0x000012e2: 93 42           	cmp	r3, r2
0x000012e4: 00 d1           	bne.n	0x12e8 <memset+10>
0x000012e6: 70 47           	bx	lr
0x000012e8: 03 f8 01 1b     	strb.w	r1, [r3], #1
0x000012ec: f9 e7           	b.n	0x12e2 <memset+4>
```

Custom env:
```
0x000005b4: 00 30           	adds	r0, #0
0x000005b6: a0 e1           	b.n	0x8fa           <--- branch to address with no fault and hard-fault
0x000005b8: 02 20           	movs	r0, #2
0x000005ba: 80 e0           	b.n	0x6be
0x000005bc: 02 00           	movs	r2, r0
0x000005be: 53 e1           	b.n	0x868
0x000005c0: 1e ff 2f 01     	vrhadd.u16	d0, d14, d31
0x000005c4: 01 10           	asrs	r1, r0, #32
0x000005c6: c3 e4           	b.n	0xffffff50
0x000005c8: fb ff ff ea     			; <UNDEFINED> instruction: 0xfffbeaff
```

The one from my environment is somewhat nonsensical, but isn't horribly garbled either.

I've tried to match the compiler flags used by adafruit as closely as possible; same goes with the code generated by Atmel Studio (although that's in a windows VM, making things slightly different). I've tried a wide variety of flags (expliclty setting -marm, -mcpu=cortex-m4, -mfloat-abi=softfp, -mthumb), although I notice that some of them only show up in the compiler stages and not at the linker, which I where I think this problem is occurring. Both environments use the same toolchain, and consequently, should be using the same standard libraries.

#### Steps to Reproduce
Hardware dependent, but I think you could use an Adafruit Grand Central in lieu of our custom PCB.

I run the following code as soon as possible: just after the reset handler, but before anything device/board specific; only CPU setup needs to happen first.

```
#include <string.h>
void foo(void) {

  volatile char x[128];
  volatile char y[128];
  memset((void*)x, 'A', sizeof(x)); //custom fails here
  memset((void*)y, 'q', sizeof(y));
  char letter = 'a';
  while(1) {
  	memcpy((void*)y, x, sizeof(x));
	letter++;
	memset((void*)x, letter, sizeof(x));
  }

}
```
### Actual Results
Hard fault internal to the first memset call (unless the compiler decided to inline memset, which I get around by making x, y larger) as it branches to an undefined location. If I run in the adafruit environment, it runs in the infinite loop as I would expect it to.

### Expected Results
This test code should simply initialize two arrays, fill them with values per a standard library function. In a loop, one array should be copied into the other, and then the copy-source should have all values incremented by 1 through another memset operation. It should never generate an error or fall into a interrupt handler (namely, hard-fault)

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:commonsense]
platform = https://bitbucket.org/ccsg-res/commonsense-pio-platform.git 
board = commonsense
framework = commonsense

debug_tool = jlink
upload_protocol = jlink

build_flags = ; Not all of these are strictly necessary, and many have been modified or removed while testing with no effect on the issue
    -Og
    -DDEBUG
    -g3
    -mlong-calls 
    --specs=nano.specs
    -L"/Users/work-reese/.platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/lib"
    -fdata-sections
    -ffunction-sections
    -mfloat-abi=softfp
    -march=armv7e-m
    -mfpu=fpv4-sp-d16
    -marm
    -mthumb-interwork

build_unflags = 
  -Os
  -mfloat-abi=hard

```

### Additional info
I figure this problem is related to linking, but it doesn't make sense that it works in one environment and not another. The adafruit env uses Arduino whereas mine is bare-metal. Perhaps one thing that would help is a way to specifically set linker flags in the INI file. I can set build flags, but only some of those show up while linking. I suppose I could also do this from the custom platform, but that's a very tedious thing to work with when debugging a problem like this.

I can also provide the output of verbose builds; mine is much shorter and simpler than the adafruit one because it lacks all the Arduino stuff.

Please let me know if anyone has ideas on what could be causing this issue. I haven't found this issue anywhere else online.

reese-grimsley, 2020-12-14:
I believe this is effectively solved; the issue was with some compiler flags for floating point support.

My issue was that I was using hybrid FP support with the -mfloat-abi=softfp option. Using -mfloat-abi=hard works for me. I confirmed that this change breaks both my custom build environment and the adafruit one. 

I believe there is some issue in the arm-none-eabi-gcc toolchain causing some issues when softfp at the same time as standard library functions. Those should each have their own multilib options; based on --print-multi-lib, both variants with the proper flags are present. I am not sure why this error occurs; it may be an issue with that version of the library or something else entirely, but I am certain this compiler flag/switch causes issue. I am happy to cross-post this on a repo that manages that toolchain package (although I'm unsure where it's hosted).

ivankravets, 2020-12-23:
Great to see that you fixed this issue. It's not a part of PlatformIO Core.

closed 2020-12-23 by ivankravets