platformio/platformio-core #3769: Download Right framework but install wrong.

GeneKong, 2020-12-18:
It's strange that upgrade from 4.3.4 to 5.03, our self defined framework download right, but move to packages with another unkown files.

```
Tool Manager: Installing git+https://github.com/espressif/ESP8266_RTOS_SDK.git#release/v3.2
git version 2.24.1.windows.2
Cloning into 'D:\ChargePile\Wifi\platformio\.cache\tmp\pkg-installing-myzo1weu'...
remote: Enumerating objects: 3707, done.
remote: Counting objects: 100% (3707/3707), done.
remote: Compressing objects: 100% (2950/2950), done.
remote: Total 3707 (delta 634), reused 2937 (delta 577), pack-reused 0
Receiving objects: 100% (3707/3707), 18.27 MiB | 6.73 MiB/s, done.
Resolving deltas: 100% (634/634), done.
Updating files: 100% (3181/3181), done.
Tool Manager: framework-esp8266-rtos-sdk @ 1.1.0+sha.f50cb20 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/custom8266/esp07.html
PLATFORM: custom Espressif 8266 (1.0.0+sha.2431a41) > Espressif Generic ESP8266 ESP-07
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-esp8266-rtos-sdk 1.1.0+sha.f50cb20
 - toolchain-xtensa-lx106-elf 5.2.0+100
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 6 compatible libraries
Scanning dependencies...
Dependency Graph
```
We get all files download in tmp, but packages wrong. Look at log, framework version is wrong too.

- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: Win10

**PlatformIO Version** (`platformio --version`): 5.0.3

### Description of problem

#### Steps to Reproduce

### Actual Results

Get "JSON Patch Tests" files.

### Expected Results

Get right framework files.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
Insert here...
```

**Source file to reproduce issue:**
```cpp
[env:esp07]
platform = git@github.com:GeneKong/framework-esp8266-rtos-sdk-idf-platformio.git
framework = esp8266-rtos-sdk
board = esp07
```

### Additional info

ivankravets, 2020-12-23:
Everything looks correct. See 
```
PLATFORM: custom Espressif 8266 (1.0.0+sha.2431a41) > Espressif Generic ESP8266 ESP-07
```

**sha.2431a41**

[image]

closed 2020-12-23 by ivankravets

GeneKong, 2020-12-23:
OK, we will test later.