platformio/platformio-core #3770: pio update command uninstalls GCC needed by pio run

cbc02009, 2020-12-21:
### Configuration

**Operating system**: Windows

**PlatformIO Version** (`platformio --version`): PlatformIO, version 5.0.3

### Description of problem
Every time I run the `pio update` command it uninstalls GCC, so the next time I run `pio run` it has to reinstall GCC

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```[env:seeed_xiao]
platform = atmelsam
board = seeed_xiao
framework = zephyr```

### Additional info

console output of `pio update`:
```
Updating platformio/contrib-piohome           3.3.1 @ ~3.3.1                     [Up-to-date]
Updating platformio/contrib-pysite            2.39.201019 @ ~2.39.0              [Up-to-date]
Updating platformio/tool-unity                1.20500.200612 @ ~1.20500.0        [Up-to-date]
Updating platformio/tool-scons                4.40001.0 @ ~4.40001.0             [Up-to-date]

Platform Manager
================
Platform atmelsam
--------
Updating platformio/atmelsam                  5.0.2                              [Up-to-date]
Updating platformio/framework-zephyr          2.20400.201210 @ ~2.20400.0        [Up-to-date]
Updating platformio/framework-zephyr-cmsis    0.0.0-alpha+sha.542b2296e6 @ 0.0.0-alpha+sha.542b2296e6[Up-to-date]
Updating platformio/framework-zephyr-canopennode 0.0.0-alpha+sha.468d350028 @ 0.0.0-alpha+sha.468d350028[Up-to-date]
Updating platformio/framework-zephyr-civetweb 0.0.0-alpha+sha.99129c5efc @ 0.0.0-alpha+sha.99129c5efc[Up-to-date]
Updating platformio/framework-zephyr-fatfs    0.0.0-alpha+sha.13697783bf @ 0.0.0-alpha+sha.13697783bf[Up-to-date]
Updating platformio/framework-zephyr-hal-atmel 0.0.0-alpha+sha.9f0a699a2f @ 0.0.0-alpha+sha.9f0a699a2f[Up-to-date]
Updating platformio/framework-zephyr-hal-st   0.0.0-alpha+sha.5b3ec3e182 @ 0.0.0-alpha+sha.5b3ec3e182[Up-to-date]
Updating platformio/framework-zephyr-libmetal 0.0.0-alpha+sha.0b23894a04 @ 0.0.0-alpha+sha.0b23894a04[Up-to-date]
Updating platformio/framework-zephyr-lvgl     0.0.0-alpha+sha.928b61c7c8 @ 0.0.0-alpha+sha.928b61c7c8[Up-to-date]
Updating platformio/framework-zephyr-mbedtls  0.0.0-alpha+sha.aef137b1af @ 0.0.0-alpha+sha.aef137b1af[Up-to-date]
Updating platformio/framework-zephyr-mcuboot  0.0.0-alpha+sha.a5d79cf8cc @ 0.0.0-alpha+sha.a5d79cf8cc[Up-to-date]
Updating platformio/framework-zephyr-mcumgr   0.0.0-alpha+sha.5051f9d900 @ 0.0.0-alpha+sha.5051f9d900[Up-to-date]
Updating platformio/framework-zephyr-open-amp 0.0.0-alpha+sha.724f7e2a45 @ 0.0.0-alpha+sha.724f7e2a45[Up-to-date]
Updating platformio/framework-zephyr-loramac-node 0.0.0-alpha+sha.3f545d76a2 @ 0.0.0-alpha+sha.3f545d76a2[Up-to-date]
Updating platformio/framework-zephyr-openthread 0.0.0-alpha+sha.07f430dac6 @ 0.0.0-alpha+sha.07f430dac6[Up-to-date]
Updating platformio/framework-zephyr-segger   0.0.0-alpha+sha.874d9e9696 @ 0.0.0-alpha+sha.874d9e9696[Up-to-date]
Updating platformio/framework-zephyr-tinycbor 0.0.0-alpha+sha.40daca97b4 @ 0.0.0-alpha+sha.40daca97b4[Up-to-date]
Updating platformio/framework-zephyr-tinycrypt 0.0.0-alpha+sha.3e9a49d267 @ 0.0.0-alpha+sha.3e9a49d267[Up-to-date]
Updating platformio/framework-zephyr-littlefs 0.0.0-alpha+sha.9e4498d1c7 @ 0.0.0-alpha+sha.9e4498d1c7[Up-to-date]
Updating platformio/framework-zephyr-mipi-sys-t 0.0.0-alpha+sha.957d46bc3c @ 0.0.0-alpha+sha.957d46bc3c[Up-to-date]
Updating platformio/framework-zephyr-trusted-firmware-m 0.0.0-alpha+sha.143df67555 @ 0.0.0-alpha+sha.143df67555[Up-to-date]
Updating platformio/tool-cmake                3.16.4 @ ~3.16.0                   [Up-to-date]
Updating platformio/tool-dtc                  1.4.7 @ ~1.4.7                     [Up-to-date]
Updating platformio/tool-ninja                1.9.0 @ ^1.7.0                     [Up-to-date]
Tool Manager: Removing toolchain-gccarmnoneeabi @ 1.80201.190214
Tool Manager: toolchain-gccarmnoneeabi @ 1.80201.190214 has been removed!

Platform nordicnrf52
--------
Updating platformio/nordicnrf52               5.0.1                              [Up-to-date]
Updating platformio/framework-zephyr          2.20400.201210 @ ~2.20400.0        [Up-to-date]
Updating platformio/framework-zephyr-cmsis    0.0.0-alpha+sha.542b2296e6 @ 0.0.0-alpha+sha.542b2296e6[Up-to-date]
Updating platformio/framework-zephyr-canopennode 0.0.0-alpha+sha.468d350028 @ 0.0.0-alpha+sha.468d350028[Up-to-date]
Updating platformio/framework-zephyr-civetweb 0.0.0-alpha+sha.99129c5efc @ 0.0.0-alpha+sha.99129c5efc[Up-to-date]
Updating platformio/framework-zephyr-fatfs    0.0.0-alpha+sha.13697783bf @ 0.0.0-alpha+sha.13697783bf[Up-to-date]
Updating platformio/framework-zephyr-hal-nordic 0.0.0-alpha+sha.d8a6ea9695 @ 0.0.0-alpha+sha.d8a6ea9695[Up-to-date]
Updating platformio/framework-zephyr-hal-st   0.0.0-alpha+sha.5b3ec3e182 @ 0.0.0-alpha+sha.5b3ec3e182[Up-to-date]
Updating platformio/framework-zephyr-libmetal 0.0.0-alpha+sha.0b23894a04 @ 0.0.0-alpha+sha.0b23894a04[Up-to-date]
Updating platformio/framework-zephyr-lvgl     0.0.0-alpha+sha.928b61c7c8 @ 0.0.0-alpha+sha.928b61c7c8[Up-to-date]
[... 180 lines cut ...]
Archiving .pio\build\adafruit_feather_nrf52840\zephyr\drivers\gpio\libdrivers__gpio.a
Archiving .pio\build\adafruit_feather_nrf52840\zephyr\arch\arch\arm\core\aarch32\libarch__arm__core__aarch32.a
Archiving .pio\build\adafruit_feather_nrf52840\zephyr\lib\posix\liblib__posix.a
Archiving .pio\build\adafruit_feather_nrf52840\modules\framework-zephyr-hal-nordic\lib..__framework-zephyr-hal-nordic.a
Archiving .pio\build\adafruit_feather_nrf52840\zephyr\libzephyr.a
Archiving .pio\build\adafruit_feather_nrf52840\zephyr\kernel\libkernel.a
Linking .pio\build\adafruit_feather_nrf52840\firmware-pre.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       14384 B         1 MB      1.37%
            SRAM:        4024 B       256 KB      1.54%
        IDT_LIST:          56 B         2 KB      2.73%
Generating ISR list .pio\build\adafruit_feather_nrf52840\zephyr\isrList.bin
Generating ISR table .pio\build\adafruit_feather_nrf52840\zephyr\isr_tables.c
Compiling .pio\build\adafruit_feather_nrf52840\zephyr_final\zephyr\isr_tables.c.o
Linking .pio\build\adafruit_feather_nrf52840\firmware.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       14384 B         1 MB      1.37%
            SRAM:        4024 B       256 KB      1.54%
        IDT_LIST:          0 GB         2 KB      0.00%
Checking size .pio\build\adafruit_feather_nrf52840\firmware.elf
Building .pio\build\adafruit_feather_nrf52840\firmware.hex
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   1.6% (used 4007 bytes from 248832 bytes)
Flash: [          ]   1.7% (used 13938 bytes from 815104 bytes)
============================================= [SUCCESS] Took 35.54 seconds =============================================```

ivankravets, 2020-12-23:
Do you use 3rd party spy antivirus software?

Please provide an output of 
```
pio system info
```

cbc02009, 2020-12-23:
I am not using any 3rd party anti-malware or antivirus.

pio system info:
```--------------------------  ------------------------------------------------------
PlatformIO Core             5.0.3
Python                      3.9.1-final.0
System Type                 windows_amd64
Platform                    Windows-10
File System Encoding        utf-8
Locale Encoding             cp1252
PlatformIO Core Directory   C:\Users\*****\.platformio
PlatformIO Core Executable  C:\Users\*****\.platformio\penv\Scripts\platformio.exe
Python Executable           c:\users\*****\.platformio\penv\scripts\python.exe
Global Libraries            0
Development Platforms       2
Tools & Toolchains          37
--------------------------  ------------------------------------------------------
```

ivankravets, 2020-12-23:
Is your project located on the special drive?

cbc02009, 2020-12-23:
> 
> 
> Is your project located on the special drive?

Sorry, what do you mean by special drive?

ivankravets, 2020-12-23:
OneDrive? Dropbox? Google Drive? Network Disk?

cbc02009, 2020-12-23:
Oh, gotcha. Nope it's just on the C:\ drive in the Windows Documents folder.

mcspr, 2021-01-04:
I wonder if this is related?

Was trying to build ESPEasy with the new toolchain package when already have these installed via Espurna builder:
https://github.com/xoseperez/espurna/blob/47538b9658d36777f81fdc78ce3592f2d8db9e68/code/platformio.ini#L125-L127
https://github.com/letscontrolit/ESPEasy/blob/e693db524076c35bf9f16cea59fafb7687426cac/platformio_esp82xx_envs.ini#L75
https://github.com/letscontrolit/ESPEasy/blob/e693db524076c35bf9f16cea59fafb7687426cac/platformio_core_defs.ini#L135-L137

Also notice the `platformio/framework-arduinoespressif8266` vs. `framework-arduinoespressif8266` in .inis

The pio-run managed to *delete* my existing git tree with an active development branch and 2 existing toolchain packages :/ It feels like it is too eager to optimize out the outdated versions:
```
>  C:\Users\maxim\.platformio\penv\Scripts\pio.exe run -e custom_beta_ESP8266_4M1M
Processing custom_beta_ESP8266_4M1M (platform: https://github.com/platformio/platform-espressif8266.git; board: esp12e; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Platform Manager: Installing git+https://github.com/platformio/platform-espressif8266.git
git version 2.29.2.windows.2
Cloning into 'C:\Users\maxim\.platformio\.cache\tmp\pkg-installing-5_nvu_5o'...
remote: Enumerating objects: 108, done.
remote: Counting objects: 100% (108/108), done.
Receiving objects:  15% (17/108)% (89/89), done.
remote: Total 108 (delta 57), reused 35 (delta 5), pack-reused 0
Receiving objects: 100% (108/108), 34.87 KiB | 430.00 KiB/s, done.
Resolving deltas: 100% (57/57), done.
Platform Manager: espressif8266 @ 2.6.2+sha.468a625 has been installed!
Tool Manager: Removing framework-arduinoespressif8266 @ 3.0.0-dev+sha.417aacc
Tool Manager: framework-arduinoespressif8266 @ 3.0.0-dev+sha.417aacc has been removed! // <--- my existing tree
Tool Manager: Removing toolchain-xtensa @ 5.100200.201223 // <--- toolchain I already had installed
Tool Manager: toolchain-xtensa @ 5.100200.201223 has been removed!
Tool Manager: Removing toolchain-xtensa @ 5.100200.200918 // <-- old toolchain, which I might've wanted to keep :/
Tool Manager: toolchain-xtensa @ 5.100200.200918 has been removed!
The platform 'https://github.com/platformio/platform-espressif8266.git' has been successfully installed!
The rest of the packages will be installed later depending on your build environment.
Tool Manager: Installing mcspr/toolchain-xtensa @ 5.100200.201223
Unpacking  [####################################]  100%
Tool Manager: toolchain-xtensa @ 5.100200.201223 has been installed!
Tool Manager: Installing git+https://github.com/esp8266/Arduino.git
```

mcspr, 2021-01-07:
Plus, I also noticed the same issue with the `pio platform update`. The culprit seems to be
https://github.com/platformio/platformio-core/blob/1ec2e55322d02ce331a2ad33c22c8b618db2a83d/platformio/package/manager/platform.py#L149

Since it purges everything that is not attached to platform as a dependency.
(and I also noticed that even with `--only-check` it still uninstalls my toolchain package :/)
```
> C:\Users\maxim\.platformio\penv\Scripts\pio.exe platform update -c
Platform espressif8266
--------
Checking platformio/espressif8266             2.6.2                              [Up-to-date]
Checking platformio/toolchain-xtensa          2.40802.200502 @ ~2.40802.191122   [Up-to-date]
Checking platformio/framework-arduinoespressif8266 3.20704.0 @ ~3.20704.0             [Up-to-date]
Checking platformio/tool-esptool              1.413.0 @ <2                       [Up-to-date]
Checking platformio/tool-esptoolpy            1.20800.0 @ ~1.20800.0             [Up-to-date]
Tool Manager: Removing framework-arduinoespressif8266 @ 3.0.0-dev+sha.100a8df
Tool Manager: framework-arduinoespressif8266 @ 3.0.0-dev+sha.100a8df has been removed!
Tool Manager: Removing toolchain-xtensa @ 5.100200.201223
Tool Manager: toolchain-xtensa @ 5.100200.201223 has been removed!
```

Another place that seems to manage uninstall comes from `get_core_package_dir()` named as `_remove_unnecessary_packages()`:
https://github.com/platformio/platformio-core/blob/1ec2e55322d02ce331a2ad33c22c8b618db2a83d/platformio/package/manager/core.py#L72
Which filters out anything without owner="platformio" and... removes it? This does seem to need an actual project to be called, which I have not tried just yet.

mcspr, 2021-01-23:
@ivankravets does this make sense then?
mcspr/platformio-core@bf393ac9

but, since this is not closed, I think you will come up with a better solution :)

ivankravets, 2021-01-23:
Resolved in https://github.com/platformio/platformio-core/commit/92655c30c18fdf75acc28650ad9d14c80c87e778

See also new command https://docs.platformio.org/en/latest/core/userguide/system/cmd_prune.html

closed 2021-01-23 by ivankravets