platformio/platformio-core #3775: Just upgraded to Platformio Core 5.0.3 but “Platformio Home” does not work anymore

sblantipodi, 2020-12-23:
As title

When I run:
platformio.exe -c clion home

it opens the web browser and then I see a page that says: “Loading…” and it never ends loading.

```
pio system info

PlatformIO Core 5.0.3
Python 3.7.2-final.0
System Type windows_x86
Platform Windows-10
File System Encoding utf-8
Locale Encoding cp1252
PlatformIO Core Directory C:\Users\sblantipodi.platformio
PlatformIO Core Executable C:\Users\sblantipodi.platformio\penv\Scripts\platformio.exe
Python Executable c:\users\sblantipodi.platformio\penv\scripts\python.exe
Global Libraries 0
Development Platforms 3
Tools & Toolchains 18
```

I wrote here:
https://community.platformio.org/t/platformio-core-5-0/15721/9

and Ivan said me to open an issue.

Thanks you guys, PlatformIO is super awesome!

ivankravets, 2020-12-24:
1. Please uninstall ALL Pythons from the system.
2. Restart VSCode.
Does it work now?

sblantipodi, 2020-12-24:
@ivankravets using Clion here, no, it not solved.

sblantipodi, 2020-12-25:
oops same problem on Visual Studio Code 5.0.3, I would not like to uninstall my pythons, is there a way to solve it?

sblantipodi, 2020-12-25:
I have the same issue here too:
https://community.platformio.org/t/platformio-libraries-updates-not-working-after-updating-to-5-x/16638

but 
platformio upgrade --dev
doesn't solved the problem

sblantipodi, 2020-12-25:
removing .platformio folder did the trick

closed 2020-12-25 by sblantipodi