platformio/platformio-core #3777: Using built-in library fails with KeyError: 'framework-stm32cube'

maxgerhardt, 2020-12-26:
What kind of issue is this?

- [X ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: Windows 10 x64

**PlatformIO Version** (`platformio --version`): `version 5.0.4b1`

### Description of problem

When adding a built-in library to lib-deps (SD card library for Arduino Uno WiFi rev2 in this case) by the name in the `library.properties`, which is `SD (MegaCore)`, installation and building of the library fails with a `KeyError: 'framework-stm32cube'` when the PIO core somehow tries to iterate through all packages / libraries.

#### Steps to Reproduce

1. Create a new project for the Arduino Uno WiFi Rev2
2. Add reference to built-in SD card library by adding `lib_deps = SD (MegaCore)`
3. Press Build

### Actual Results

```
> Executing task in folder atmel_hello: C:\Users\Max\AppData\Local\Programs\Python\Python38\Scripts\pio.exe run --environment uno_wifi_rev2 <

Processing uno_wifi_rev2 (platform: atmelmegaavr; board: uno_wifi_rev2; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelmegaavr/uno_wifi_rev2.html
PLATFORM: Atmel megaAVR (1.3.0) > Arduino Uno WiFi Rev2
HARDWARE: ATMEGA4809 16MHz, 6KB RAM, 47.50KB Flash
PACKAGES:
 - framework-arduino-megaavr 1.8.6
 - toolchain-atmelavr 1.70300.191015 (7.3.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Library Manager: Installing SD (MegaCore)
KeyError: 'framework-stm32cube':
  File "C:\Users\Max\AppData\Local\Programs\Python\Python38\Lib\site-packages\platformio\builder\main.py", line 170:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\Max\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Max\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Max\.platformio\platforms\atmelmegaavr\builder\main.py", line 149:
    target_elf = env.BuildProgram()
  File "C:\Users\Max\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Max\AppData\Local\Programs\Python\Python38\Lib\site-packages\platformio\builder\tools\platformio.py", line 62:
    env.ProcessProjectDeps()
  File "C:\Users\Max\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Max\AppData\Local\Programs\Python\Python38\Lib\site-packages\platformio\builder\tools\platformio.py", line 140:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\Max\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Max\AppData\Local\Programs\Python\Python38\Lib\site-packages\platformio\builder\tools\piolib.py", line 1062:
    project.install_dependencies()
  File "C:\Users\Max\AppData\Local\Programs\Python\Python38\Lib\site-packages\platformio\builder\tools\piolib.py", line 898:
    lm.install(spec)
  File "c:\users\max\appdata\local\programs\python\python38\lib\site-packages\platformio\package\manager\_install.py", line 48:
    pkg = self._install(
  File "c:\users\max\appdata\local\programs\python\python38\lib\site-packages\platformio\package\manager\library.py", line 93:
    if is_builtin_lib(spec.name):
  File "c:\users\max\appdata\local\programs\python\python38\lib\site-packages\platformio\commands\lib\helpers.py", line 49:
    for storage in storages or get_builtin_libs():
[... 181 lines cut ...]
The nRF52 Series are built for speed to carry out increasingly complex tasks in the shortest possible time and return to sleep, conserving precious battery power. They have a Cortex-M4F processor and are the most capable Bluetooth Smart SoCs on the market.

Home: https://www.nordicsemi.com/Products/nRF52-Series-SoC
Frameworks: arduino, mbed, zephyr
Packages: toolchain-gccarmnoneeabi, framework-mbed, framework-arduinoadafruitnrf52, framework-arduinonordicnrf5, framework-arduino-nrf52-mbedos, framework-zephyr, framework-zephyr-cmsis, framework-zephyr-canopennode, framework-zephyr-civetweb, framework-zephyr-fatfs, framework-zephyr-hal-nordic, framework-zephyr-hal-st, framework-zephyr-libmetal, framework-zephyr-lvgl, framework-zephyr-mbedtls, framework-zephyr-mcuboot, framework-zephyr-mcumgr, framework-zephyr-open-amp, framework-zephyr-loramac-node, framework-zephyr-openthread, framework-zephyr-segger, framework-zephyr-tinycbor, framework-zephyr-tinycrypt, framework-zephyr-littlefs, framework-zephyr-mipi-sys-t, framework-zephyr-trusted-firmware-m, tool-sreccat, tool-openocd, tool-nrfjprog, tool-jlink, tool-bossac-nordicnrf52, tool-cmake, tool-dtc, tool-ninja, tool-gperf
Version: 5.0.1

ststm32 ~ ST STM32
==================
The STM32 family of 32-bit Flash MCUs based on the ARM Cortex-M processor is designed to offer new degrees of freedom to MCU users. It offers a 32-bit product range that combines very high performance, real-time capabilities, digital signal processing, and low-power, low-voltage operation, while maintaining full integration and ease of development.

Home: http://www.st.com/web/en/catalog/mmc/FM141/SC1169?sc=stm32
Frameworks: arduino, cmsis, libopencm3, mbed, spl, stm32cube, zephyr
Packages: toolchain-gccarmnoneeabi, framework-mbed, framework-cmsis, framework-cmsis-stm32f0, framework-cmsis-stm32f1, framework-cmsis-stm32f2, framework-cmsis-stm32f3, framework-cmsis-stm32f4, framework-cmsis-stm32f7, framework-cmsis-stm32g0, framework-cmsis-stm32g4, framework-cmsis-stm32h7, framework-cmsis-stm32l0, framework-cmsis-stm32l1, framework-cmsis-stm32l4, framework-cmsis-stm32l5, framework-spl, framework-libopencm3, framework-arduinoststm32, framework-arduinoststm32-maple, framework-arduinostm32mxchip, framework-arduinoststm32l0, framework-stm32cubef0, framework-stm32cubef1, framework-stm32cubef2, framework-stm32cubef3, framework-stm32cubef4, framework-stm32cubef7, framework-stm32cubeg0, framework-stm32cubeg4, framework-stm32cubeh7, framework-stm32cubel0, framework-stm32cubel1, framework-stm32cubel4, framework-stm32cubel5, framework-zephyr, framework-zephyr-cmsis, framework-zephyr-canopennode, framework-zephyr-civetweb, framework-zephyr-fatfs, framework-zephyr-hal-st, framework-zephyr-hal-stm32, framework-zephyr-libmetal, framework-zephyr-lvgl, framework-zephyr-mbedtls, framework-zephyr-mcuboot, framework-zephyr-mcumgr, framework-zephyr-open-amp, framework-zephyr-loramac-node, framework-zephyr-openthread, framework-zephyr-segger, framework-zephyr-tinycbor, framework-zephyr-tinycrypt, framework-zephyr-littlefs, framework-zephyr-mipi-sys-t, framework-zephyr-trusted-firmware-m, tool-stm32duino, tool-openocd, tool-jlink, tool-dfuutil, tool-cmake, tool-dtc, tool-ninja, tool-gperf, tool-ldscripts-ststm32
Version: 10.0.1

teensy ~ Teensy
===============
Teensy is a complete USB-based microcontroller development system, in a very small footprint, capable of implementing many types of projects. All programming is done via the USB port. No special programmer is needed, only a standard USB cable and a PC or Macintosh with a USB port.

Home: https://www.pjrc.com/teensy
Frameworks: arduino, mbed
Packages: toolchain-atmelavr, toolchain-gccarmnoneeabi, framework-arduinoteensy, framework-mbed, tool-teensy, tool-jlink
Version: 4.11.1
```

valeros, 2020-12-28:
Thanks for reporting, fixed directly in the `ststm32` repository.

closed 2020-12-28 by valeros

thinkyhead, 2026-03-12:
If I run a test on the current Marlin codebase (`bugfix-2.1.x` branch) with this small set of changes to the vanilla config…
```sh
#
# Test esp32 (1) 
#
restore_configs
opt_enable WIFISUPPORT WEBSUPPORT OTASUPPORT
opt_set MOTHERBOARD BOARD_ESPRESSIF_ESP32 \
  TX_BUFFER_SIZE 64 \
  WIFI_SSID '"ssid"' \
  WIFI_PWD '"password"'
```

… the failure result is that weird **KeyError: 'framework-stm32cube':** Even though we have our imports specifying the versions to use. I thought maybe some new overlapping/conflicting dependency was somehow being included due to outside changes. Anyway, this is what now shows up before the build gets to the compile phase, while PlatformIO is in the library mangling phase…

```
PACKAGES: 
 - framework-arduinoespressif32 @ 3.10004.210126 (1.0.4) 
 - tool-esptoolpy @ 1.30000.201119 (3.0.0) 
 - toolchain-xtensa-esp32s3 @ 12.2.0+20230208 
 - toolchain-xtensa32 @ 2.50200.97 (5.2.0)
Converting Marlin.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Library Manager: Installing https://github.com/MarlinFirmware/TMCStepper/archive/v0.8.9.zip
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: TMCStepper@0.8.9 has been installed!
Library Manager: Installing esp32async/AsyncTCP @ 3.3.3
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80%
Library Manager: AsyncTCP@3.3.3 has been installed!
Library Manager: Installing mathieucarbou/ESP Async WebServer @ 3.0.6
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: ESP Async WebServer@3.0.6 has been installed!
Library Manager: Resolving dependencies...
Library Manager: Installing mathieucarbou/Async TCP @ ^3.1.4
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80%
Library Manager: Async TCP@3.1.4 has been installed!
Library Manager: Installing esphome/ESPAsyncTCP-esphome @ ^2.0.0
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: ESPAsyncTCP-esphome@2.0.0 has been installed!
KeyError: 'framework-stm32cube':
  File "/Users/thinkyhead/.platformio/penv/lib/python3.11/site-packages/platformio/builder/main.py", line 173:
    env.SConscript("$BUILD_SCRIPT")
  File "/Users/thinkyhead/.platformio/packages/tool-scons/scons-local-4.8.1/SCons/Script/SConscript.py", line 620:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/Users/thinkyhead/.platformio/packages/tool-scons/scons-local-4.8.1/SCons/Script/SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/Users/thinkyhead/.platformio/platforms/espressif32@2.1.0/builder/main.py", line 224:
    target_elf = env.BuildProgram()
  File "/Users/thinkyhead/.platformio/packages/tool-scons/scons-local-4.8.1/SCons/Util/envs.py", line 252:
    return self.method(*nargs, **kwargs)
  File "/Users/thinkyhead/.platformio/penv/lib/python3.11/site-packages/platformio/builder/tools/piobuild.py", line 63:
    env.ProcessProjectDeps()
  File "/Users/thinkyhead/.platformio/packages/tool-scons/scons-local-4.8.1/SCons/Util/envs.py", line 252:
    return self.method(*nargs, **kwargs)
  File "/Users/thinkyhead/.platformio/penv/lib/python3.11/site-packages/platformio/builder/tools/piobuild.py", line 153:
    plb = env.ConfigureProjectLibBuilder()
  File "/Users/thinkyhead/.platformio/packages/tool-scons/scons-local-4.8.1/SCons/Util/envs.py", line 252:
    return self.method(*nargs, **kwargs)
  File "/Users/thinkyhead/.platformio/penv/lib/python3.11/site-packages/platformio/builder/tools/piolib.py", line 1204:
    project.install_dependencies()
  File "/Users/thinkyhead/.platformio/penv/lib/python3.11/site-packages/platformio/builder/tools/piolib.py", line 1005:
    lm.install(spec)
  File "/Users/thinkyhead/.platformio/penv/lib/python3.11/site-packages/platformio/package/manager/_install.py", line 47:
    pkg = self._install(spec, skip_dependencies=skip_dependencies, force=force)
  File "/Users/thinkyhead/.platformio/penv/lib/python3.11/site-packages/platformio/package/manager/_install.py", line 130:
    self.install_dependencies(pkg)
  File "/Users/thinkyhead/.platformio/penv/lib/python3.11/site-packages/platformio/package/manager/_install.py", line 143:
    self.install_dependency(dependency)
  File "/Users/thinkyhead/.platformio/penv/lib/python3.11/site-packages/platformio/package/manager/library.py", line 89:
    not_builtin_conds.append(not self.is_builtin_lib(spec.name))
  File "/Users/thinkyhead/.platformio/penv/lib/python3.11/site-packages/platformio/package/manager/library.py", line 120:
    for storage in cls.get_builtin_libs():
  File "/Users/thinkyhead/.platformio/penv/lib/python3.11/site-packages/platformio/util.py", line 57:
    self.cache[key] = (time.time(), func(*args, **kwargs))
  File "/Users/thinkyhead/.platformio/penv/lib/python3.11/site-packages/platformio/package/manager/library.py", line 105:
    for storage in p.get_lib_storages():
  File "/Users/thinkyhead/.platformio/penv/lib/python3.11/site-packages/platformio/platform/base.py", line 226:
    pkg = self.get_package(opts["package"])
  File "/Users/thinkyhead/.platformio/penv/lib/python3.11/site-packages/platformio/platform/_packages.py", line 29:
    return self.pm.get_package(spec or self.get_package_spec(name))
  File "/Users/thinkyhead/.platformio/penv/lib/python3.11/site-packages/platformio/platform/_packages.py", line 21:
    owner=self.packages[name].get("owner"),
```