platformio/platformio-core #3782: Sha1 mismatch on tool-esptool-linux_armv6l-1.413.0.tar.gz

keesj, 2021-01-01:
What kind of issue is this?

- [X ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:

esphome@odroid:~$ platformio --version
PlatformIO, version 4.3.4

### Description of problem

I am using esphome to create a configuration for nodemcuv2 (ESP8266) and when esphome use platformio to download the tools it fails.

```
esphome@odroid:~$ cat depth.yaml 
esphome:
  name: depth
  platform: ESP82
```

```
Processing depth (board: nodemcuv2; framework: arduino; platform: espressif8266@2.6.2)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
PackageManager: Installing tool-esptool @ <2
Warning! Package Mirror: The 'sha1' sum 'e15fba5accf70d9c1574c47055ee9aedc0c3fe1c' of downloaded file 'tool-esptool-linux_armv6l-1.413.0.tar.gz' is not equal to remote 'bdfd4890e232df1876df63c8e676b12119e89c9f'
Looking for another mirror...
Error: Could not install 'tool-esptool' with version requirements '<2' for your system 'linux_aarch64'.

Please try this solution -> http://bit.ly/faq-package-manager
```
I found several issues that have the same problem (same sha1 mismatch) 

https://github.com/platformio/platform-espressif8266/issues/230
https://github.com/esphome/issues/issues/1510

I downloaded the "official" file from https://sourceforge.net/projects/platformio-storage/files/packages/ and manualy checked the checksum to be the expected checksum 

why@machine:~/Downloads$ sha1sum tool-esptool-linux_armv6l-1.413.0.tar.gz 
bdfd4890e232df1876df63c8e676b12119e89c9f  tool-esptool-linux_armv6l-1.413.0.tar.gz

I therefore think something is wrong on the mirror as both the issues 230 and 1510 show that the downloaded file have e15fba5accf70d9c1574c47055ee9aedc0c3fe1c as sha1sum can this be fixed?

keesj, 2021-01-02:
After performing pio upgrade and platformio upgrade the problem disappeared

closed 2021-01-02 by keesj

baomin, 2021-01-03:
pip3 install --upgrade platformio