platformio/platformio-core #3785: c_cpp_properties.json.tpl Incorrect

atestani, 2021-01-02:
The template  is generating comment lines with "//" comments which are not allowed in json files.  These lines (87-90):
//
// !!! WARNING !!! AUTO-GENERATED FILE!
// PLEASE DO NOT MODIFY IT AND USE "platformio.ini":
// https://docs.platformio.org/page/projectconf/section_env_build.html#build-flags
//

should be removed and some other way to include the warning be added.

maxgerhardt, 2021-01-03:
Reference to thread: https://community.platformio.org/t/failed-to-parse-c-cpp-properties-json/17861

ivankravets, 2021-01-04:
VSCode uses its own JSON parser that allows comments. It seems like you use some extension that tries to parse VSCode JSON file and as a bare JSON.

atestani, 2021-01-04:
I fixed it by finding I had *TWO* C/C++ extensions where one was an
extension pack by Microsoft v1.0.0 and the other C/C++ v 1.13 also by
Microsoft.  I was unable to uninstall either of the C/C++ extensions
because PIO "required them".  I uninstalled PIO and both extensions and
reinstalled PIO and the v1.13 C/C++ extension and the issue is resolved.

Thanks for the lead.

ivankravets, 2021-01-04:
Great that you resolved it!

Happy coding with PlatformIO!

closed 2021-01-04 by ivankravets