platformio/platformio-core #3787: Serial monitor EOL and send_on_enter filter not working properly

schleb, 2021-01-04:
Issue: 
- Serial monitor sends no EOL bytes with send_on_enter filter enabled (CR and/or LF bytes are missing at line end)
- Send-on_enter does not work at all with EOL set to CR or EOL set to LF. It only sends the line if EOL is set to CRLF

Complete description:
I have a simple setup in which an ESP32 microcontroller replies the bytes sent to it via serial port. It replys the bytes in decimal representation so that I can read them easily:

Code:
```
void loop()
{
   int byte;

   if (Serial.available())
   {
    byte = Serial.read();

    Serial.print("received:");
    Serial.println(byte, DEC);
  }
}
```

Monitor snap (of typing [a] [Enter] [b] [Enter] [c] [Enter])

> a
> received:97
> b
> received:98
> c
> received:99

The issue is as you can see easily: there is no CR or LF or CRLF sent by the monitor. If I use another Terminal it works. E.g. with Arduino IDE I receive LF (dec 10). The Arduino IDE has no echo of the sent bytes, so sending the same bytes (a, b, c) it looks like this:

> received:97
> received:10
> received:98
> received:10
> received:99
> received:10

The EOL parameter of the monitor is set to CRLF (default). If I change the EOL parameter of the monitor to CR or LF, the send_on_enter filter does not work. Nothing is beeing sent. If I then change the parameter back to CRLF odd things happen - it sends out all the old stuff with CR(13) or LF(10):

> a
> received:97
> b
> received:98
> c
> received:99
> --- EOL: CR ---
> a
> b
> c
> --- EOL: LF ---
> a
> b
> c
> --- EOL: CRLF ---
> a
> received:97
> received:13
> received:98
> received:13
> received:99
> received:13
> received:97
> received:10
> received:98
> received:10
> received:99
> received:10
> received:97
> b
> received:98
> c
> received:99

Platformio.ini:
monitor_speed = 115200
monitor_flags=
    --echo
monitor_filters=
    send_on_enter

System:
PIO Core 5.0.3 Home 3.3.1
VS Code 1.52.0 
Windows 10

The issue seems to be happened also to this guy: https://community.platformio.org/t/monitor-send-on-enter-filter-not-work-with-eol-lf/17356

ivankravets, 2021-01-27:
Thanks for the report! 

Please re-test with `pio upgrade --dev`.

niclasku, 2021-02-10:
This fix does not work for me. [`platformio/commands/device/filters/send_on_enter.py`](https://github.com/platformio/platformio-core/blob/d0cc06f7667c26723d02bad24355ae10318eee23/platformio/commands/device/filters/send_on_enter.py#L34) seems to remove the EOL from the buffer after it gets appended a few lines above. Changing the method to this works for me: 
```python
def tx(self, text):
  self._buffer += text
  if self._buffer.endswith(self._eol):
    return self._buffer
  return ""
```
Not sure if this hack has any side effects.

ivankravets, 2021-02-10:
This is the correct behavior. We send text ON ENTER and we don't send "enter". If you need  EOL chars - then do not use this filter.

@schleb Am I correct?

niclasku, 2021-02-10:
I wonder how I could send something like `command parameter\r\n` then. With defaults it would send `c\r\n`, `o\r\n`, `m\r\n` and so on. I would expect that if `send_on_enter` is enabled it sends `command parameter` with no `--eol` specified, `command parameter\r` with `-eol=CR` and `command paramter\r\n` with `--eol=CRLF`.

Same problem is mentioned here: [Forum](https://community.platformio.org/t/is-there-anyway-to-send-manually-typing-messages-to-arduino-through-serial-port-in-platformio/15326/19)

closed 2021-02-10 by ivankravets

niclasku, 2021-02-10:
Thank you very much for the fast response! This works pretty well for me, but I am not sure if there exists a use case where you want to use `send_on_enter` with no EOL chars at all. If you don't use the `--eol` flag it currently defaults to CRLF and there seems to be no option to disable sending of EOL chars if `send_on_enter` is enabled. Supporting for example `--eol=None` could help people in this situation.

ivankravets, 2021-02-10:
PySerial does not support this https://github.com/pyserial/pyserial/blob/master/serial/tools/miniterm.py#L902

schleb, 2021-02-11:
@ivankravets
> This is the correct behavior. We send text ON ENTER and we don't send "enter". If you need EOL chars - then do not use this filter.
> 
> @schleb Am I correct?

Mh, I also expected the EOL characters to be sent while `send_on_enter` is enabled. So the fix also did not work for me as I expected.

@niclasku
> Thank you very much for the fast response! This works pretty well for me, but I am not sure if there exists a use case where you want to use `send_on_enter` with no EOL chars at all. If you don't [...]

That's an interesting question :-) I would have said no, but indeed I just had this use case one week ago as I had to communicate with a matchbox laser module (https://integratedoptics.com/) which has a weird communication protocol. In order to send a command you have to send all command-bytes in a very short timing without EOL. The laser accepts the command in the moment that no further bytes follow. That is not possible by typing the command by hand in a terminal because after each byte you type the laser thinks the command is complete - and says "error" because the command is not complete. So here you need the possibility to send a complete line without EOL.

I agree: It would be nice to have all options: sending CR, LF, CRLF and none

Martinspen, 2022-01-10:
Hello,

Has anyone found a way to send a configurable EOL character with the send_on_enter filter?

I have reproduced basically the same steps as schleb. Using an Arduino Nano board, VSCode 1.63.2, python 3.10.1, macOS 12.0.1. Same code for troubleshooting:

```
void loop()
{
   int byte;
   if (Serial.available())
   {
    byte = Serial.read();
    Serial.print("received:");
    Serial.println(byte, DEC);
  }
}
```

Platformio.ini (I also tried the similar option with monitor_filters instead):

```
monitor_flags =
    --echo
    --eol
    LF
    --filter
    send_on_enter
```

Under these conditions, I can type whatever, nothing is sent/received.

What I find really weird is that if use pio device monitor in the terminal, the monitor works almost as I expect:

> % pio device monitor -b 9600 --echo --eol LF --filter send_on_enter
> --- Available filters and text transformations: colorize, debug, default, direct, hexlify, log2file, nocontrol, printable, send_on_enter, time
> --- More details at https://bit.ly/pio-monitor-filters
> --- Miniterm on /dev/cu.usbserial-AB0LRFSL  9600,8,N,1 ---
> --- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
> 
> 
> received:10
> 
> received:10
> a
> received:97
> received:10
> b
> received:98
> received:10
> c
> received:99
> received:10
> 
> --- exit ---
I say almost because I do have to type Enter twice the first time, which is weird but not a big deal.

Any suggestion on how to ensure in platformio.ini (or some other configuration file) that the serial monitor sends the requested EOL character when using send_on_enter?

commit 808ba603 (2021-01-27): Fixed an issue when "pio device monitor –eol" and “send_on_enter” filter do not work properly // Resolve #3787
commit 7c8508b6 (2021-02-10): Fixed an issue with device monitor when the “send_on_enter” filter didn’t send EOL chars // Resolve #3787