platformio/platformio-core #3788: [FEATURE REQUEST] Options merge through extends-like option

turgu1, 2021-01-05:
Version: PlatformIO IDE V2.2.1
Editor: VSCode
OS: Linux

Hello, it would be beneficial that the following platformio.ini would merge options, such that in my (over simplified) example below, for [env:device1], there would be two entries in the build_flags option, instead of one (with the current version of platformio, builds_flags in [env:device1] replaces the one from devices when `extends` is used. I was expecting that there would be some merge option available):

```
[devices]
build_flags =
  -D MAIN_DEFINE=2

[env:device1]
extends_with_merge = devices
build_flags =
  -D OTHER_DEFINE=1
```

I could have added `${device.build_flags}` under `-D OTHER_DEFINE=1` but when you have many of these repeated options, it defeat the purpose of clarity. Here is a real example that would benefit from this  feature (don't tell me to use [env] for this, the platformio.ini contains a lot more that can't rely on [env]):

```
[platformio]
default_envs = inkplate_6

[inkplate_common]
platform               = espressif32
board                  = esp-wrover-kit
framework              = espidf
monitor_speed          = 115200
upload_speed           = 256000
monitor_filters        = colorize
board_build.partitions = partitions.csv
build_flags =
  -O3
  -D DEBUGGING=0
  -D CONFIG_SPIRAM_CACHE_WORKAROUND
  -D EPUB_INKPLATE_BUILD=1
  -D EPUB_LINUX_BUILD=0
  -D INCLUDE_vTaskSuspend=1
  -D SHOW_TIMING=0
  !/usr/bin/pkg-config --cflags --libs lib_freetype/lib/pkgconfig/freetype2.pc
  -std=gnu++17
  -I include_global
lib_extra_dirs =
  lib_esp32
board_build.embed_files =
  embed/favicon.ico
  embed/upload_script.html
lib_deps =
  https://github.com/turgu1/ESP-IDF-InkPlate.git

[env:inkplate_6]
extends_with_merge     = inkplate_common
build_type  = release
build_flags =
  -D INKPLATE_6

[env:inkplate_6plus]
extends_with_merge     = inkplate_common
build_type  = release
build_flags =
  -D INKPLATE_6PLUS

[env:inkplate_10]
extends_with_merge     = inkplate_common
build_type  = release
build_flags =
  -D INKPLATE_10

```

Did I miss something from what is currently possible with the platform.ini structure? Is there another way to do it without compromising the clarity of the ini file? Maybe a new option named `merges` or something?
Thanks a lot for your help!!

Guy

ivankravets, 2021-01-05:
Thanks, I moved to PlatformIO Core.

lnlp, 2021-05-12:
Also see #3952 which is similar but does not limit to `build_flags` only.

softwaregravy, 2022-11-24:
That seems reasonable

turgu1, 2022-11-24:
This is it! Thanks guys!

Jason2866, 2022-11-24:
👍

CONSULitAS, 2023-02-15:
@ivankravets any news on this topic?

I have a different use case to merge. Please look at this platformio.ini snippet:

```
[OTA8266]
upload_protocol = espota
; use secret from _credentials_OTA.ini for later use in [env:My-Device-OTA]
ota_auth = ${secrets.ota_auth}
upload_flags =
  --auth=${ota_auth}
  --port=8266

[env:MyDevice1]
board = esp01_1m

; set device for Config.h
build_flags = ${env.build_flags}
    -DMyDevice1

[env: MyDevice1_OTA]
extends = MyDevice1, OTA8266
upload_port = MyDevice1.local
```

With the current implementation we get 
`Error: BoardConfig: Board is not defined`

But there are no conflicting properties which would be overwritten.

mawildoer, 2024-07-09:
This is exactly what I was hoping to do as well; multiple upload protocols to configure, multiple build configurations and I'd like to matrix them together.

Conceptually, this is the first thing to come to mind: https://docs.github.com/en/actions/using-jobs/using-a-matrix-for-your-jobs#using-a-matrix-strategy