platformio/platformio-core #3790: Inspect Code fails on certain board types with "Bad JSON"

sannoxdev, 2021-01-05:
Inspect memory works, inspect code fails with “Bad JSON”.  Certain board types work ok e.g. Arduino Uno (memory and code), while others  fail the Code analysis e.g. Adafruit Feather M0 or SeeedStudios Wio Terminal. The problem can be reproduced with the simple blink example used on the platformio website and the following platformio.ini:

[env:uno]
platform = atmelavr
board = uno
framework = arduino

[env:zero]
platform = atmelsam
board = zero
framework = arduino

[env:seeed_wio_terminal]
platform = atmelsam
board = seeed_wio_terminal
framework = arduino
platform_packages = framework-arduino-samd-seeed@https://github.com/Seeed-Studio/ArduinoCore-samd.git

All boards will compile ok. Tested against VSCODE 1.51.1 / PIO 5.0.3  on Windows 10 20H2 and Ubuntu 20.04 with the same result.

SERJSOCHI, 2021-01-17:
Same problem. Wemos D1 mini PRO.

Platform.ini
```
[env:d1_mini_pro]
platform = espressif8266
board = d1_mini_pro
framework = arduino
upload_speed = 921600
build_type = debug
build_flags = -Wall -Wextra -Wlogical-op -Wempty-body -Werror 
```

About:
```
Version: 1.52.1 (user setup)
Commit: ea3859d4ba2f3e577a159bc91e3074c5d85c0523
Date: 2020-12-16T16:34:46.910Z
Electron: 9.3.5
Chrome: 83.0.4103.122
Node.js: 12.14.1
V8: 8.3.110.13-electron.0
OS: Windows_NT x64 10.0.19042
```

sannoxdev, 2021-01-31:
Ivan,  can I assist in troubleshooting the problem? Happy to assist if I can.

commit ae26079e (2021-03-09): Fixed an issue when code inspection fails with "Bad JSON" // Resolve #3790
closed 2021-03-09 by ivankravets