platformio/platformio-core #3792: "Command line is too long" with esp32 under Windows

nathanjel, 2021-01-06:
### Configuration
Windows 10
**PlatformIO Version** (`platformio --version`):

### Description of problem

Hello, I just ran over the same issue. Replication - as I work on a proprietary solution - is not straightforward, but what should help is that my project imports 76 external libraries thru lib_deps. That results in the final linker command of 10280 bytes. 

Please see my platformio.ini and the resulting linker command for building the 21-pin version - it should give You a good idea how to reproduce this issue. I had to mask the files a bit - sorry for this.

[Too long line issue - platformio.ini - Pastebin.com](https://pastebin.com/4NL3F1Ng)
[Too long line issue - resulting linker command - Pastebin.com](https://pastebin.com/NK94WDRb)

I will also file an issue to the core library on GitHub
#### Steps to Reproduce

1.
2.
3.

### Actual Results

### Expected Results

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
Insert here...
```

**Source file to reproduce issue:**
```cpp
Insert here...
```

### Additional info

justoke, 2021-01-10:
I am having the same issue, although out of 3 env defined in the platformio.ini only the node32s builds, whereas the others fail with the "path too long". I tried moving the whole project to a folder in the root of my drive to short it as much as possible without any effect. I am using Windows 10. 

[image]

nathanjel, 2021-01-11:
I just tried with `lib_archive = false` and it fails on Windows 10

```ps
Linking .pio\build\esp32dev-21pin\bootloader.elf
Building .pio\build\esp32dev-21pin\bootloader.bin
esptool.py v3.0
Linking .pio\build\esp32dev-21pin\firmware.elf
xtensa-esp32-elf-g++: error: CreateProcess: No such file or directory
*** [.pio\build\esp32dev-21pin\firmware.elf] Error 1
=============================================================== [FAILED] Took 233.96 seconds ===============================================================
Environment     Status    Duration
--------------  --------  ------------
esp32dev-21pin  FAILED    00:03:53.956
========================================================== 1 failed, 0 succeeded in 00:03:53.956 ========================================================== 
PS C:\Work\E2-Sample-Demonstration> 
```

justoke, 2021-01-11:
> I just tried with `lib_archive = false` and it fails on Windows 10
```
I tried the same with the same failure

nathanjel, 2021-01-13:
I just managed to create a workaround based on original PIO `builder/tools/piomaxlen.py`. The fact it works confirms a bug in platformio-core. The [existing version supports large linker command line](https://github.com/platformio/platformio-core/blob/1ec2e55322d02ce331a2ad33c22c8b618db2a83d/platformio/builder/tools/piomaxlen.py#L84) only in case of large set of linker source files, but it does not handle libraries as of yet.

The workaround involves creating an additional script and referencing it from `platformio.ini`.

#### 1. place `win_linker_workaround.py` in project root directory, aside `platformio.ini`
```python
Import("env")

from os.path import isfile, join
import hashlib
import os

def _file_long_data_workaround(env, _data):
	data = env.subst(_data).replace("\\", "/")
	build_dir = env.subst("$BUILD_DIR")
	tmp_file = join(
		build_dir, "longcmd-%s" % hashlib.md5(data.encode()).hexdigest()
	)
	retval = '@"%s"' % tmp_file
	if not isfile(tmp_file):
		with open(tmp_file, "w") as fp:
			fp.write(data)
	return retval

def implant_hook_for_libflags():
	env.Replace(_long_libflags_hook=_file_long_data_workaround)
	coms = {}
	key = "LINKCOM"
	coms[key] = env.get(key, "").replace(
		"$_LIBFLAGS", "${_long_libflags_hook(__env__, _LIBFLAGS)}"
	)
	env.Replace(**coms)

implant_hook_for_libflags()
```

#### 2. do needed changes to `platformio.ini`
```ini
...
[env]
...
extra_scripts =
  pre:win_linker_workaround.py
...
```

#### And for me it solves the problem:
```ps
Linking .pio\build\esp32dev-21pin\firmware.elf
Building .pio\build\esp32dev-21pin\firmware.bin
Retrieving maximum program size .pio\build\esp32dev-21pin\firmware.elf
Checking size .pio\build\esp32dev-21pin\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  14.7% (used 48324 bytes from 327680 bytes)
Flash: [=======   ]  72.5% (used 1140630 bytes from 1572864 bytes)
esptool.py v3.0
============================================================================= [SUCCESS] Took 203.75 seconds =============================================================================

Environment     Status    Duration
--------------  --------  ------------
esp32dev-21pin  SUCCESS   00:03:23.755
============================================================================== 1 succeeded in 00:03:23.755 ==============================================================================
```

justoke, 2021-01-13:
I've tried that but still get the error:
[image]

IS there anyway to confirm the script has run?

nathanjel, 2021-01-13:
@justoke To my best guess - You're facing a different error - the one You reported occurs during source file compilation. My workaround only resolves the issue at final linking step. I guess in Your case a similar workaround for compile command could do the trick, but explaining it is beyond this forum.

Regarding checking if the script is being run - sure, add something like `print("HELLO, IT's ME, THE SCRIPT!!!!!!!!!!!!!")` in the `_file_long_data_workaround` function. You will notice this in the output.

justoke, 2021-01-13:
> @justoke To my best guess - You're facing a different error - the one You reported occurs during source file compilation. My workaround only resolves the issue at final linking step. I guess in Your [...]

Thanks for the tip - you are right this is compile time, but this could be promising as a workaround. I'll share my findings

closed 2021-01-27 by ivankravets

ivankravets, 2021-01-27:
Please re-test with `pio upgrade --dev`  Does it work now?

nathanjel, 2021-01-28:
```ps
esp32dev-21pin  SUCCESS   00:00:43.636
=========================================== 1 succeeded in 00:00:43.636 ===========================================
```

Works! :) 
@justoke looking at the resolution in the code, which is perfect <3, I think Your issue and the overall problem will be finally gone for good!

Big thanks to PlatformIO team!!

justoke, 2021-01-28:
> Big thanks to PlatformIO team!!

Thank you!

commit e2906e3b (2021-01-27): Refactored a workaround for a maximum command line character limitation // Resolve #3792