platformio/platformio-core #3793: Debugging not accessible on commandline

maxgerhardt, 2021-01-08:
What kind of issue is this?

- [X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: Win10 x64

**PlatformIO Version** (`platformio --version`): `PlatformIO Core, version 5.0.5a1`

### Description of problem

In the [documentation](https://docs.platformio.org/en/latest/core/userguide/cmd_debug.html#examples), an example is shown where the command `pio debug --interface=gdb -x .pioinit` gives one a GDB shell. 

In Windows at least, this seems to be hardly working. In a Windows CMD terminal, as well as the VSCode PlatformIO CLI, one is just "stuck" after the GDB invocation, all characters are being "swallowed" without being displayed or reacted to. Also, no openOCD output is visible, which would be nice debug indiciator.

#### Steps to Reproduce

1. Create project with configs below, use Bluepill as HW
2. Run `pio run -t upload` to verify upload via STLink is working
3. Run `pio debug --interface=gdb -x .pioinit` to reproduce behavior

### Actual Results

[image]

### Expected Results

Debugging starts over the CLI with OpenOCD output and controllable GDB shell.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:genericSTM32F103C8]
platform = ststm32
board = genericSTM32F103C8
framework = arduino
debug_tool = stlink
upload_protocol = stlink
```

**Source file to reproduce issue:**
```cpp
#include <Arduino.h>

void setup()
{
}

void loop()
{
}
```

### Additional info

ivankravets, 2021-01-22:
Yes, this is a known issue. There is a bug with Windows CMD and Python/Twisted bridge when forwarding STDIN to the GDB subprocess. We plan to refactor our unified debugger soon and port it to native Python 3 Async I/O.

We did something similar with PIO Home, see https://github.com/platformio/platformio-core/commit/6ff67aeadfc63b421c0f652c70ac0df5b8bb4098

So, now is a time for debugging!

ivankravets, 2021-03-17:
Hi @maxgerhardt,

We've started a massive work on refactoring PlatformIO Unified Debugger. A lot of interesting are comings soon.

Could you try the first 5.2.0a1? We moved away from Twisted and ported everything to native Python Async I/O. You should be able to debug now from CLI. 

Thanks!

closed 2021-03-17 by ivankravets

maxgerhardt, 2021-03-22:
I upgraded to 5.2.0a3 and now the text input to the GDB client is working in a standard `cmd.exe` shell. 

```
>pio debug --interface gdb -x .pioinit
Preparing firmware for debugging...
Processing stm8sblue_stm8s103f3 (platform: https://github.com/platformio/platform-ststm8.git; board: stm8sblue)
------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm8/stm8sblue.html
PLATFORM: ST STM8 (1.0.3+sha.dbba972) > ST STM8S103F3 Breakout Board
HARDWARE: STM8S103F3P6 16MHz, 1KB RAM, 8KB Flash
DEBUG: Current (stlink) External (stlink)
PACKAGES:
 - tool-stm8binutils 0.230.0 (2.30)
 - toolchain-sdcc 1.40100.12072 (4.1.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 1 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <STM8_headers> 0.0.1
Building in debug mode
Checking size .pio\build\stm8sblue_stm8s103f3\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   0.5% (used 5 bytes from 1024 bytes)
Flash: [          ]   1.4% (used 115 bytes from 8192 bytes)
============================================= [SUCCESS] Took 1.66 seconds =============================================

Environment           Status    Duration
--------------------  --------  ------------
stm8sblue_stm8s103f3  SUCCESS   00:00:01.657
============================================= 1 succeeded in 00:00:01.657 =============================================
Reading symbols from C:\Users\Max\Documents\PlatformIO\Projects\stm8_headers_baremetal_pio\.pio\build\stm8sblue_stm8s103f3\firmware.elf...done.
PlatformIO Unified Debugger -> http://bit.ly/pio-debug
PlatformIO: debug_tool = stlink
PlatformIO: Initializing remote target...
xPack OpenOCD, x86_64 Open On-Chip Debugger 0.10.0+dev-00378-ge5be992df (2020-06-26-09:29)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
srst_only separate srst_gates_jtag srst_open_drain connect_deassert_srst

Info : tcl server disabled
Info : telnet server disabled
Info : STLINK V2J37S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.253616
Info : clock speed 800 kHz
Info : starting gdb server for stm8s.cpu on pipe
Info : accepting 'gdb' connection from pipe
target halted due to debug-request, pc: 0x0000805b
warning: Can not parse XML target description; XML support was disabled at compile time
0x0000805b in main () at src\main.c:112
112         for (i=0; i<uninit; i++)
Loading section DATA, size 0x4 lma 0x1
Loading section INITIALIZED, size 0x1 lma 0x5
Loading section SSEG, size 0x1 lma 0x6
Loading section HOME, size 0x7 lma 0x8000
Loading section GSINIT, size 0x1a lma 0x8007
Loading section GSFINAL, size 0x3 lma 0x8021
Loading section INITIALIZER, size 0x1 lma 0x8024
Loading section CODE, size 0x4e lma 0x8025
Start address 0x8007, load size 121
Warn : keep_alive() was not invoked in the 1000 ms timelimit. GDB alive packet not sent! (1622 ms). Workaround: increase "set remotetimeout" in GDB
Transfer rate: 68 bytes/sec, 15 bytes/write.
target halted due to debug-request, pc: 0x00008000
target halted due to debug-request, pc: 0x00008000
Temporary breakpoint 1 at 0x802c: file src\main.c, line 96.
PlatformIO: Initialization completed
(gdb) PlatformIO: Resume the execution to `debug_init_break = tbreak main`
PlatformIO: More configuration options -> http://bit.ly/pio-debug
Continuing.

Temporary breakpoint 1, main () at src\main.c:96
96        sfr_CLK.CKDIVR.byte = 0x00;
(gdb) b main.c:109
Breakpoint 2 at 0x8048: file src\main.c, line 109.
(gdb) c
Continuing.

Breakpoint 2, main () at src\main.c:109
109         toggle_pin(&LED_PORT, LED_PIN);
(gdb)
(gdb) info registers
pc             0x8048   32840
a              0x0      0
x              0x9      9
y              0x27c0   10176
sp             0x3ff    1023
cc             0x28     40
xh             0x0      0
xl             0x9      9
yh             0x27     39
yl             0xc0     192
(gdb) backtrace
#0  main () at src\main.c:109
#1  0x00000000 in ?? ()
```

Works very nicely! Great work!

ivankravets, 2021-03-22:
@maxgerhardt thanks for the feedback! Yes, it works super fast now. Especially when you try to debug 2nd time. We cache a lot of information to speed up debugging.

commit b54a8b40 (2021-03-17): Refactor Unified Debugger to native Python Asynchronous I/O stack // Resolve #3793 , Resolve #3595