platformio/platformio-core #3796: Strange "extra_scripts" behavior

gsieben, 2021-01-08:
I have a workspace with two projects (A&B). In one of the projects (A) I have an extra script that automatically increments the build number on build. This works quite well. But now I have the problem when I compile the second project (B) without extra script. Despite that only this project (B) is compiled, the exstra script in project (A) is still executed and the build number is incremented. This is of course not wanted. What am I doing wrong?

[platformio]
default_envs = d1_mini

[env:d1_mini]
platform = espressif8266
board = d1_mini
framework = arduino

extra_scripts =
    pre:risebuilnumber.py ; Pyton script increasing the build number

lib_deps = bblanchon/ArduinoJson @ ^6.17.2
           Wire
           knolleary/PubSubClient @ ^2.8
           fastled/FastLED @ ^3.4.0

ivankravets, 2021-01-09:
Please provide a simple project to reproduce this issue.

gsieben, 2021-01-10:
No problem.

Attached is a folder with a workspace containing two (empty) projects. When compiling project A the extra script is executed. It increments, as mentioned, the build number in the Version.h file of project A. 

When you compile project B, the extra script of project A is also executed in the background. Also then the build number in project A is increased. 

[Workspace.zip](https://github.com/platformio/platformio-core/files/5792437/Workspace.zip)