platformio/platformio-core #3797: Inclusion of ArduinoJSON causes cppcheck to not recognize a call to a function

maxgerhardt, 2021-01-08:
What kind of issue is this?

- [X ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: Win x64

**PlatformIO Version** (`platformio --version`): `PlatformIO Core, version 5.0.5a1`

### Description of problem

Well, this is a weird one. See https://community.platformio.org/t/wrong-issues-from-cppcheck/18410. 

At first, a project is created for an ESP32Dev board. Then, a `display.h` and `display.cpp` with two functions is added. Those two functions are called in `main.cpp`. 

```cpp
#include <Arduino.h>
#include "display.h"

// cppcheck-suppress unusedFunction
void setup() {
   setup_display();
}

// cppcheck-suppress unusedFunction
void loop() {
   show_display("Test");
}
```

Running the a check gives

```
>pio check --skip-packages
Checking esp32dev > cppcheck (platform: espressif32; board: esp32dev; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
No defects found
============ [PASSED] Took 0.16 seconds ============
```

However, if one does a `#include <ArduinoJSON>` at the top, one gets

```
>pio check --skip-packages
Checking esp32dev > cppcheck (platform: espressif32; board: esp32dev; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
src\display.cpp:10: [low:style] The function 'setup_display' is never used. [unusedFunction]
src\display.cpp:20: [low:style] The function 'show_display' is never used. [unusedFunction]
```

#### Steps to Reproduce

1. Clone reproduction project at https://github.com/maxgerhardt/pio-cppcheck-arduinojson
2. Make sure it compiles: `pio run`
3. Run `pio check --skip-packages` to see that there are no defects
4. Modify the `#include <ArduinoJSON.h>` in `main.cpp`
5. Rerun ``pio check --skip-packages`
6. Observe wrong defects

### Actual Results

Detected defects which are no defects since function *is* called in `main.cpp`

### Expected Results

No defects are found. Inclusion of a library does not change the workings of the cppcheck tool.

### If problems with PlatformIO Build System:

All files found in reproduction project

### Additional info

```
t>pio system info
--------------------------  --------------------------------------------------------------------------
PlatformIO Core             5.0.5a1
Python                      3.8.6-final.0
System Type                 windows_amd64
Platform                    Windows-10
File System Encoding        utf-8
Locale Encoding             cp1252
PlatformIO Core Directory   C:\Users\Max\.platformio
PlatformIO Core Executable  C:\Users\Max\AppData\Local\Programs\Python\Python38\Scripts\platformio.exe
Python Executable           c:\users\max\appdata\local\programs\python\python38\python.exe
Global Libraries            0
Development Platforms       14
Tools & Toolchains          99
--------------------------  --------------------------------------------------------------------------
```

maxgerhardt, 2021-01-09:
Addendum: When I manually overwrite the files in `C:\Users\<user>\.platformio\packages\tool-cppcheck` with the [2.1.0](https://bintray.com/platformio/tool-packages/download_file?file_path=1da7e57-tool-cppcheck-windows_amd64-1.210.0.tar.gz) version (except `package.json` and `.piopm` file), to force the usage of version 2.1.0 instead of the latest 2.3.0, it does **not** report a defect.

GunZi200, 2021-01-09:
I'm seeing the same issue on my end after following the reproduction steps. Strange 🤔 Isn't this a bug on cppcheck's end? Maybe would should mention this at https://sourceforge.net/p/cppcheck/discussion/

maxgerhardt, 2021-01-09:
The question is whether it's an error in the way PlatformIO calls `cppcheck` or whether it's a bug in `cppcheck` itself. I could e.g. imagine that, since ArduinoJSON has a lot of `.ino` example files which also implement `setup()` and `loop()` of course, that those are somehow mistaken as the project's `setup()` and `loop()` implementations, in which the display functions are not called of course. But that could of course be entirely not the case, too.

I'll oben a bugtracker ticket on https://trac.cppcheck.net/, since that's where it should end up.. The page says if "just" a forum post is opened it takes another person to actually create the ticket.

maxgerhardt, 2021-01-10:
I could further narrow it down to which part in ArduinoJson.h causes the defect: `#include "ArduinoJson/Variant/VariantCompare.hpp"`. The reproduction project has been updated. A ticket was opened: https://trac.cppcheck.net/ticket/10103

valeros, 2021-03-11:
Hi @maxgerhardt ! Actually, the result when you include ArduinoJSON headers is the right one. Cppcheck is conifgured to analyze all source files in the `src` folder. Both files `display.cpp` and `main.cpp` contain unused functions and if we take into account suppressing two functions in the `main.cpp` file, then we should see two `unusedFunction` defects from `display.cpp` which is correctly reported in your second check with ArduinoJSON includes. But the problem here is that even if we remove the suppress comments in the `main.cpp` we still don't see the defects from this file and it seems like another issue with Cppcheck preprocessor. The situation is worsened in our case when sources are analyzed in bulk and Cppcheck doesn't report defects at all, even from other sources not affected by ArduinoJSON.

As you mentioned, Cppcheck v2.1 works better in this case, but downgrading the package to v2.1 could possibly introduce even more issues for other projects because v2.3 contains several fixes for their preprocessor. The best we can do here is to at least report defects from files independently from broken ones.

closed 2021-03-11 by valeros