platformio/platformio-core #3801: Problems with src_filter

Nullfati, 2021-01-17:
What kind of issue is this?

- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

### Configuration

**Operating system**: Windows 10

**PlatformIO Version** (`platformio --version`): 5.0.3

### Description of problem
Don't work 'src_filet' like expected. Have header file which generated by CMake by 'configure_file' command. This file stored in 'lib/targets/<target_name>/src'.

### Actual Results
The header file is not included in the build.

The error of build:
```Error
Compiling .pio\build\blackpill_f411ce\lib064\config\project_config.cpp.o
lib\config\src\project_config.cpp:3:10: fatal error: project_config_macros.h: No such file or directory

*******************************************************************************
* Looking for project_config_macros.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:project_config_macros.h"
* Web  > https://platformio.org/lib/search?query=header:project_config_macros.h
*
*******************************************************************************

    3 | #include <project_config_macros.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

### Expected Results
The header file is added to the project and build be successes.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[common_env]
lib_deps_external =
    ArduinoJson @ ^6.16.1
    ivanseidel/LinkedList @ 0.0.0-alpha+sha.dac3874d28
build_unflags =
build_flags =
    -D SERIAL_RX_BUFFER_SIZE=1024
default_src_filter =
    +<*>
    -<.git/>
    -<.svn/>
    -<example/>
    -<examples/>
    -<test/>
    -<tests/>
    -<lib/targets/>

[env:blackpill_f411ce]
framework = arduino
platform = ststm32
board = blackpill_f411ce
debug_tool = stlink
src_filter =
    ${common_env.default_src_filter}
    +<lib/targets/blackpill_f411ce/>
lib_deps =
    ${common_env.lib_deps_external}
build_unflags =
    ${common_env.build_unflags}
build_flags =
    ${common_env.build_flags}
```

Nullfati, 2021-01-18:
Also my cpp file:
```
#include "project_config.h"

#include <project_config_macros.h>

namespace project_cfg {

const int bluetooth_global_tx = BLUETOOTH_GLOBAL_TX;
const int bluetooth_global_rx = BLUETOOTH_GLOBAL_RX;
const int bluetooth_global_serial_speed = BLUETOOTH_GLOBAL_SERIAL_SPEED;
const int bluetooth_global_size_of_json = BLUETOOTH_GLOBAL_SIZE_OF_JSON;

const int task_manager_task_count = TASK_MANAGER_TASK_COUNT;

const int led_pin = LED_PIN;

}
```

ivankravets, 2021-01-18:
`src_filter` works only for project source files. If you need to exclude library source files, you need https://docs.platformio.org/en/latest/librarymanager/config.html#srcfilter

closed 2021-01-18 by ivankravets