platformio/platformio-core #3802: Please add support for CH32V103 MCU

FredBill1, 2021-01-19:
- [x] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

Hi, it would be great if you could add support for the CH32V103 MCU produced by WCH Technologies Corporation.

The official site of the MCU: [http://www.wch.cn/products/CH32V103.htm](url)
Product manual: [http://www.wch.cn/downloads/CH32V103DS0_PDF.html](url)
EVT: [http://www.mounriver.com](url)
Official IDE: [http://www.mounriver.com](url)

Kashouryo, 2021-04-01:
+1 for this request. This MCU is extremely cheap, powerful, and accessible to everyone that wants to getting into RISC-V

bsvdoom, 2023-01-18:
+1 after one year

MrAMS, 2023-02-06:
+1 after one year, or you can try [EIDE](https://em-ide.com/), it's a good alternative to platformio

maxgerhardt, 2023-02-18:
We're working on it in https://github.com/Community-PIO-CH32V/platform-ch32v.

davidbittonmbp, 2023-02-25:
@maxgerhardt that's great to hear. the hackster.io email today has an article on the MCU

https://www.hackster.io/news/muselab-s-nanoch32v003-puts-a-32-bit-risc-v-core-on-your-breadboard-for-just-1-50-e21955e89987?mc_cid=2546b1db81&mc_eid=2b8f7ba0dc