platformio/platformio-core #3804: UnicodeDecodeError: 'utf-8' codec

pvonmoradi, 2021-01-20:
### Configuration
Xubuntu 18.04
PlatformIO Core, version 5.0.4
### Description of problem
PIO upload shows errors:
```
make upload
platformio -f -c vim run --target upload
Processing waveshare_ble400 (platform: nordicnrf51; board: waveshare_ble400; framework: arduino)
-------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nordicnrf51/waveshare_ble400.html
PLATFORM: Nordic nRF51 (6.1.0) > Waveshare BLE400
HARDWARE: NRF51822 32MHz, 32KB RAM, 256KB Flash
DEBUG: Current (blackmagic) External (blackmagic, jlink, stlink)
PACKAGES: 
 - framework-arduinonordicnrf5 1.700.201209 (7.0) 
 - tool-jlink 1.68801.0 (6.88.1) 
 - tool-nrfjprog 1.90702.1 (9.7.2) 
 - tool-openocd 2.1000.200630 (10.0) 
 - tool-sreccat 1.164.0 (1.64) 
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 9 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Checking size .pio/build/waveshare_ble400/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   0.7% (used 244 bytes from 32768 bytes)
Flash: [          ]   0.9% (used 2484 bytes from 262144 bytes)
Configuring upload protocol...
AVAILABLE: blackmagic, jlink, nrfjprog, stlink
CURRENT: upload_protocol = jlink
Uploading .pio/build/waveshare_ble400/firmware.hex
SEGGER J-Link Commander V6.88b (Compiled Nov 27 2020 16:43:11)
DLL version V6.94a, compiled Jan 14 2021 11:51:14

J-Link Command File read successfully.
Processing script file...

J-Link connection not established yet but required for command.
Connecting to J-Link via USB...O.K.
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/home/pooya/.pyenv/versions/3.8.2/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/home/pooya/.pyenv/versions/3.8.2/lib/python3.8/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/home/pooya/.local/pipx/venvs/platformio/lib/python3.8/site-packages/platformio/proc.py", line 47, in run
    self.do_reading()
  File "/home/pooya/.local/pipx/venvs/platformio/lib/python3.8/site-packages/platformio/proc.py", line 70, in do_reading
    for byte in iter(lambda: self._pipe_reader.read(1), ""):
  File "/home/pooya/.local/pipx/venvs/platformio/lib/python3.8/site-packages/platformio/proc.py", line 70, in <lambda>
    for byte in iter(lambda: self._pipe_reader.read(1), ""):
  File "/home/pooya/.pyenv/versions/3.8.2/lib/python3.8/codecs.py", line 322, in decode
    (result, consumed) = self._buffer_decode(data, self.errors, final)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xc3 in position 101: invalid continuation byte
============================ [SUCCESS] Took 1.53 seconds ============================
```

#### Steps to Reproduce
1. create a blinky project for `nordicnrf51/waveshare_ble400` and choose `jlink` as upload method
2. make upload
3. target blinks but pio shows python errors

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:waveshare_ble400]
platform = nordicnrf51
board = waveshare_ble400
framework = arduino
upload_protocol = jlink
build_flags = -DNRF51_S130
; extra_scripts = post:extra_script.py
; lib_deps =

```

**Source file to reproduce issue:**
```cpp
#include <Arduino.h>

void setup() {
    Serial.begin(115200);
    pinMode(PIN_LED1, OUTPUT);
}
void loop() {
    digitalWrite(PIN_LED1, 1);
    delay(200);
    digitalWrite(PIN_LED1, 0);
    delay(200);
}
```
### Additional info
Installed PIO in a venv via pipx. Python version of that venv is `3.8.2`

ivankravets, 2021-01-20:
Could you provide an output of `pio system info`?

pvonmoradi, 2021-01-20:
> Could you provide an output of `pio system info`?

```
λ pio system info
--------------------------  ---------------------------------------------------
PlatformIO Core             5.0.4
Python                      3.8.2-final.0
System Type                 linux_x86_64
Platform                    Linux-5.4.0-62-generic-x86_64-with-glibc2.2.5
File System Encoding        utf-8
Locale Encoding             UTF-8
PlatformIO Core Directory   /home/pooya/.platformio
PlatformIO Core Executable  /home/pooya/.local/bin/platformio
Python Executable           /home/pooya/.local/pipx/venvs/platformio/bin/python
Global Libraries            0
Development Platforms       5
Tools & Toolchains          26
--------------------------  ---------------------------------------------------
```

ivankravets, 2021-01-20:
Are your source files in UTF-8? Do you print something in the extra script that has non-ASCII chars?

pvonmoradi, 2021-01-20:
@ivankravets I don't think so? Is there a way to check?
Note that this happens only on `flash` and only when `jlink` is used. (for black magic probe method this error was not shown).

ivankravets, 2021-01-20:
Please re-test with the latest development version of PIO Core 

```
pio upgrade --dev
```

ivankravets, 2021-01-20:
Does it work now?

closed 2021-01-20 by ivankravets

pvonmoradi, 2021-01-20:
@ivankravets 
Yes. Jlink method works now. Does the bug cause corruption in upload? Cause blinky was working fine is working fine now. How can I revert to latest stable?
```
Checking size .pio/build/my_env/firmware.elf
Building .pio/build/my_env/firmware.hex
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   0.8% (used 492 bytes from 65536 bytes)
Flash: [          ]   1.8% (used 9600 bytes from 524288 bytes)
Configuring upload protocol...
AVAILABLE: blackmagic, jlink, nrfjprog
CURRENT: upload_protocol = jlink
Uploading .pio/build/my_env/firmware.hex
SEGGER J-Link Commander V6.88b (Compiled Nov 27 2020 16:43:11)
DLL version V6.94a, compiled Jan 14 2021 11:51:14

J-Link Command File read successfully.
Processing script file...

J-Link connection not established yet but required for command.
Connecting to J-Link via USB...O.K.
Firmware: J-Link V9 compiled Dec 13 2019 11:14:50
Hardware version: V9.50
S/N: ***
License(s): }\xc3\xc, RDI, FlashBP, FlashDL, JFlash
VTref=3.250V
Target connection not established yet but required for command.
Device "NRF52832_XXAA" selected.

Connecting to target via SWD
InitTarget() start
InitTarget() end
Found SW-DP with ID 0x2BA01477
DPv0 detected
Scanning AP map to find all available APs
AP[2]: Stopped AP scan as end of AP map has been reached
AP[0]: AHB-AP (IDR: 0x24770011)
AP[1]: JTAG-AP (IDR: 0x02880000)
Iterating through AP map to find AHB-AP to use
AP[0]: Core found
AP[0]: AHB-AP ROM base: 0xE00FF000
CPUID register: 0x410FC241. Implementer code: 0x41 (ARM)
Found Cortex-M4 r0p1, Little endian.
FPUnit: 6 code (BP) slots and 2 literal slots
CoreSight components:
ROMTbl[0] @ E00FF000
ROMTbl[0][0]: E000E000, CID: B105E00D, PID: 000BB00C SCS-M7
ROMTbl[0][1]: E0001000, CID: B105E00D, PID: 003BB002 DWT
ROMTbl[0][2]: E0002000, CID: B105E00D, PID: 002BB003 FPB
ROMTbl[0][3]: E0000000, CID: B105E00D, PID: 003BB001 ITM
ROMTbl[0][4]: E0040000, CID: B105900D, PID: 000BB9A1 TPIU
ROMTbl[0][5]: E0041000, CID: B105900D, PID: 000BB925 ETM
Cortex-M4 identified.
PC = 000015FC, CycleCnt = 0005905F
R0 = 00003F87, R1 = 000003E8, R2 = 000821C9, R3 = 00000000
R4 = 00000064, R5 = 00003F3C, R6 = 00000000, R7 = 00000000
R8 = 00000000, R9 = 00000000, R10= 20000000, R11= 00000000
R12= 00000000
SP(R13)= 2000FFE0, MSP= 2000FFE0, PSP= 00000000, R14(LR) = 000015FD
XPSR = 21000000: APSR = nzCvq, EPSR = 01000000, IPSR = 000 (NoException)
CFBP = 00000000, CONTROL = 00, FAULTMASK = 00, BASEPRI = 00, PRIMASK = 00

FPS0 = 00000000, FPS1 = 00000000, FPS2 = 00000000, FPS3 = 00000000
FPS4 = 00000000, FPS5 = 00000000, FPS6 = 00000000, FPS7 = 00000000
FPS8 = 00000000, FPS9 = 00000000, FPS10= 00000000, FPS11= 00000000
FPS12= 00000000, FPS13= 00000000, FPS14= 00000000, FPS15= 00000000
FPS16= 00000000, FPS17= 00000000, FPS18= 00000000, FPS19= 00000000
FPS20= 00000000, FPS21= 00000000, FPS22= 00000000, FPS23= 00000000
FPS24= 00000000, FPS25= 00000000, FPS26= 00000000, FPS27= 00000000
FPS28= 00000000, FPS29= 00000000, FPS30= 00000000, FPS31= 00000000
FPSCR= 00000000

Downloading file [.pio/build/my_env/firmware.hex]...
Comparing flash   [100%] Done.
J-Link: Flash download: Bank 0 @ 0x00000000: Skipped. Contents already match
O.K.

Reset delay: 0 ms
Reset type NORMAL: Resets core & peripherals via SYSRESETREQ & VECTRESET bit.
Reset: Halt core after reset via DEMCR.VC_CORERESET.
Reset: Reset device via AIRCR.SYSRESETREQ.

Script processing completed.

=========================== [SUCCESS] Took 10.50 seconds ===========================
```

ivankravets, 2021-01-20:
No, the fix does not affect code building. Please review the source code of your proejct.

commit 52b22b57 (2021-01-20): Fixed a "UnicodeDecodeError: 'utf-8' codec can't decode byte" // Resolve #3804 , Resolve #3417