platformio/platformio-core #3805: Please add support for RPi Pico (RP2040)

harry1453, 2021-01-21:
Please could you add support for the new [RPi Pico (RP2040 MCU)](https://www.raspberrypi.org/products/raspberry-pi-pico/)?

It is a Cortex M0+ chip but I don't think any of the existing platforms cover it as it's the first RPi to not support Linux (out of the box, at least...).

[Board Datasheet](https://datasheets.raspberrypi.org/pico/pico_datasheet.pdf)
[MCU Datasheet](https://datasheets.raspberrypi.org/rp2040/rp2040_datasheet.pdf)
[C/C++ SDK Documentation](https://datasheets.raspberrypi.org/pico/sdk/pico_c_sdk.pdf)

ivankravets, 2021-01-23:
Could you move this issue to https://github.com/raspberrypi/pico-sdk? Or, as minimum, duplicate it. Thanks.

PlatformIO is open source platform, so, anyone can contribute with own dev-platform. Let's wait when @raspberrypi team provide official integration of RP2040 for PlatformIO.

/cc @liamfraser, @kilograham

blackketter, 2021-01-23:
I'm trying to get my head around how this works, long term.  

I assume that the RP2040 would be added as a new platform and the pico-sdk would be a new framework.  

The Arduino team has stated that they've already started adding RP2040 chip with the Arduino ecosystem, which would be another another framework.  

Finally, the RPi Pico is the first of many RP2040-based boards that have already been announced, so those would need to be added too, on an ongoing basis.

So... In the mean time, what help is needed to get this going?

Spongman, 2021-01-29:
> I don't believe this is something we would do, ... Seems more like a job for the PlatformIO devs

this finger-pointing graph has a cycle.

does this mean PlatformIO won't support the RPi pico?

harry1453, 2021-01-30:
I don't have much experience with the internals of PlatformIO, but I'm tentatively starting to write a platform repo. Might not go anywhere.

maxgerhardt, 2021-01-31:
I have experience with custom platforms since I've developed https://github.com/maxgerhardt/platform-w60x (yes it's WIP but it can successfully compile the Arduino core, the native vendor's SDK, has the openocd upload tools and has debugging), so I might give this a shot, too.  Since the Raspberry Pi team per quote above isn't thinking about PlatformIO support, we have to get our hands dirty ourselves.

On a sidenote, if the platform is well-developed, it might soon well be the standard of Raspberry Pi Pico C/C++ programming. The [SDK installation + VSCode setup process](https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf) is lengthy and includes a lots of working in the shell, exporting environment variables, freaking compiling OpenOCD yourself and creating the VSCode configurations yourself, The build system is CMake and this is a thing new users will have to learn to configure too if they're unfamiliar with it. Just listen to https://youtu.be/cVHCllbN3bQ?t=846 about how this works out for some users.

If this is in PlatformIO, it's going to be a one-click installation / "create new Pico project" and PlatformIO will download and install everything, and users can use the familiar library management and `platformio.ini` configs, unified across pretty much all of PlatformIO, instead of CMake configs. 

@ivankravets what do you think about a user-developed platform after the responses in the linked pico-sdk issue?

ivankravets, 2021-02-02:
> the build system is CMake and this is a thing new users will have to learn to configure too if they're unfamiliar with it

Yes, you are right. The main problem here is CMake that has a limited API (see https://gitlab.kitware.com/cmake/cmake/-/issues/20463) if you decide to automate the build process. What is more, we even not saying anything about its complex syntax. We have a totally different vision on how embedded development workflow should look, and we do not agree that every embedded developer should learn these new build systems and their syntaxes (Make, CMake, Ninja, etc). Build tools change very often. That was the reason why we didn't re-invent wheels, and use industry-leading [SCons](https://scons.org/) build engine based on the native Python interpreter.

> what do you think about a user-developed platform 

It does not mean that we don't want to provide integration for Raspberry Pi Pico. PlatformIO.Org is an open-source project, and everyone can contribute. We work on community requests and report every month what we did for the PlatformIO.Org project. You can check reports at https://piolabs.com/blog/news/

If we talk about PlatformIO.Org, PlatformIO Core is our main priority, the second is maintaining what we did before for the last 6 years, and the last (if we have spare time for OSS) we work on the community requests related to 3rd party hardware and software products.

RPi SDK is based on CMake that didn't have a good API to export build metadata. What does it mean? It means, that you need to manually reproduce the whole build workflow using PlatformIO Build System API (based on SCons) and replicate everything that CMake does. A few problems:

- The only RPi team releases a new version of their SDK - you will need to re-evaluate the whole build chain and apply all changes (sometimes it can take weeks, ESP-IDF is a good example)
- You will always receive bad feedback from the community that "there is a new version of SDK "X" and why PlatformIO does not support it".
- The only you start doing some work WITHOUT support from the original hardware/software vendor - you will be responsible for everything that you do. It means user complaints, criticism, and etc. 

----------

We already passed this way. There is no gratitude here. **The ONLY ONE** reasonable way to solve this issue is to create a joint value proposition with both parties. We already contacted the RPi Trade team and have not received an answer yet. We will keep you updated.

Happy coding with PlatformIO! 😊 

Regards,
Your friends at PlatformIO Labs

blackketter, 2021-02-02:
Thank you, @ivankravets, for the thorough explanation!  I really hope that the Raspberry Pi team can support PlatformIO, it would really lower the barrier to entry for getting folks started with developing on the Pico.

Spongman, 2021-02-03:
> to solve this issue is to create a joint value proposition with both parties

jhmaloney, 2021-02-06:
One approach would be to wait until Arduino releases their framework for the RP2040, then integrate that into PlatformIO. Since the MicroBlocks virtual machine depends on many of the standard Arduino API's, that approach would be ideal for us. (We already use PlatformIO to build for several dozen boards).

I've just installed the pico-sdk and its workflow is complex and cumbersome, as others have pointed out. What I love about PlatformIO is how easy it is. and how it manages all the tools, frameworks, and libraries for you. To add a new board, I typically just add a few lines to my platformio.ini file and PlatformIO installs the entire toolchain, framework, and libraries. I waste zero time setting up my workflow for the new board so I can focus on the stuff that actually matters.

Wiz-IO, 2021-02-20:
just experiment as beginning...
https://github.com/Wiz-IO/wizio-pico

2bndy5, 2021-04-03:
Don't know why we haven't seen any official arduino core for the rp2040, but someone got impatient and released their own [unofficial arduino core for the rp2040](https://github.com/earlephilhower/arduino-pico).

jpraczyk, 2021-04-09:
ArduinoCore-mbed 2.0.0 now supports rp2040
https://github.com/arduino/ArduinoCore-mbed/releases/tag/2.0.0

valeros, 2021-04-23:
Hi guys! We've added initial support for RP2040, but for now only with Arduino framework. Unfortunately, we don't have any RP2040-based board at hand, so it would be great if someone could evaluate essential functionality (compiling, uploading, etc.). For further discussion please use the  https://github.com/platformio/platform-raspberrypi repository.  

And just in case, here is a typical configuration for the upstream version of the platform:
```ini
[env:pico]
platform = https://github.com/platformio/platform-raspberrypi.git
framework = arduino
board = pico
```

danielkucera, 2021-04-23:
It builds, uploads and blinks. Thank you!
The only thing missing was upload port, in my case:
```
upload_port = /media/danman/RPI-RP2
```

valeros, 2021-04-26:
@danielkucera ! Do you mean PlatformIO wasn't able to upload firmware without specifying the upload port? Does the Arduino IDE requires the upload port as well? Anyway, please open an issue https://github.com/platformio/platform-raspberrypi/issues if something doesn't work as expected.

aallan, 2021-05-27:
> We already contacted the RPi Trading team and have not received an answer yet. 

As we have said previously, we're happy to support you if you have any questions. However we don't have any developer time available to contribute. Sorry, but this isn't something we're going to do.

ivankravets, 2023-06-12:
Reopen as a part of https://github.com/platformio/platform-raspberrypi/pull/36#issuecomment-1587171274