platformio/platformio-core #3806: library.json on native platforms

jcw, 2021-01-21:
- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

I have a project which needs to supports lots of platform. The platform-specific parts are placed in separate libraries, e.g. `lib/arch-native/`, `lib/arch-stm32/`, etc. Common code, including `main()` should remain as portable as possible, preferably with few `#if`'s, so my solution is to have each of these libraries provide an include file called `arch.h`, with a matching `#include <arch.h>` in the main applications (there are many, including demos).

To keep the libraries from interfering with each other, I added a `library.json` file to each lib, as follows:

* `lib/arch-native/library.json` contains: `{ "platform": "native" }`
* `lib/arch-stm32/library.json` contains: `{ "platform": "ststm32" }`
* etc ...

It looks like this does not work. The "platforms" setting is not treated as filter by the library dependency manager.

I tried the same using "frameworks" instead (with settings "native", "stm32cube", etc), but "native" is not recognised as a framework. It _does_ work to prevent use in a native build when set to "stm32cube", but it does not work to prevent an stm32 build from seeing a library marked as "native".

I did not include name+version info in `library.json`, since these libraries are not meant to be packaged separately.

ivankravets, 2021-01-22:
Have you tried to set `strict` mode for a library using https://docs.platformio.org/en/latest/librarymanager/config.html#libcompatmode ?

jcw, 2021-01-22:
Woohoo, **that works perfectly!** (I've added it to `[env]`, so that it applies everywhere unless overridden).

 I think this approach offers a very convenient way to split off platform-dependent source code, with minimal impact on the main application code. Perhaps also useful as suggestion in PIO's documentation.

Thank you, Ivan. Feel free to close this issue.

ivankravets, 2021-01-22:
I think this is a bug and PlatformIO should handle it automatically. We will work on the fix.

ivankravets, 2021-01-22:
Please upgrade to the latest development version via `pio upgrade --dev`.

Doesit work now without any extra configuration steps?

closed 2021-01-22 by ivankravets

jcw, 2021-01-22:
# Yes! \o/

ivankravets, 2021-01-24:
Sorry, I did a mistake. You need manually change the compatibility mode for your library using https://docs.platformio.org/en/latest/librarymanager/config.html#libcompatmode

The default mode is `soft`. See a related issue caused by recent changes https://github.com/platformio/platformio-core/issues/3811

I'll revert these changes.

/cc @mcspr

ivankravets, 2021-01-24:
I also added a note to docs https://github.com/platformio/platformio-docs/commit/89ee78e13f83b0f3347423e1494c6e139eb6b29b

commit f9384ded (2021-01-22): Fixed an issue when “strict” compatibility mode was not used for a library with custom “platforms” field in library.json manifest // Resolve #3806
commit bd4d3b91 (2021-01-24): Revert "lib_compat_mode" changes // Resolve #3811 Resolve #3806