platformio/platformio-core #3807: GDB-py from toolchain-gccarmnoneeabi can't find Python

MasterLufier, 2021-01-21:
### Configuration

**Operating system**: 
Ubuntu 20.04.1 LTS 

**PlatformIO Version** (`platformio --version`): 
PlatformIO Core, version 5.0.4

**Board**:
Nucleo-F429ZI

**GDB Version** (from gdb without python)
`.platformio/packages/toolchain-gccarmnoneeabi/bin/arm-none-eabi-gdb --version`
 #`
GNU gdb (xPack GNU Arm Embedded GCC, 64-bit) 8.3.0.20190709-git
`

### Description of problem
I split this issue from my [another one]( https://github.com/platformio/platform-ststm32/issues/478 ),  because it is not linked problems. arm-none-eabi-gdb-py from toolchain-gccarmnoneeabi can't find Python path

I manually install different version of gcc taked from [here](https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads/9-2019-q4-major) and it works fine.

`.platformio/packages/toolchain-gccarmnoneeabi-9/bin/arm-none-eabi-gdb-py --version`
`
GNU gdb (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 8.3.0.20190709-git
`

#### Steps to Reproduce
`$.platformio/packages/toolchain-gccarmnoneeabi/bin/arm-none-eabi-gdb-py --version`

## Output:
```
Could not find platform independent libraries <prefix>
Could not find platform dependent libraries <exec_prefix>
Consider setting $PYTHONHOME to <prefix>[:<exec_prefix>]
ImportError: No module named site
```

valeros, 2021-01-21:
Hi @MasterLufier ! Thanks for reporting. The `toolchain-gccarmnoneeabi` is an internal package and not meant to be used outside of PlatformIO ecosystem.

closed 2021-01-21 by valeros

MasterLufier, 2021-01-21:
@valeros sorry, you don't wrote about it [there](https://github.com/platformio/platform-ststm32/issues/478)