platformio/platformio-core #3808: ccls support is broken

ettom, 2021-01-22:
What kind of issue is this?

- [X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

### Configuration

**PlatformIO Version** (`platformio --version`):
PlatformIO Core, version 5.0.4
### Description of problem
Starting with 5.0.4, ccls support is broken for `--ide vim` (possibly others), no language server related features seem to be working. It seems that the problem is the first line of the generated `.ccls` file. It's now the full path to the used compiler, however according to [ccls documentation](https://github.com/MaskRay/ccls/wiki/Project-Setup#compiler-driver), it should contain the name of a compiler driver - either `clang` or `clang++`. This was changed in 32181d1bd298c368993bd1264d7e595163fb70a4.

ivankravets, 2021-01-23:
Please re-test with `pio upgrade --dev`. Does it work now?

closed 2021-01-23 by ivankravets

ettom, 2021-01-23:
Yes, it works now. Thank you!

ivankravets, 2021-01-23:
Thanks that reported!

Happy coding with PlatformIO!

commit e695e30a (2021-01-23): Fixed an issue with compiler driver for ".ccls" language server // Resolve #3808