platformio/platformio-core #3810: nested library dependency issue

jcw, 2021-01-23:
- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

This issue may be related to the previous one I reported: #3806.
This is with PIO 5.0.5b4, as required for #3806.

I've moved an external library dependency from the main `platformio.ini` file into a `library.json` file, because the main application no longer depends on that library (called `JeeH`). IOW, I removed `lib_deps = JeeH` from the main `platformio.ini` file, and added this to `lib/arch-stm32/library.json`:

    "dependencies": { "JeeH": "*" }

But this fails to identify the dependency from `<arch-stm32>` to `<JeeH> 1.15`.

It's easy to replicate, if you download the whole project and build it (it's a small repo, no other dependencies):

    git clone https://git.jeelabs.org/jcw/monty.git
    cd monty
    pio run -t size

The expected dependency output is:
```
Dependency Graph
|-- <arch-stm32>
|   |-- <JeeH> 1.15
|   |-- <monty>
|-- <monty>
```

But what I see, is:
```
Dependency Graph
|-- <arch-stm32>
|   |-- <monty>
|-- <monty>
```

Which leads to: "lib/arch-stm32/stm32.cpp:4:10: fatal error: jee.h: No such file or directory".

(I did not have this issue at first, because I still had `PLATFORMIO_LIB_EXTRA_DIRS` set to a local copy of JeeH)

jcw, 2021-01-24:
On a related note: removing `lib_deps = JeeH` from `platformio.ini` and then doing a build will _continue_ to find the library, even after a `pio run -t clean`. But after `rm -rf .pio` the build will fail, due to the now-missing library. This makes it tricky to nail down the issue - I found it surprising that "pio run -t clean" and "rm -rf .pio" behave differently.

ivankravets, 2021-01-24:
1. Please re-test with the latest 5.0.5b5. We reverted back all changes.
2. Please provide a simple project to reproduce this issue.

jcw, 2021-01-24:
1. done
2. the above project is super-simple: a few subdirs in `lib/`, one of them with an external dependency

Please try the above 3 commands as described. The behaviour is still the same with 5.0.5b5, the dependency in `lib/arch-stm32/library.json` is not picked up, causing the build to fail on the missing `jee.h` header. For comparison, the other variant works: `pio run -e native`.

ivankravets, 2021-01-26:
Duplicate of https://github.com/platformio/platformio-core/issues/2573

closed 2021-01-26 by ivankravets