platformio/platformio-core #3823: provide a BUILD_COMMIT flag in the platformio.ini

jerabaul29, 2021-01-27:
I really like the ```BUILD_TIMESTAMP``` flag that can be set in the ```platformio.ini``` file already now:

```
build_flags =
    -DBUILD_TIMESTAMP=$UNIX_TIME
```

**Would it be possible to similarly get a ```BUILD_COMMIT``` flag** that would return:

- None if the project root is not a git repo
- the commit ID otherwise

At present I implement this by doing, when I am in a git repo:

```
build_flags =
    !echo "-DBUILD_COMMIT="$(git rev-parse --verify HEAD)
```

But I suppose this is not portable (I am on linux, but what for windows users of my project?), and I do not have the None logics, as it would start to be a bit long command.

jerabaul29, 2021-01-27:
PS: my idea is to allow to make such things portable between linux / windows etc, i.e. this is the typical use case; posting here so that people googling "platformio build flag commit version" and similar also get the help about the stringify tricks by the same occasion :) :

```cpp
#include <Arduino.h>

#define STRINGIFY2(X) #X
#define STRINGIFY(X) STRINGIFY2(X)

void setup() {
  Serial.begin(9600);
  Serial.println(F("booted"));
  delay(100);

  Serial.print(F("compiled at UNIX time: "));
  Serial.println(F(STRINGIFY(BUILD_TIMESTAMP)));
  delay(100);

  Serial.print(F("built from commit: "));
  Serial.println(F(STRINGIFY(BUILD_COMMIT)));
  delay(100);

  Serial.println(F("done setup"));
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

ivankravets, 2021-01-27:
There is an example of how to make it portable https://docs.platformio.org/en/latest/projectconf/section_env_build.html#dynamic-build-flags

See `Use Case: Create a “PIO_SRC_REV” macro with the latest Git revision` section

Or, you can automate with https://docs.platformio.org/en/latest/projectconf/advanced_scripting.html#before-pre-and-after-post-actions

closed 2021-01-27 by ivankravets