platformio/platformio-core #3824: Platformio overwrites .vscode/launch.json, breaking official VSCode documentation, preventing project-specific settings

ul-gh, 2021-01-28:
Hi,

as already described by johnjiabinzhang in pio community posting [VSCode IDE automatically generate of launch.json every time](https://community.platformio.org/t/vscode-ide-automatically-generate-of-launch-json-every-time/3305), the behavior breaks the [official documented way](https://code.visualstudio.com/docs/editor/debugging#_platformspecific-properties) how VSCode is configured.

Worse, in VSCode "Run" panel, there is also a GUI for adding custom settings to the project-specific launch.json, which are then - unexpectedly - overwritten when VSCode with platformio is restarted.

While I admit there is a warning in launch.json such as "do not edit this file", the provided links to the PIO documentaion do not help when custom launch.json tasks are needed instead of the PIO unified debugger.

Modifying the template file in PIO user config directory (.platformio/penv/lib/python3.8/site-packages/platformio/ide/tpls/vscode/.vscode) is also not a solution as this place is not shared with the project and as this is also not a place for permanent user or project-specific config.

My suggestion would be to only generate the .vscode/launch.json in case it does not yet exist and otherwise either ask the user to merge the settings or just append the PIO specific JSON objects to the launch.json.

Please let me know if I can be of any help implementing this.. however I have only rudimentary experience with the PIO project structure..

Regards, Ulrich

ul-gh, 2021-01-28:
A work-around for this is to add platform-specific launch tasks to the global (per user) configuration as described here.

Use case for me is adding to VSCode with platformio the [vue.js debugging and dev tooling as described here](https://vuejs.org/v2/cookbook/debugging-in-vscode.html#Launching-the-Application-from-VS-Code) - and this is a per-project setup.

ivankravets, 2021-01-28:
Why do you need Vue.JS debugger for the embedded project?

ul-gh, 2021-01-28:
Hi Ivan, and thanks for the reply.

I built an AJAX web application front-end and back-end running on an embedded web server on the ESP32.
This is a remote control for a hardware application. It uses server-side push messages to update the live view of the remote control view:

[ESP-AJAX-Lab](https://github.com/ul-gh/esp_ajax_if)

You could do this without an application framework, and in fact the current version, which is not ready for production, does not even have the vue.js application running, but has some custom JavaScript for this. But the modular architecture of the framework is supposed to help when more features are added, and also you have pre-built components for visualisation of streaming measurement data from the microcontroller etc.

iocafe, 2021-01-29:
Hi Ivan,
I have same challenge as ul-gh, overwritten "launch.json". I write libraries which compile to several platforms, like micro-controller, linux, windows, etc. PC builds use cmake, while micro-controller builds use platform IO. Since I always have CMakeLists.txt and platformio.ini in same project root directory, it is not possible to use Visual Studio Code for cmake builds. Workaround has been to use qtcreator (I quess any other IDE but visual studio code) for these. It would be nice to use same Visual Studio Code trough. I guess it would be possible just overwrite ""type": "platformio-*" launch configurations, instead of overwriting the whole while. Stil,l I am not sure if this is common enough issue to justify the effort. It might also create user issues if both user and platformIO are allowed to modify the same file. For me this is just minor inconvenience. 

Best regards, Pekka

yunuscukran, 2021-01-30:
Hi, this is an inconvenience on my end as well. I'd love to be able to change this. Could you point out where/how in the code the launch.json rewrite action happens? @ul-gh mentioned the template being within the /penv folder, but I'm also seeing a launch.json template in the package.json of `platformio-vscode-ide`

Thanks,
Yunus

ivankravets, 2021-01-30:
Sorry for the issue. We will fix it in 5.1.1 soon.

closed 2021-03-10 by ivankravets

AndrewCapon, 2023-01-03:
Hi Guys,

I'm on:
Core: 6.1.5
Home: 3.4.3

My launch.json is getting overwritten.

As far as I can see here this shouldn't be happening?

ivankravets, 2023-01-03:
If you have broken JSON, it will be overwritten. https://jsonformatter.curiousconcept.com/

AndrewCapon, 2023-01-03:
Thanks, not sure what broken JSON is. Whitespace should have no impact?

ivankravets, 2023-01-03:
Please share the raw JSON here.

AndrewCapon, 2023-01-03:
```
     {
        "name": "MyDebug",
        "type": "cppdbg",
        "request": "launch",
        "miDebuggerPath": "/Applications/ARM/bin/arm-none-eabi-gdb",
        "miDebuggerArgs": "--baud=115200  /Users/bob/Development/teensy/TeensyTest/.pio/build/teensy41debug/firmware.elf",
        "MIMode": "gdb",
        "targetArchitecture": "arm",
        "program": "/Users/bob/Development/teensy/TeensyTest/.pio/build/teensy41debug/firmware.elf",
        "launchCompleteCommand": "None",
        "filterStderr": false,
        "filterStdout": false,
        "externalConsole": false,
        "cwd": "${workspaceRoot}",
        "setupCommands": [
          {"text": "set target-async off"},
          {"text": "target extended-remote /dev/cu.usbserial-14410"},   
          ]
      }
```

That is what is removed, Perfectly valid JSON.

Putting it through the pretty printer does stop it being removed though.

AndrewCapon, 2023-01-03:
Also the only reason I am doing it like this is I can't work out how to get a normal `pic debug' to pass through `--baud` on the gdb command line, `debug_speed = 115200` has no effect at this level. Is there a way to do this?

ivankravets, 2023-01-04:
Yes, you need https://docs.platformio.org/en/latest/projectconf/sections/env/options/debug/debug_extra_cmds.html

- https://sourceware.org/gdb/onlinedocs/gdb/Remote-Configuration.html

```ini
debug_extra_cmds = 
   set serial baud 115200
```

Does it work? If does not, please try `set remotebaud 115200`.

AndrewCapon, 2023-01-04:
Hi @ivankravets 

Thanks very much for the info, unfortunately neither of these worked with `debug_extra_cmds=`. I'm guessing as they would be run after the `target extended-remote $DEBUG_PORT` which is where it hangs.

I had tried `set remotebaud 115200` in the `debug_init_cmds =` before and that didn't seem to be supported by the version of arm-none-eabi-gdb in PlatformIO.

I just tried `set serial baud 115200` before `target extended-remote $DEBUG_PORT` and that works perfectly:

```
debug_port = /dev/cu.usbserial-14410         # Teensy, computer and OS dependent -- NEEDS TO BE ADJUSTED
debug_tool = custom
debug_load_mode = manual
debug_server = 
debug_init_cmds = 
  set serial baud 115200
  target extended-remote $DEBUG_PORT
  $INIT_BREAK
  define pio_reset_run_target
  interrupt
  tbreak loop
  continue
  end
  define pio_restart_target
  echo Restart is undefined for now.
  end

debug_init_break =
```

Thanks very much for the help with this it is much appreciated.

Cheers

Andy

dzid26, 2023-07-21:
I think this should be reopened.
- `launch.json` gets overwritten after doing even the slightest change to the autogenerated file,
- Adding non-platformio configs to the end of file gets it overwritten as well
- Still, there should be an option to disable updating 'launch.json' automatically.

One could use relative paths and distribute the file with the repository. E.g. of relative paths:
```json
{
    "type": "platformio-debug",
    "request": "launch",
    "name": "PIO Debug",
    "executable": "${workspaceFolder}/semihosting/.pio/build/nucleo_l452re/firmware.elf",
    "projectEnvName": "nucleo_l452re",
    "toolchainBinDir": "${userHome}/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/bin",
    "internalConsoleOptions": "openOnSessionStart",
    "svdPath": "${userHome}/.platformio/platforms/ststm32/misc/svd/STM32L4x2.svd",
    "preLaunchTask": {
        "type": "PlatformIO",
        "task": "Pre-Debug"
    }
},
```

leon0399, 2023-09-06:
+1, I tried a lot of different approaches. Adding anything other than autogenerated tasks to `launch.json` causes overrides. I would love to see an option to disable `launch.json` updates

leon0399, 2023-09-06:
+1, I tried a lot of different approaches. Adding anything other than autogenerated tasks to `launch.json` causes overrides. @ivankravets I would love to see an option to disable `launch.json` updates

ivankravets, 2023-09-07:
@leon0399 Please validate your JSON via https://jsonformatter.curiousconcept.com/ before saving.

miathedev, 2023-10-03:
Have the same issue. Want to simply debug two devices at once. So i added a secondary entry for debugging a secondary environment.

Its absolutely unusable for this use case as the launch.json is always overwritten.
However i am supposed to debug two devices at once using platformio.

(My launch.json is valid json)

AndrewCapon, 2023-10-14:
Same here (again) loosing my cortex-debug sections, how can we stop platform doing this, I don't event want to use the platform debugger!

ivankravets, 2023-10-15:
Please ensure that your JSON is valid using https://jsonformatter.curiousconcept.com/

AndrewCapon, 2023-10-16:
Of course it is valid!

ivankravets, 2023-10-17:
Could you share the RAW `launch.json` somewhere?

AnthonyDTU, 2023-12-19:
I have a problem with this as well, and I think I have isolated the issue for my specific case.

My goal is to create a "***PIO Attach***" launch configuration. The way I am trying to achieve this, is to create an env in the platformio.ini file, so that I have both a debug and an attach env, which looks like the following: 

```code
[platformio]
default_envs = esp32_debug

[env]
platform = espressif32@6.4.0 
board = esp32dev
framework = espidf 
monitor_speed = 115200
upload_speed = 921000
upload_protocol = esp-prog

[env:esp32_debug]
build_type = debug
debug_load_mode = modified
debug_tool = esp-prog
debug_port = :3333
debug_init_break = tbreak setup

[env:esp32_attach]
build_type = debug
debug_tool = esp-prog
debug_port = :3333
debug_load_mode = manual
debug_init_break = tbreak setup
debug_init_cmds =
  target extended-remote $DEBUG_PORT
  $INIT_BREAK
  monitor halt
  monitor init
  monitor resume
```

Since my default env is ***esp32_debug***, the launch.json file is generated with the following configurations:

```json
{
    "version": "0.2.0",
    "configurations": [
        {
            "type": "platformio-debug",
            "request": "launch",
            "name": "PIO Debug",
            "executable": "/workspace/.pio/build/esp32_debug/firmware.elf",
            "projectEnvName": "esp32_debug",
            "toolchainBinDir": "/root/.platformio/packages/toolchain-xtensa-esp32/bin",
            "internalConsoleOptions": "openOnSessionStart",
            "preLaunchTask": {
                "type": "PlatformIO",
                "task": "Pre-Debug"
            }
        },
        {
            "type": "platformio-debug",
            "request": "launch",
            "name": "PIO Debug (skip Pre-Debug)",
            "executable": "/workspace/.pio/build/esp32_debug/firmware.elf",
            "projectEnvName": "esp32_debug",
            "toolchainBinDir": "/root/.platformio/packages/toolchain-xtensa-esp32/bin",
            "internalConsoleOptions": "openOnSessionStart"
        },
        {
            "type": "platformio-debug",
            "request": "launch",
            "name": "PIO Debug (without uploading)",
            "executable": "/workspace/.pio/build/esp32_debug/firmware.elf",
            "projectEnvName": "esp32_debug",
            "toolchainBinDir": "/root/.platformio/packages/toolchain-xtensa-esp32/bin",
            "internalConsoleOptions": "openOnSessionStart",
            "loadMode": "manual"
        }
    ]
}
```

I would then like to add a configuration to the launch.json file, that looks like this:

```json
{
    "type": "platformio-debug",
    "request": "launch",
    "name": "PIO Attach",
    "executable": "/workspace/.pio/build/esp32_attach/firmware.elf",
    "projectEnvName": "esp32_attach",
    "toolchainBinDir": "/root/.platformio/packages/toolchain-xtensa-esp32/bin",
    "internalConsoleOptions": "openOnSessionStart",
    "preLaunchTask": {
        "type": "PlatformIO",
        "task": "Pre-Debug"
    }
}
```

This all work. If I don't want to actively debug, I can upload the firmware via the PlatformIO pane, with the *Upload* task. If I then later want to attach the debugger, I can select ***PIO Attach*** in the debug pane, and a debug session will start without restarting the target device. 

However, when the platformio.ini file is updated/saved, the launch.json files is overwritten (as this thread is all about). I believe that this is handled by this file: https://github.com/platformio/platformio-core/blob/develop/platformio/project/integration/tpls/vscode/.vscode/launch.json.tpl.

In this file, the code which is responsible for "saving" external configurations, looks for all configurations ***not*** of the type: "platformio-debug" (line 65). However, my ***PIO Attach*** configuration is of that type, and it therefore does not get saved, resulting in it getting overwritten. 

Is there a possibility to only look for the three launch configurations: ***PIO Debug***, ***PIO Debug (skip Pre-Debug)*** and ***PIO Debug (without uploading)***?

ivankravets, 2023-12-19:
@AnthonyDTU , could you file a new issue at https://github.com/platformio/platformio-core/issues ? We will implement this in the nearest release.