platformio/platformio-core #3827: "The command line is too long" when linking

ldursw, 2021-01-30:
What kind of issue is this?

- [X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: Windows 10 x64

**PlatformIO Version** (`platformio --version`): `PlatformIO Core, version 5.1.0`

### Description of problem

After upgrading from 5.0.4 to 5.1.0 the build fails with `The command line is too long` when linking `libSodium.a`. This regression may have been caused by e2906e3 when fixing #3792. Downgrading to 5.0.4 fixes the issue.

### Additional info

Output of `pio run -v` on 5.0.4 and 5.1.0
[log510.txt](https://github.com/platformio/platformio-core/files/5898660/log510.txt) line 229
[log504.txt](https://github.com/platformio/platformio-core/files/5898655/log504.txt) line 190
[longcmd-5c2b0877ea68fe2ac65a2da8ceb030c0.txt](https://github.com/platformio/platformio-core/files/5898666/longcmd-5c2b0877ea68fe2ac65a2da8ceb030c0.txt)
[longcmd-5b96ba999a3e908e52610e59e8d95b4e.txt](https://github.com/platformio/platformio-core/files/5898669/longcmd-5b96ba999a3e908e52610e59e8d95b4e.txt)

ldursw, 2021-01-31:
In this line
https://github.com/platformio/platformio-core/blob/698594525f06c4aa1f30bfa4f54cca91dd3df4ae/platformio/builder/tools/piomaxlen.py#L30
If I change `256` to `300` the project compiles successfully.

ivankravets, 2021-02-01:
Thanks for the report! Please re-test with `pio upgrade --dev`.

closed 2021-02-01 by ivankravets

ldursw, 2021-02-01:
It is working now. Thanks for fixing the issue.

commit e63019c4 (2021-02-01): Fixed a "The command line is too long" issue with a linking process on Windows // Resolve #3827