platformio/platformio-core #3828: Package publish rejects the package

mcspr, 2021-01-30:
### Configuration

**Operating system**:

Windows 20H2

**PlatformIO Version** (`platformio --version`):

PlatformIO Core, version 5.1.0

### Description of problem

Following up on the https://github.com/platformio/platformio-core/issues/3776

After downloading release files from:
https://github.com/earlephilhower/esp-quick-toolchain/releases

```
> dir | foreach {echo $_.Name}
aarch64-linux-gnu.xtensa-lx106-elf-48f7b08.210128.tar.gz
x86_64-apple-darwin14.xtensa-lx106-elf-48f7b08.210128.tar.gz
x86_64-linux-gnu.xtensa-lx106-elf-48f7b08.210128.tar.gz
x86_64-w64-mingw32.xtensa-lx106-elf-48f7b08.210128.zip
> dir | foreach {pio package publish $_.Name}
Error: Invalid request data for pkgtype -> 'Must be one of: library, platform, tool'
Error: Invalid request data for pkgtype -> 'Must be one of: library, platform, tool'

Aborted!
>
```

I have not tried to 'repackage', like before, since I assumed this method of directly uploading .tar.gz or .zip should work now.

### Additional info

`package.json` is generated here:

https://github.com/earlephilhower/esp-quick-toolchain/blob/48f7b08d1679365ea462675dd3bfb9d83c9bf11a/Makefile#L280-L290

maxgerhardt, 2021-01-30:
Judging from 

https://github.com/platformio/platformio-core/blob/c623a6aaccf5dfc77e58a3d7610218bc842c1499/platformio/clients/registry.py#L44-L46

and

https://github.com/platformio/platformio-core/blob/c623a6aaccf5dfc77e58a3d7610218bc842c1499/platformio/package/meta.py#L54-L66

and

https://github.com/platformio/platformio-core/blob/c623a6aaccf5dfc77e58a3d7610218bc842c1499/platformio/package/manifest/parser.py#L34-L39

The logic whereit fails is that it can't decide whether it's a library, platform, or tool. And that is decided by the package having a `package.json` (--> tool), `platform.json` (--> platform) or library.json/library.properties/module.json (--> library)

The e.g. https://github.com/earlephilhower/esp-quick-toolchain/releases/download/3.0.0-newlib4.0.0-gnu22/x86_64-linux-gnu.xtensa-lx106-elf-48f7b08.210128.tar.gz file has 

[image]

.. as a direct first thing a folder, not the files directly. Only in there

[image]

Is the file we're looking for.

So I **think** the bug here is that the archive file contains a folder which contains all the files, but it should contain all the files directly. Bug in the generation logic.

mcspr, 2021-01-30:
Afaik, archive structure is made to be compatible with ArduinoIDE's downloader, as these archives are also used as-is here when tagged version is released:
https://github.com/esp8266/Arduino/blob/master/package/package_esp8266com_index.template.json
(as development versions needs manual sync through the use of tools/get.py)

So, the gist is:
- PackageType.from_archive is not raising exception and is not checked by the caller whether it is None
- `"/v3/packages/mcspr/None" URL is valid on the server side?
- from_archive should be able to determine package root like it already does with `platform_packages = ...` installation?
https://github.com/platformio/platformio-core/blob/698594525f06c4aa1f30bfa4f54cca91dd3df4ae/platformio/package/manager/base.py#L128-L134

maxgerhardt, 2021-01-30:
Oh well handy that such a `find_pkg_root` function already exists, then I'm sure it can be integrated here rather easily (if the bug is as I'm seeing it that it doesn't walk the directory structure). CC @ivankravets

ivankravets, 2021-02-01:
So, and where is the issue? I'm not sure that I understand what is going wrong. There is PlatformIo Registry with its own rules, and there 3rd party archives that are not compatible. Is it correct?

mcspr, 2021-02-01:
Package archive above do not contain `package.json` in the 'root' of the archive, but `pio package publish` still tries to upload it without raising an error until the very last moment. The very last message describes just that - allow it to discover the root, just like the normal installation does when we do `pio package pack`

ivankravets, 2021-02-01:
This is normal behavior. It is a popular practice to put files in the interim folder to avoid issues with ZIP archive that has issues with putting files in the root. It is common practice to ship libraries in ZIP archives and put `library.json` in the "nested" folder.

mcspr, 2021-02-01:
...so shouldn't we be able to publish an archive with `<root>/somedir/package.json`?

Hence the comments about the `PackageType.from_archive`. It only checks for `<root>/package.json`, thus causing the publish to use `/v3/packages/mcspr/None` as an URL instead of `/v3/packages/mcspr/tool` (i.e. see the 'map' just above).

ivankravets, 2021-02-01:
> Hence the comments about the PackageType.from_archive. It only checks for <root>/package.json

Aha, now I understand. Yes, it seems like an issue. Thanks for the report! We will think about providing lightweight solution for this.

commit 1269ce06 (2022-04-08): Improved detection of a package type from the tarball archive // Resolve #3828
closed 2022-04-08 by ivankravets