platformio/platformio-core #3829: 5.1.0 breaks STM8 framework

paulo-fernando-silva, 2021-01-31:
First of all, thank you so much for the vscode plugin. It's really great. I just started using it recently, and was TI's, ST's, and the Arduino IDEs, when I came across platformio. It was an easy choice. Great work. I just need to figure out how to make it less black box because I can't see what the plugin is doing. That leads me to this bug.
### Configuration

**Operating system**:
Linux Mint 20.4 or so

**PlatformIO Version** (`platformio --version`):
5.1.0

### Description of problem
After the automatic update even a simple blink doesn't build anymore. 
See build_on_510.txt for the failing build output and build_on_504.txt for the successful build output
[build_on_504.txt](https://github.com/platformio/platformio-core/files/5899984/build_on_504.txt)
[build_on_510.txt](https://github.com/platformio/platformio-core/files/5899985/build_on_510.txt)

Fortunately I found that I can downgrade the version here:
https://community.platformio.org/t/downgrade-platformio/217
with the command:
 
 ~/.platformio/penv/bin/pip install -U "platformio<5.1.0"

Now it's working again.

#### Steps to Reproduce

1. Install or update  "platformio == 5.1.0"
2. create an stm8 sample, for example the native blink from platformio examples
3. build...

### Actual Results
see build_on_510.txt

### Expected Results
see build_on_504.txt

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
It's whatever platformio creates by default

**Source file to reproduce issue:**
platformio's stm8 native blink example seems to reproduce this issue.
might reproduce with other projects too.

### Additional info

maxgerhardt, 2021-02-01:
Issue in platform: https://github.com/platformio/platform-ststm8/issues/27

valeros, 2021-02-01:
Closed as duplicate https://github.com/platformio/platform-ststm8/issues/27

closed 2021-02-01 by valeros