platformio/platformio-core #3831: Disabled telemetry is ignored

lfuelling, 2021-02-03:
Hi,

I've disabled telemetry and crash reporting in VSCode and after installing the PlatformIO IDE into VSCode, VSCode suddenly "does requests" (according to the firewall) to Google Analytics. 

I've never seen an option to accept this or even opt-out and the only way to disable it seems to be blocking VSCode to access `google-analytics.com` this feels very scammy.

ivankravets, 2021-02-03:
Thanks, re-opened. It seems like analytics in PIO Home. We will fix it soon.

ivankravets, 2021-03-09:
The telemetry is disabled on the PlatformIO Home side, see https://github.com/platformio/platformio-home/blob/develop/app/store/telemetry.js#L49

Could you provide more details on how to re[produce this issue? Could try to disable the PlatformIO IDE extension for VSCode? Do you still see requests to GA?

ivankravets, 2021-03-15:
Please reopen if you still need help

closed 2021-03-15 by ivankravets

steve-bate, 2025-08-03:
> The telemetry is disabled on the PlatformIO Home side, see https://github.com/platformio/platformio-home/blob/develop/app/store/telemetry.js#L49
>
> Could you provide more details on how to re[produce [...]

I'm seeing this too. This is what's causing the extension timeout failure when trying open PIO Home on Windows 11. My guess it's because I have a DNS blocker on my LAN (Pi-Hole) to prevent this kind of behavior.
```
    at ClientRequest.<anonymous> (c:\Users\Steve\.vscode\extensions\platformio.platformio-ide-3.3.4-win32-x64\node_modules\got\dist\source\core\index.js:970:111)
    at Object.onceWrapper (node:events:633:26)
    at ClientRequest.emit (node:events:530:35)
    at ClientRequest.emit (node:domain:489:12)
    at ClientRequest.origin.emit (c:\Users\Steve\.vscode\extensions\platformio.platformio-ide-3.3.4-win32-x64\node_modules\@szmarczak\http-timer\dist\source\index.js:43:20)
    at emitErrorEvent (node:_http_client:104:11)
    at TLSSocket.socketErrorListener (node:_http_client:518:5)
    at TLSSocket.emit (node:events:518:28)
    at TLSSocket.emit (node:domain:489:12)
    at emitErrorNT (node:internal/streams/destroy:170:8)
    at emitErrorCloseNT (node:internal/streams/destroy:129:3)
    at processTicksAndRejections (node:internal/process/task_queues:90:21)
    at GetAddrInfoReqWrap.onlookupall [as oncomplete] (node:dns:122:26)
```
Unfortunately, I don't have the time to track down the precise place where the telemetry is being attempted.