platformio/platformio-core #3832: Remote monitor unable to use control characters

chippey, 2021-02-03:
When trying to use the remote monitor / miniterm, I'm unable to do things like hit Ctl + T to bring up the menu, or toggle various lines with Ctl + R or Ctl + D
------------------------------------------------------------------

### Configuration

**Operating system**:  Linux (Pop!_OS 20.04 LTS

**PlatformIO Version** (`platformio --version`): PlatformIO Core, version 5.1.0

### Description of problem

#### Steps to Reproduce

1. Install pio via Python on remote host and local
2. Initiate pio monitor with : `pio remote device monitor`
3. Hit Ctl + T, or any other Ctl combo other than Ctl + C

### Actual Results
```
chippey@acornduex:~$ pio remote device monitor
Starting Serial Monitor on nvidia-nano:/dev/ttyACM0
--- Available filters and text transformations: colorize, debug, default, direct, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at http://bit.ly/pio-monitor-filters
--- Miniterm on socket://localhost:40225  9600,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
Unhandled Error
Traceback (most recent call last):
  File "/home/chippey/.platformio/packages/contrib-pysite/twisted/python/log.py", line 103, in callWithLogger
    return callWithContext({"system": lp}, func, *args, **kw)
  File "/home/chippey/.platformio/packages/contrib-pysite/twisted/python/log.py", line 86, in callWithContext
    return context.call({ILogContext: newCtx}, func, *args, **kw)
  File "/home/chippey/.platformio/packages/contrib-pysite/twisted/python/context.py", line 122, in callWithContext
    return self.currentContext().callWithContext(ctx, func, *args, **kw)
  File "/home/chippey/.platformio/packages/contrib-pysite/twisted/python/context.py", line 85, in callWithContext
    return func(*args,**kw)
--- <exception caught here> ---
  File "/home/chippey/.platformio/packages/contrib-pysite/twisted/internet/posixbase.py", line 614, in _doReadOrWrite
    why = selectable.doRead()
  File "/home/chippey/.platformio/packages/contrib-pysite/twisted/internet/tcp.py", line 243, in doRead
    return self._dataReceived(data)
  File "/home/chippey/.platformio/packages/contrib-pysite/twisted/internet/tcp.py", line 249, in _dataReceived
    rval = self.protocol.dataReceived(data)
  File "/home/chippey/.local/lib/python3.7/site-packages/platformio/commands/remote/client/device_monitor.py", line 33, in dataReceived
    self.factory.send_to_server(data)
  File "/home/chippey/.local/lib/python3.7/site-packages/platformio/commands/remote/client/device_monitor.py", line 68, in send_to_server
    self.cdm.acwrite_data(data)
  File "/home/chippey/.local/lib/python3.7/site-packages/platformio/commands/remote/client/device_monitor.py", line 218, in acwrite_data
    self._acwrite_buffer += data
builtins.TypeError: can only concatenate str (not "bytes") to str

--- exit ---
Exception in thread rx:
Traceback (most recent call last):
  File "/home/chippey/.local/lib/python3.7/site-packages/serial/urlhandler/protocol_socket.py", line 171, in read
    raise SerialException('socket disconnected')
serial.serialutil.SerialException: socket disconnected

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 926, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/home/chippey/.local/lib/python3.7/site-packages/serial/tools/miniterm.py", line 499, in reader
    data = self.serial.read(self.serial.in_waiting or 1)
  File "/home/chippey/.local/lib/python3.7/site-packages/serial/urlhandler/protocol_socket.py", line 178, in read
    raise SerialException('read failed: {}'.format(e))
serial.serialutil.SerialException: read failed: socket disconnected
```

### Expected Results
Get the remote monitor menu, or other functions that are invoked with Ctl

mulles, 2021-05-04:
You might use the baud rate 115200 as suggest in https://docs.nvidia.com/jetson/archives/l4t-archived/l4t-321/index.html#page/Tegra%2520Linux%2520Driver%2520Package%2520Development%2520Guide%2Fsoftware_features_jetson_nano.html%23wwpID0E0DB0HA
`pio remote device monitor  --baud 115200`

Moreover I have a similar problem with the combo Raspberrypi3B+ and STM Nucleo-64 (USB to Serial): 

In my case:   Ctrl + T followed by  Ctl + H works to get help. And   Ctrl + T followed by  Ctl + I to get information: 

When connecting directly to my laptop (PopOS 20.10) with `pio device monitor  --echo` everythings works as expected. Echo is used so I see the commands I typing on the serial console.

mulles, 2021-05-05:
@chippey  A workarround would be to run `ssh -t  username@host pio device monitor -b 115200` in case you got ssh access to nvidia-nano. ` -t` stand for Terminal 
If you enable SSH Reverse tunnel you would even be able to access the nvidia-nano outside of a router without NAT: 
In that case the command would be: `ssh -t -p <port> <pi@lsserver.uber.space> pio device monitor -b 115200`, where `port` the reverse ssh port.

commit f1638c9c (2021-11-05): Fixed an issue when PIO Remote device monitor crashes on the first keypress // Resolve #3832