platformio/platformio-core #3835: How to download packages manually?

Gianthard-cyh, 2021-02-06:
I am a user from China and the speed of downloading packages is extremely low.
I've found the comment from @jiladahe1997 in #3425 :
>If it's slow with downloading platformio packages (like stm32HAL library, gcctools, which platformio will download automatically after you create a new project).
>you could download it manually here:http://dl.bintray.com/platformio/ . And then just copy them to .platformio\packages\ .

the speed of http://dl.bintray.com/platformio/ is high.
However, after I downloaded the tar.gz file and unzipped it, I started building and get this:
[image]
It seems that PIO didn't recognize the package and still tries to download it.
So how do I install the package manually from http://dl.bintray.com/platformio/?
[image]

Gianthard-cyh, 2021-02-06:
我是一个来自中国的用户。PIO自带的包管理器下载速度很慢。我在 #3425 中找到了 @jiladahe1997 的回复：
>If it's slow with downloading platformio packages (like stm32HAL library, gcctools, which platformio will download automatically after you create a new project).
you could download it manually here:http://dl.bintray.com/platformio/ . And then just copy them to .platformio\packages\

所以我从http://dl.bintray.com/platformio/ 下载了.tar.gz文件放在了.platformio\packages\目录下，但是我再次运行生成任务的时候，PIO的包管理器又开始下载我已经下好的包。
所以如何正确地手动安装PIO包？

ivankravets, 2021-02-10:
Could you try these links from our own CDNs?

- CDN1 => https://dl.registry.nm1.platformio.org/tools/c9/17/f2c97dab926bb74d65343a647d20b75450201aface6a1ab6071556a640a1/toolchain-xtensa-windows_amd64-2.100100.200706.tar.gz
- CDN2 => https://dl.registry.ns1.platformio.org/tools/c9/17/f2c97dab926bb74d65343a647d20b75450201aface6a1ab6071556a640a1/toolchain-xtensa-windows_amd64-2.100100.200706.tar.gz

So, where is a good speed?

Gianthard-cyh, 2021-02-10:
> Could you try these links from our own CDNs?
>
> * CDN1 => [...]

the first one. How do I change the CDN to it?

ivankravets, 2021-02-10:
We changed the priority for China on our side. So, please run `pio system prune -f` and try building project.

Does it download fast automatically now?

Gianthard-cyh, 2021-02-10:
> We changed the priority for China on our side. So, please run `pio system prune -f` and try building project.
> 
> Does it download fast automatically now?

[image]
It looks nice now. Thank you very much.

Gianthard-cyh, 2021-02-10:
> > We changed the priority for China on our side. So, please run `pio system prune -f` and try building project.
> > Does it download fast automatically now?
> 
> [image]
> It looks nice now. Thank you very much.

ivankravets, 2021-02-10:
Thanks for the report! Happy coding with PlatformIO!

Gianthard-cyh, 2021-03-25:
I'm sorry to say that, but the speed when I init a project for the first time is really low.
Have you changed the CDN when init a project for the first time to CDN 2?

ivankravets, 2021-03-25:
What is the speed for https://dl.registry.nm1.platformio.org/tools/c9/17/f2c97dab926bb74d65343a647d20b75450201aface6a1ab6071556a640a1/toolchain-xtensa-windows_amd64-2.100100.200706.tar.gz ?

Gianthard-cyh, 2021-03-26:
> What is the speed for https://dl.registry.nm1.platformio.org/tools/c9/17/f2c97dab926bb74d65343a647d20b75450201aface6a1ab6071556a640a1/toolchain-xtensa-windows_amd64-2.100100.200706.tar.gz ?

It's slow. About 30kB/s

ivankravets, 2021-03-26:
Do you have a Linux machine? Please provide an output of the next commands:
# 1
```
traceroute platformio.org
```

# 2

```
traceroute nm1.platformio.org
```

Gianthard-cyh, 2021-03-26:
> Do you have a Linux machine? Please provide an output of the next commands:
> 
> # 1
> ```
> traceroute platformio.org
> ```
> 
> # 2
> ```
> traceroute nm1.platformio.org
> ```

[image]
[image]

ivankravets, 2021-03-29:
How about a speed for this package?
- https://dl.registry.platformio.org/libraries/03/df/56123fae2b834c2b805c4c1b5e6d3382e38849c8b5b44a03f9b71b2d7680/LovyanGFX-0.3.10.tar.gz

Gianthard-cyh, 2021-04-03:
> How about a speed for this package?
> 
> * https://dl.registry.platformio.org/libraries/03/df/56123fae2b834c2b805c4c1b5e6d3382e38849c8b5b44a03f9b71b2d7680/LovyanGFX-0.3.10.tar.gz

about 1MB/s.

Gianthard-cyh, 2021-04-03:
However , the speeds of all the CDNs are swinging. They are different between machines ,even from time to time,

Gianthard-cyh, 2021-04-03:
> However , the speeds of all the CDNs are swinging. They are different between machines ,even from time to time,

maybe a speed test of usable CDNs is needed before downloading.

ivankravets, 2021-04-04:
What is the right solution for this case? Is it possible to contact your government and ask them to unblock our servers? I don't see another option here. We tried 3 different data centers and all are blocked and speed is limited.

ivankravets, 2021-04-04:
I also know that China government blocked https://pypi.org/. I don't have any words on how to explain this reason.

Gianthard-cyh, 2021-04-05:
> What is the right solution for this case? Is it possible to contact your government and ask them to unblock our servers? I don't see another option here. We tried 3 different data centers and all are blocked and speed is limited.

Thank you very much for your reply.
I'm so sorry that China government has blocked your servers. I do not have the right to ask the government to unblock it . Although the speeds are swinging, some of them have acceptable speed at a certain time. Set up a mirror server in China may be a long time goal. So Focus on the present, just choose the CDN with better speed automatically before every download. It will increase user experience.
Thank you for making PIO better!

ivankravets, 2021-04-05:
Could you advise us of a few VPS providers in China that work with non-Chinese customers?

Gianthard-cyh, 2021-04-05:
> Could you advise us of a few VPS providers in China that work with non-Chinese customers?

https://www.alibabacloud.com/
https://intl.cloud.tencent.com/

Thank you very much for your reply.
According to the law, websites in China must be registered, including VPS and CDN, and must use Chinese ID card, so you should find a person in charge in China to apply for the record. Otherwise, ports 80 and 443 will be blocked when resolving the domain name, and testpage184 will be displayed. So if you want to use a domain name and want to use port 80, the only solution is to choose a person in charge in China. If not, you can use another port. If you just want to build an image for storing files in China, a more affordable way is to use OSS (object storage service). It will provide a storage space, as well as a default domain name, so that you can access the above files. However, if you want to use other domain names, you still need to apply for the record. Also, servers in China Hongkong is an option, and they don't need record.

I'm still asking my friends for the providers they advise.

ivankravets, 2021-04-05:
Sounds crazy strange. Can't imagine how to resolve this issue in the 21st century.

ivankravets, 2021-04-05:
Sounds crazy strange. Can't imagine how to resolve this issue in the 21st century.

P.S: I tried to register at Tencent and their service works "well". See a picture below. It seems they block people outside China as well.

[image]

Gianthard-cyh, 2021-04-09:
> Sounds crazy strange. Can't imagine how to resolve this issue in the 21st century.
> 
> P.S: I tried to register at Tencent and their service works "well". See a picture below. It seems they block people outside China as well.
> 
> [image]

nodecache.com
This may be a good choice, which doesn't need record in China.

ivankravets, 2021-04-09:
@Gianthard-cyh do you mean if use direct IP on a non HTTP port your government will not block us?

Gianthard-cyh, 2021-04-09:
> @Gianthard-cyh do you mean if use direct IP on a non HTTP port your government will not block us?

yes, and some providers like https://www.nodecache.com/Pricing doesn't need record, which means you can use your domain and HTTP port when using their CDN product

Gianthard-cyh, 2021-04-09:
> @Gianthard-cyh do you mean if use direct IP on a non HTTP port your government will not block us?

Just for servers in China. This can avoid record. But for your servers outside China, They block IPs, which means that there is no way except setting up a server in China.

ivankravets, 2021-04-09:
Could you email me directly via me@ikravets.com? I'll provide some internal links based on different IPs and ports. We would be thankful if you test them. Thanks!

Gianthard-cyh, 2021-04-09:
> Could you email me directly via [me@ikravets.com](mailto:me@ikravets.com)? I'll provide some internal links based on different IPs and ports. We would be thankful if you test them. Thanks!

OK. Thank you!

ivankravets, 2021-04-19:
Should be fixed on our side. Please try to install any package. Does it work fast now?

closed 2021-04-19 by ivankravets

dkwiqmd, 2022-09-05:
so vpn

ivankravets, 2022-09-27:
@dkwiqmd , do experience ay problems with downloading packages?

developer-ken, 2022-10-07:
Is it possible to configure which mirror should pio use, just like `source.list` for apt? 
It would be great that we can mirror pio packages in our school lab and make it easier for students to learn and use pio. 

I now providing them with a compressed file of my `~/.platformio` folder, but when any package got an update, pio will download them again, at a very slow speed. Actually we can have a whole afternoon doing no work just downloading packages. It will be great if we can just have a local mirror, then we only need to download it once and fast distribute them to local computers.

radonyl, 2022-11-05:
According to https://docs.platformio.org/en/latest/core/userguide/pkg/cmd_install.html#local-folder
Since version 6.0, `pio pkg install` can install packages manually from directories, archive files or custom links.

@Gianthard-cyh 
a package will only be recongized when there is a `.piopm` file in its directory which is generate locally, so directly copy and paste downloaded packagse will not work. Use the `pio pkg install file://path_to_directory` instead.

developer-ken, 2023-01-22:
> Is it possible to configure which mirror should pio use, just like `source.list` for apt? It would be great that we can mirror pio packages in our school lab and make it easier for students to learn and use pio.
> 
> I now providing them with a compressed file of my `~/.platformio` folder, but when any package got an update, pio will download them again, at a very slow speed. Actually we can have a whole afternoon doing no work just downloading packages. It will be great if we can just have a local mirror, then we only need to download it once and fast distribute them to local computers.
> 
> I am in China, 'Inter'net is not quite international here and developers have to eventually figure out a way to got a proxy. We can hardly view github without a proxy. There is actually someone selling pio packages here because they are not so easy to get by official mirror. Tha's awful.

I am now running a reverse proxy in China with my registered domain, which works well for me and most students in the lab. Also I noticed the official server got a huge download speed increase now, installing packages is possible even without special tricks. I guess the problem is solved somehow.  
Can't understand why our gov block those download servers which contains absolutely legal contents.

nikito7, 2023-10-01:
Slow again. Europe

ivankravets, 2023-10-02:
@nikito7 , what is your country? What is your download speed for https://dl.registry.platformio.org/download/espressif/tool/toolchain-xtensa-esp32/12.2.0+20230208/toolchain-xtensa-esp32-darwin_arm64-12.2.0+20230208.tar.gz ?

nikito7, 2023-10-02:
> @nikito7 , what is your country? What is your download speed for https://dl.registry.platformio.org/download/espressif/tool/toolchain-xtensa-esp32/12.2.0+20230208/toolchain-xtensa-esp32-darwin_arm64-12.2.0+20230208.tar.gz ?

It's ok now. But it happen sometimes.

[image]

dizjis, 2024-05-17:
So slow, very slow. Russia
[image]
> https://dl.registry.platformio.org/download/espressif/tool/toolchain-xtensa-esp32/12.2.0+20230208/toolchain-xtensa-esp32-darwin_arm64-12.2.0+20230208.tar.gz ?

on this link the speed is 15kb/s 
[image]

>tracert platformio.org

[image]

ivankravets, 2024-05-17:
Could you try again?

dizjis, 2024-05-17:
> Could you try again?

its ok.Thank you very much