platformio/platformio-core #3836: Make package registry searchable in the CLI or GUI

maxgerhardt, 2021-02-10:
What kind of issue is this?
- [X] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

------------------------------------------------------------------

### Configuration

**Operating system**: Win10 x64

**PlatformIO Version** (`platformio --version`): `version 5.1.0`

### Description of Feature Request 

I've just noticed this morning that the bintray website (https://bintray.com/platformio/) doesn't contain the `tool-packages` sub-package in which I was able to search which packages and package versions are available in PlatformIO for e.g. usage in the `platform_packages` directive in the `platformio.ini`. 

Only using this website I was able to answer a lot of forum questions regarding "How can I upgrade my compiler" or "What package do need to use for an older mbed-os package" or "why is that package of that version missing" (when the package of another versin might be usable). 

The CLI doesn't support searching for generic packages, like `toolchain-atmelavr`, as far as I understand. `pio lib search toolchain-atmelavr` and `pio platform search toolchain-atmelavr` both don't list what `toolchain-atmelavr` package versions are available. I also don't see any other command [documented](https://docs.platformio.org/en/latest/core/userguide/index.html) that would do that.

What I would be expecting is that there is a general `pio package search` or `pio registry search` command that lets one enumerate available package sources / versions for a given domain, and e.g. even download them.

maxgerhardt, 2021-02-10:
Addendum: Per [community post](https://community.platformio.org/t/platformio-ssl-error-while-upgrading-using-vscode/16431) I remembered the API package and figured out that a GET query like

https://api.registry.platformio.org/v3/packages/platformio/tool/toolchain-atmelavr

would do what I want, but this is very cumbersome.

ivankravets, 2021-02-10:
Yes, we didn't expect this. [JFrog closed the Bintray project](https://jfrog.com/blog/into-the-sunset-bintray-jcenter-gocenter-and-chartcenter/). We used Bintray as one of our OSS mirrors. 

There is a new PlatformIO Trusted Registry that was launched in the summer of 2020. The only missing thing is a frontend for it. We will provide soon. Sorry for the issue. 

Meantime, you can use our API 3.0 => https://redocly.github.io/redoc/?url=https://api.registry.platformio.org/v3/openapi.yaml

Search example - https://api.registry.platformio.org/v3/packages?query=type:tool%20atmelavr

ivankravets, 2022-03-31:
The GUI and CLI has been implemented:

- https://registry.platformio.org/
- https://docs.platformio.org/en/latest/core/userguide/pkg/cmd_search.html

closed 2022-03-31 by ivankravets