platformio/platformio-core #3839: Feature Request: Add ${PROJECT_NAME} to Targets in generated CMakeLists.txt

MikeMitterer, 2021-02-11:
### Configuration

**Operating system**: OSX 10.15.7

**PlatformIO Version** (`platformio --version`): 5.1.0

### Description of problem
In a multi-project environment e.g.
```
├── CMakeLists.txt
├── Door
│   ├── CMakeLists.txt
│   ├── CMakeListsPrivate.txt
│   ├── CMakeListsUser.txt
├── LaserSensor
│   ├── CMakeLists.txt
│   ├── CMakeListsPrivate.txt
│   ├── CMakeListsUser.txt

```
you generated CMakeLists.txt in 'Door' and in 'LaserSensor' does not work anymore because
each of the files has as target "Production" and "Debug"

The solution is easy:
```
add_custom_target(
        ${PROJECT_NAME}_Production ALL
        COMMAND platformio -c clion run "$<$<NOT:$<CONFIG:All>>:-e${CMAKE_BUILD_TYPE}>"
        WORKING_DIRECTORY ${CMAKE_CURRENT_SOURCE_DIR}
)
```
But every time I have to regenerate the `pio init --ide clion` CMakeLists it throws out my changes

valeros, 2021-03-04:
Hi @MikeMitterer ! Thanks for the FR. Could you please explain in more details how it works in your case? I tried your solution and it seems that renaming that targets breaks PlatformIO configurations (at least on the latest CLion v2020.3.2):

[image]

MikeMitterer, 2021-03-05:
Hi @valeros the problem is `_Z_DUMMY_TARGET`. Just add 
`add_executable(${PROJECT_NAME}_Z_DUMMY_TARGET ${SRC_LIST})` and the problem is solved.

Here is a CMakeLists.txt:
```cmake
# !!! WARNING !!! AUTO-GENERATED FILE, PLEASE DO NOT MODIFY IT AND USE
# https://docs.platformio.org/page/projectconf/section_env_build.html#build-flags
#
# If you need to override existing CMake configuration or add extra,
# please create `CMakeListsUser.txt` in the root of project.
# The `CMakeListsUser.txt` will not be overwritten by PlatformIO.

cmake_minimum_required(VERSION 3.13)
set(CMAKE_SYSTEM_NAME Generic)
set(CMAKE_C_COMPILER_WORKS 1)
set(CMAKE_CXX_COMPILER_WORKS 1)

project("HelloESP32" C CXX)

include(CMakeListsPrivate.txt)

if(EXISTS ${CMAKE_CURRENT_SOURCE_DIR}/CMakeListsUser.txt)
include(CMakeListsUser.txt)
endif()

add_custom_target(
    ${PROJECT_NAME}_Production ALL
    COMMAND platformio -c clion run "$<$<NOT:$<CONFIG:All>>:-e${CMAKE_BUILD_TYPE}>"
    WORKING_DIRECTORY ${CMAKE_CURRENT_SOURCE_DIR}
)

add_custom_target(
    ${PROJECT_NAME}_Debug ALL
    COMMAND platformio -c clion run --target debug "$<$<NOT:$<CONFIG:All>>:-e${CMAKE_BUILD_TYPE}>"
    WORKING_DIRECTORY ${CMAKE_CURRENT_SOURCE_DIR}
)

add_executable(${PROJECT_NAME}_Z_DUMMY_TARGET ${SRC_LIST})
```

[image]

valeros, 2021-03-05:
Thanks, but that didn't help and it still doesn't work with the native CLion plugin for PlatformIO:

[image]

Do you use that plugin, right? Based on your screenshot it seems that you're just using bare CMake files?

MikeMitterer, 2021-03-08:
Hm, funny - I don't use the plugin at all. OK - thats not really true, it runs somehow in the background but usually I use the native cMake support built into CLion. My modified CMakeLists.txt generates the necessary run-entries