platformio/platformio-core #3841: `pio device monitor` does not exit on SIGINT

timoschwarzer, 2021-02-12:
What kind of issue is this?

- [X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: Arch Linux

**PlatformIO Version** (`platformio --version`): PlatformIO Core, version 5.1.0

### Description of problem

Sending SIGINT to `platformio device monitor` prints `--- exit ---` but does not actually exit.

#### Steps to Reproduce

1. Run `exec pio device monitor` in your shell
2. Find the PID with `ps -aux | grep pio` or your system monitor GUI
3. Run `kill -INT <PID>` in another shell

### Additional info

Background: I've configured CLion to run `pio device monitor` after uploading the project. CLion sends SIGINT to the process when clicking the stop or restart button in the IDE.

ivankravets, 2021-03-15:
This is an issue of PySerial Python package. Please forward it to https://github.com/pyserial/pyserial/issues

It's very easy to reproduce:
```
python -m serial.tools.miniterm
```

closed 2021-03-15 by ivankravets

timoschwarzer, 2021-03-15:
Thank you for clarification! :)

ivankravets, 2021-03-15:
I forgot to mention, double SIGINT works for me.

timoschwarzer, 2021-03-15:
> I forgot to mention, double SIGINT works for me.

Interesting. Unfortunately, it's not possible to make CLion send SIGINT twice.