platformio/platformio-core #3842: Pre/Post Actions on buildfs not called

qistoph, 2021-02-12:
### Configuration

**Operating system**: Ubuntu 20.04.2 LTS

**PlatformIO Version**: PlatformIO Core, version 5.1.0

### Description of problem
According to [the docs](https://docs.platformio.org/en/latest/projectconf/advanced_scripting.html#before-pre-and-after-post-actions) `buildfs` is a valid built-in target for Pre and Post Actions. However the pre and post events don't get called.

#### Steps to Reproduce

1. Define buildfs pre or post action
2. Run `pio run -t buildfs`
3. Check the output

### Actual Results vs Expected Results

```diff
 Processing d1_mini (platform: espressif8266; board: d1_mini; framework: arduino)
 --------------------------------------------------------------------------------
 Verbose mode can be enabled via `-v, --verbose` option
 CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
 PLATFORM: Espressif 8266 (2.6.2) > WeMos D1 R2 and mini
 HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
 PACKAGES: 
  - framework-arduinoespressif8266 3.20704.0 (2.7.4) 
  - tool-esptool 1.413.0 (4.13) 
  - tool-esptoolpy 1.20800.0 (2.8.0) 
  - tool-mklittlefs 1.203.200522 (2.3) 
  - tool-mkspiffs 1.200.0 (2.0) 
  - toolchain-xtensa 2.40802.200502 (4.8.2)
 LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
 LDF Modes: Finder ~ chain, Compatibility ~ soft
 Found 30 compatible libraries
 Scanning dependencies...
 No dependencies
 Building in release mode
+1. before_buildfs ...
 Building file system image from 'data' directory to .pio/build/d1_mini/spiffs.bin
 /index.html
+2. after_buildfs ...
 ========================= [SUCCESS] Took 0.32 seconds =========================
 ```

### If problems with PlatformIO Build System:

**platformio.ini**
```
[env:d1_mini]
platform = espressif8266
board = d1_mini
framework = arduino
extra_scripts = custom_tools.py
```

**custom_tools.py**
```
Import("env")

def before_buildfs(source, target, env):
    print("1. before_buildfs", source, target)

def after_buildfs(source, target, env):
    print("2. after_buildfs", source, target)

env.AddPreAction("buildfs", before_buildfs)
env.AddPostAction("buildfs", after_buildfs)
```

### Work around
 
 A work around that I've come up with is binding the events to the produced file's name.

 ```
# Work around for SPIFFS FS
env.AddPreAction("$BUILD_DIR/spiffs.bin", before_buildfs)
env.AddPostAction("$BUILD_DIR/spiffs.bin", after_buildfs)

# Work around for LittleFS
env.AddPreAction("$BUILD_DIR/littlefs.bin", before_buildfs)
env.AddPostAction("$BUILD_DIR/littlefs.bin", after_buildfs)
```

closed 2021-03-08 by valeros

valeros, 2021-03-08:
Hi @mcspr ! Thanks for reporting, should be fixed in the dev branch of the `espressif8266` platform. You can try it out using the [upstream version](https://docs.platformio.org/en/latest/platforms/espressif8266.html#stable-and-upstream-versions) of the platform.

valeros, 2021-03-08:
Hi @qistoph! Thanks for reporting, should be fixed in the dev branch of the `espressif8266` platform. You can try it out using the [upstream version](https://docs.platformio.org/en/latest/platforms/espressif8266.html#stable-and-upstream-versions) of the platform.